gnss-sdr/gnss-sdr #11: Google code is shutting down.

jiegec, 2015-08-16:
Consider change gflags and glog download location.

jiegec, 2015-08-16:
Patch for v0.0.5:
```
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -427,7 +427,7 @@
      message (STATUS " gflags library has not been found.")
      message (STATUS " gflags will be downloaded and built automatically ")
      message (STATUS " when doing 'make'. ")
-     set(gflags_MD5 "5fd4554d2ba84bf222a2fec0870d28df")
+     set(gflags_MD5 "99f54cc74fd8c934a507e8c57666870e")
      if("${CMAKE_CXX_COMPILER_ID}" STREQUAL "Clang")
          file(WRITE ${CMAKE_CURRENT_BINARY_DIR}/gflags-${gflags_RELEASE}/tmp/configure_osx
 "#!/bin/sh
@@ -454,7 +454,7 @@
      ExternalProject_Add(
           gflags-${gflags_RELEASE}
           PREFIX ${CMAKE_CURRENT_BINARY_DIR}/gflags-${gflags_RELEASE}
-          URL http://gflags.googlecode.com/files/gflags-${gflags_RELEASE}.zip
+          URL https://github.com/gflags/gflags/archive/v${gflags_RELEASE}.zip
           DOWNLOAD_DIR ${CMAKE_CURRENT_BINARY_DIR}/download/gflags-${gflags_RELEASE}
           URL_MD5 ${gflags_MD5}
           SOURCE_DIR ${CMAKE_CURRENT_SOURCE_DIR}/thirdparty/gflags/gflags-${gflags_RELEASE}
@@ -573,8 +573,7 @@
               glog-${glog_RELEASE}
               DEPENDS ${TARGET_GFLAGS}
               PREFIX ${CMAKE_CURRENT_BINARY_DIR}/glog-${glog_RELEASE}
-              SVN_REPOSITORY http://google-glog.googlecode.com/svn/trunk
-              SVN_REVISION -r "142"
+              GIT_REPOSITORY https://github.com/google/glog.git
               DOWNLOAD_DIR ${CMAKE_CURRENT_BINARY_DIR}/download/glog-${glog_RELEASE}
               SOURCE_DIR ${CMAKE_CURRENT_SOURCE_DIR}/thirdparty/glog/glog-${glog_RELEASE}
               BINARY_DIR ${CMAKE_CURRENT_BINARY_DIR}/glog-${glog_RELEASE}
@@ -589,7 +588,7 @@
               glog-${glog_RELEASE}
               DEPENDS ${TARGET_GFLAGS}
               PREFIX ${CMAKE_CURRENT_BINARY_DIR}/glog-${glog_RELEASE}
-              URL http://google-glog.googlecode.com/files/glog-${glog_RELEASE}.tar.gz
+              URL https://github.com/google/glog/archive/v${glog_RELEASE}tar.gz
               DOWNLOAD_DIR ${CMAKE_CURRENT_BINARY_DIR}/download/glog-${glog_RELEASE}
               URL_MD5 ${glog_MD5}
               SOURCE_DIR ${CMAKE_CURRENT_SOURCE_DIR}/thirdparty/glog/glog-${glog_RELEASE}
```

carlesfernandez, 2015-08-25:
Already fixed in next branch. All googlecode links have been replaced by their corresponding repository at GitHub.

closed 2015-08-25 by carlesfernandez