gnss-sdr/gnss-sdr #12: Cannot find python in next branch.

jiegec, 2015-08-26:
Error:
```
cd /Users/chenjack/temp/gnss-sdr/build/volk_gnsssdr_module/build && /usr/local/Cellar/cmake/3.3.0/bin/cmake -DCMAKE_C_COMPILER=/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -DCMAKE_CXX_COMPILER=/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -DCMAKE_INSTALL_PREFIX=/Users/chenjack/temp/gnss-sdr/build/volk_gnsssdr_module/install -DENABLE_STATIC_LIBS=ON -DPYTHON_EXECUTABLE=/opt/local/bin/python "-GUnix Makefiles" /Users/chenjack/temp/gnss-sdr/src/algorithms/libs/volk_gnsssdr_module/volk_gnsssdr
CMake Error at cmake/GrPython.cmake:105 (file):
  file FILE([TO_CMAKE_PATH|TO_NATIVE_PATH] path result) must be called with
  exactly three arguments.
Call Stack (most recent call first):
  CMakeLists.txt:59 (include)

-- Python checking for python >= 2.5
-- Python checking for python >= 2.5 - not found
-- Python checking for Cheetah >= 2.0.0
-- Python checking for Cheetah >= 2.0.0 - not found
CMake Error at CMakeLists.txt:64 (message):
  Python 2.5 or greater required to build VOLK_GNSSSDR
```

PYTHON_EXECUTABLE is defined here:
https://github.com/gnss-sdr/gnss-sdr/blob/d2e3d2f3b22869180655332eededf058c15ea2e9/CMakeLists.txt#L386
Why override the PYTHON_EXECUTABLE here? What if i use homebrew or fink instead of macports?

odrisci, 2015-08-26:
If you are using homebrew you can just delete that line and everything compiles just fine.

I'm not sure if there's a nice way to detect MacPorts in CMake, but something like:
```
IF( MACPORTS_FOUND )
  set(USE_MACPORTS_PYTHON "-DPYTHON_EXECUTABLE=/opt/local/bin/python")
ENDIF(MACPORTS_FOUND)
```

would be a better option. I don't have MacPorts installed so I can't really contribute to a generic solution.

By the way, I'm using my own formula for gnuradio in homebrew as well http://github.com/odrisci/homebrew-gnuradio. Note that this is very slightly modified from the original fork by @titanous

jiegec, 2015-08-26:
@odrisci That works.

carlesfernandez, 2015-08-28:
Thanks for pointing this out. I don't know how to detect Macports properly in CMake, but at  8be9ac859bf92e9d54ac92cce6ec798db8af9d71, I pushed a workaround that should work well for Fink and Homebrew, and at least let the user define PYTHON_EXECUTABLE as a CMake variable.

Please reopen the issue if this does not work for you.

closed 2015-08-28 by carlesfernandez