gnss-sdr/gnss-sdr #13: GPS+Galileo Config Hangs

odrisci, 2015-10-08:
I've noticed that configuring gnss-sdr to run with both GPS and Galileo leads to a hang. It seems all threads are blocked, this occurs about 1 second into the data file in general.

I've created a gist with three config files: 1) gps.conf; 2) gal.conf; 3) gps-gal.conf see [here](gist.github.com/odrisci/4ab6a78c42ddc156ba8e). These config files work with the CTTC data set available [here](http://sourceforge.net/projects/gnss-sdr/files/data/)

When I run these with the current next branch the individual system configs everything runs fine (beautifully even!) - but when I run the gps-gal.conf the receiver just stops processing after about 1 second of file time with no errors. The keyboard listener is still responsive, so typing q <return> works.

I've noticed this with other files as well.

Running on Mac OS X 10.10 using homebrew. Anyone else seeing this issue?

carlesfernandez, 2015-10-09:
> typing q works.

Hi,

It seems that the satellite channels are not well configured. In
gps-gal,conf, please use something like:

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=4
Channels_1B.count=4

Channels.in_acquisition=1

Channel0.signal=1B
Channel1.signal=1B
Channel2.signal=1B
Channel3.signal=1B
Channel4.signal=1C
Channel5.signal=1C
Channel6.signal=1C
Channel7.signal=1C

You will also need to synchronize the outputs of the navigation data
demodulation of GPS L1 and Galileo E1:

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder
TelemetryDecoder_1C.dump=false
TelemetryDecoder_1C.decimation_factor=4;

;######### TELEMETRY DECODER GALILEO CONFIG  ########
TelemetryDecoder_1B.implementation=Galileo_E1B_Telemetry_Decoder
TelemetryDecoder_1B.dump=false
TelemetryDecoder_1B.decimation_factor=1;

Please note that you can find an example of this configuration at
https://github.com/gnss-sdr/gnss-sdr/blob/master/conf/gnss-sdr_Hybrid_short.conf

Hope this helps,
Carles

odrisci, 2015-10-09:
Excellent! Thanks Carles.

In fact it was sufficient simply to set the line:
TelemetryDecoder_1C.decimation_factor=4;

Now it works beautifully.

I'm not sure why it experienced a hang previously, I suppose I'll have to look into the telemetry decoder in more detail.

Thanks again,

Cillian

closed 2015-10-09 by carlesfernandez