gnss-sdr/gnss-sdr #18: Gnuradio version detection incorrect

DL7AD, 2015-12-09:
I'm running Ubuntu 10.4 and because the repositories doesn't provide gnuradio 3.7.3+, I installed gnuradio manually. (Gnuradio 3.7.2 has been installed on the computer before)

But the cmake script still assumes, I have Gnuradio 3.7.2 installed.

```{r, engine='bash', count_lines}
sven@sven-desktop:/src/gnss-sdr/build$ cmake ../
...
Found Version
3.7.2.1
-- The GNU Radio version installed in your system is too old.
-- CMake cannot find GNU Radio >= 3.7.3
Go to http://gnuradio.org/redmine/projects/pybombs/wiki
and follow the instructions to install GNU Radio in your system.
CMake Error at CMakeLists.txt:363 (message):
  GNU Radio 3.7.3 or later is required to build gnss-sdr

-- Configuring incomplete, errors occurred!
See also "/src/gnss-sdr/build/CMakeFiles/CMakeOutput.log".

sven@sven-desktop:/src/gnss-sdr/build$ gnuradio-companion --version
GNU Radio Companion v3.7.8.1-297-ga463fecc

This program is part of GNU Radio
GRC comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it.
```

DL7AD, 2015-12-11:
I hacked the CMake files, that the version detection is skipped. I ran into an error where gnss-sdr expects the Gnuradio Libs in /usr/lib/x86_64-linux-gnu/ but mine were actually located in /usr/local/lib/

carlesfernandez, 2015-12-29:
Hi,

it seems that your system still have files from your former GNU Radio installation, and that confuses CMake. Please remove **all** the files belonging to Gnu Radio 3.7.2 (at /usr/lib/x86_64-linux-gnu/ , please remove all libgnuradio-* files, and and also the includes at /usr/include) before doing ```cmake ..```.

It seems that  Gnu Radio 3.7.2 was installed by a .deb package. If this is the case, ```sudo apt-get remove gnuradio-dev``` could work for you.

Please feel free to re-open the issue if this does not solve your problem.

Best regards,
Carles

closed 2015-12-29 by carlesfernandez