gnss-sdr/gnss-sdr #19: Make check fails in Ubuntu 15.04

odrisci, 2015-12-28:
I'm running Ubuntu 15.04 with:
o gcc: 4.9.2
o GNU Radio: 3.7.9git-226-ge0a70a92
o Boost: 1.55
o gnss-sdr: next branch - 217ed14

Running ``make check`` in debug mode fails with:

```sh
1: control_thread_test: /usr/include/boost/thread/pthread/condition_variable_fwd.hpp:81: boost::condition_variable::~condition_variable(): Assertion `!ret' failed.
1/5 Test #1: control_thread_test ..............***Exception: Other  0.78 sec
```

The issue appears to be with the order of destructors and the ``concurrent_queue`` object. The concurrent queue is destroyed *before* the threads that are reading from it. These threads are still waiting on the ``condition_variable`` when the ``condition_variable`` destructor is called, which throws the above exception.

Note that I have not seen this issue when running gnss-sdr on a Mac, which has a later version of boost (1.59).

I think this is the same issue as was raised [here](http://stackoverflow.com/questions/4905114/multithreaded-queue-destruction-in-c)

odrisci, 2015-12-28:
A patch to remove the error is given here: odrisci/gnss-sdr@3f3a2a7

This is not an ideal solution, as it involves lots of sleeping, but it does cause ``make check`` to pass.

carlesfernandez, 2015-12-29:
Hi,

I reproduced your setup (Ubuntu 15.04, gcc 4.9.2, Boost 1.55) in a Virtual Machine but I cannot reproduce the error: both ```make check``` and ```run_tests``` work well for me. I'll need to dig a little bit more to see why this is happening. I do not see any problem in sleeping the collector threads during some milliseconds, but I'd like to understand why in order to come up with a more robust solution (100 ms is enough for all machines? is it too much?)

Thank you very much for spotting this issue!

Best regards,
Carles

carlesfernandez, 2015-12-30:
Hi,

I finally could reproduce the error, which is only appearing in my VM when the code has been built in ```Debug``` mode but not in ```Release```. I've pushed a commit that fix the issue, avoiding the use of sleeping time (that could be harmful in the case of channel queue, since tracking blocks could receive outdated data from acquisition blocks): c5407a51064f30d13fb0679d23cb801c6efbb1d0

I've tested it on Ubuntu 14.04.2, 15.04 and Mac OS X 10.11.2. Could you please confirm that it works for you?

Thanks, 
Carles

odrisci, 2015-12-30:
Hi Carles,

Thanks for that. I'm still getting some random segfaults in any one or more of:
```
trk_test
flowgraph_test
gnss_block_test
```
Random in the sense that running 

```sh
make check
```

will succeed about 3 times out of 5, and fail with a segfault in one of the above the other times. On the other hand I have not seen any failures in ``run_tests`` as yet.

It appears that the ``gnuradio_block_test`` and ``control_thread_test`` tests are now fine. I'll look into this, but I think it may be possible to apply a similar solution to the test harnesses as you have applied to the ``control_thread`` and ``channel`` blocks, i.e. flushing the concurrent queue and then waiting for the thread to join.

Thanks for your help on this,

Cillian

odrisci, 2016-01-01:
Ok,

I've investigated this a little further. All the crashes (segfaults) seemed to occur in the thrift library, so I thought this might be related to [this](https://gnuradio.org/redmine/projects/gnuradio/wiki/ControlPort#Thrift-092-Shutdown-Patch).

I'm not sure exactly how to apply the patch to thrift (in fact it seems to have already been incorporated into the thrift library installed via pybombs). To test this, I've reverted to gnuradio v3.7.7 and this seems to to solved the segfaults.

I think this issue can be closed here, however it seems to be a problem with gnuradio versions > 3.7.7. This might need more investigation. For now I'm happy to have a fully running, check passing version of gnss-sdr running under Ubuntu 15.04 :)

Thanks for the quick turnaround, very much appreciated.

Cillian

carlesfernandez, 2016-01-04:
Hi Cillian,

I've been also tracking down those segfaults. When updating to GR v3.7.10, errors seem to have gone away. I'm going to close this issue here, thank you so much for helping in improving the receiver!

Best regards,
Carles

closed 2016-01-04 by carlesfernandez