gnss-sdr/gnss-sdr #24: bit_transition_flag

mmajoral, 2016-03-17:
Minor bug in pcps_acquisition.cc:

if d_bit_transition_flag is set then the constructor forces d_max_dwells to be 1. The value of d_max_dwells in pcps_acquisition.cc is then not consistent with the value of max_dwells_ in GpsL1CaPcpsAcquisition, which is set to either 2 or whatever value is specified in the conf file.

luis-esteve, 2016-05-06:
Initially the aim of bit_transition_flag was enable or disable a strategy to deal with bit transitions in GPS signals: process two dwells and take maximum test statistics. This flag invalidated the value of max_dwells.

Add documentation in next commits

luis-esteve, 2016-05-10:
SInce commit [https://github.com/gnss-sdr/gnss-sdr/commit/5b9683baad55acdbaba775cfc53b105f2ec0b0ad](url) the strategy to deal with bit transitions signals is the overlap/save technique. You can read about this technique in [http://gps-ttff.tripod.com/FFTAcquisition.pdf](url), specifically see figure 5.

commit fd5cd42b (2016-05-10): Correct minor bug in pcps acquisition blocks

Change the initialization of max_dwells in pcps acquisition adpaters
due its value is forced to be 1 in gr_block constructor if boolean
"bit_transition_flag" is set to TRUE

Resolves: #24
closed 2016-05-10 by luis-esteve