gnss-sdr/gnss-sdr #26: Setting fixed PRN for each Channel does not work

JJZim, 2016-03-18:
Hello,
if I set a static PRN for a (GPS) Channel, the aquisition still is done for all available channels in a somehow wired order. The channel never checks for its preconfigured PRN.

JJZim, 2016-03-18:
I think I've worked out a patch for this problem.
How can I submit it to the project?

luis-esteve, 2016-03-20:
Hi,
you can send your patch to this mailing list or make a pull-request from
your github account. All these options are explained in our How to
contribute to the source code page
<http://gnss-sdr.org/documentation/how-contribute-source-code>

Hope this helps

Regards

Luis

JJZim, 2016-03-24:
Hey, 
Thanks for your reply.
After some more detailed research on the source code (I'm working on master branch, is this boy the way a bad idea?) I came to the conclusion, that the Channel Statemachine as well as the message passing protocol (if you can call it one) are not really made to support channels with fixed satellites. I although implemented a solution of which I think it will work (at least for my constellation). However, as I am not really into details of the whole implementation I think that this fix maybe  violates dependencies. 
Was there any progress regarding that special feature since the last commit to the master branch? It's very hard to search through the commits of the next branch.

Thanks in advance

JJZim, 2016-04-19:
Okay, so after a bit of additional investigation, I think my problem or understanding of how to use fixed satellites for each channel is different to how it is actually implemented in gnss-sdr.

For me it would be greate to fix all channels to certain satellites (each channel exactly to one) and let them do their work (even if this means that they fail during acquisition all the time).

However, the fixed sattelites for now are only "hints" for the channels and maybe (after a acquisition fail for example) totally different satellites are assigned to each channel.
Did I get it right? And if so, is this the intention of the implementation and will not be changed in the future?

Regards,
Jan

luis-esteve, 2016-04-19:
Hi Jan, you are right. The fixed satellites are the initial ones searched
by the acquisition process. If the acquisition fails, then the satellite is
tried to be acquired once again, but if this second acquistion (named
reacquisition) falis again then the satellite is discarded and the channel
try to acquire the following PRN in the queue.

I don't know if this process will be changed in future versions of channel,
but feel free to implement your own state machine and share it with the
gnss-sdr community.

Hope this helps

Regards

Luis

closed 2016-04-21 by JJZim