gnss-sdr/gnss-sdr #31: Change in VOLK's API breaks building

carlesfernandez, 2016-05-29:
The interface of  volk_32f_index_max_16u have changed as per https://github.com/gnuradio/volk/commit/cb83e1ce4280f84d85eb7d7b6c17179b57f20d98 . This breaks building if GNSS-SDR is using Volk built from the latest source code snapshot. However, Volk’s master branch version is still tagged as ‘1.2.2’, so there is no way to detect this API change (for instance if Volk was installed through the libvolk-dev package, or from the ‘1.2.2’ tag.)

carlesfernandez, 2016-05-29:
Opened an issue at https://github.com/gnuradio/volk/issues/84

carlesfernandez, 2016-05-29:
Workaround at 0aa980a9a10a6d7bdb01c8cdac7a4cf2b4e9a028 

However, this will break building of code tagged with v0.0.7 as soon as VOLK 1.2.3 is released

Alex--C, 2016-06-24:
The workaround should be included in the "next" branch, right?

I'm getting the following error when I try to build it:

..../src/algorithms/acquisition/gnuradio_blocks/pcps_acquisition_cc.cc:339:84: error: cannot convert 'unsigned int*' to 'uint16_t* {aka short unsigned int*}' in argument passing
		      volk_32f_index_max_16u(&indext, d_magnitude, effective_fft_size);

latinovic, 2016-06-27:
On Debian gnss-sdr version 0.0.7-1 mentioned workaround is not included so this behavior is expected, but I had tried latest version from git repository and build failed with the same error as well.
I investigated further and it seems that  VOLK_VERSION stuck on 1.1.
The reason for that is in cmake/Modules/FindVolk.cmake:
**set(VOLK_VERSION ${PC_VOLK_VERSION})**

After commenting this line I was able to build gnss-sdr successfully.

odrisci, 2016-08-30:
This fix caused some problems for me using the Galileo E1 pcps acquisition with a very high sampling rate. It turns out that the 16 bit index isn't enough. For example, with 50 Msps, there are 200 000 samples in every E1b code period,  but the current fix will only look in the first 65 535 (2<sup>16</sup>-1) samples for the maximum value. The situation would be much worse for E5, where the code period is 10 ms.

I've created a pull request (#34) that addresses this by using the `volk_32f_index_max_32*` functions and 32 bit unsigned indexing.

carlesfernandez, 2016-09-06:
A possible solution is to "port" the functions `volk_32f_index_max_32*` to volk_gnsssdr, so we could avoid requiring a very recent version of VOLK. I'll will work on it ASAP.

carlesfernandez, 2016-09-19:
Fixed in the `next` branch from commit https://github.com/gnss-sdr/gnss-sdr/commit/438ba45c16fac91a0aa0aab54a1453f07515c57f 
A new kernel `volk_gnsssdr_32f_index_max_32u` has been added. This allows building when using Ubuntu versions older than 16.10 (which come with versions of VOLK older than 1.3).

This fix requires the full rebuild of gnss-sdr, since it contains an addition to the VOLK_GNSSSDR API.

carlesfernandez, 2016-09-23:
Added more protokernels to the original VOLK kernel (AVX aligned and unaligned, SSE4.1 unaligned, SSE unaligned, and NEON), and contributed back to VOLK in https://github.com/gnuradio/volk/pull/99

This is what I get in my machine:

```
RUN_VOLK_GNSSSDR_TESTS: volk_gnsssdr_32f_index_max_32u(131071,1987)
a_sse4_1 completed in 140.732ms
u_sse4_1 completed in 136.963ms
a_sse completed in 181.364ms
u_sse completed in 184.856ms
a_avx completed in 71.131ms
u_avx completed in 72.978ms
generic completed in 309.554ms
Best aligned arch: a_avx
Best unaligned arch: u_avx
```

So AVX seems to achieve pretty good results. Not sure about the NEON implementation, I think there is room for improvement there. This is what a Raspberry Pi 3 gets:

```
RUN_VOLK_GNSSSDR_TESTS: volk_gnsssdr_32f_index_max_32u(131071,1987)
generic completed in 1578.3ms
neon completed in 730.067ms
Best aligned arch: neon
Best unaligned arch: neon
``` 

I'm closing this issue since it seems to be solved. Please feel free to reopen it if this change triggers other problems, or open a new one.

closed 2016-09-23 by carlesfernandez