gnss-sdr/gnss-sdr #32: Build problems with cmake in openSUSE 13.1

theGidy, 2016-08-04:
I tried to compile gnss-sdr with cmake. I had to to make some changes in the checkout code for gflags and link to another aclocal version to start `make` under openSUSE 13.1. But in the end i run in a compiler error, and i can't find out what is the cause for this. 

My first suggestion is, that i run in a compile error because is used aclocal-1.13 instead of aclocal-1.14. Unfortunately only aclocal-1.13 is available in openSuse 13.1.  Is it really necessary to use this version or it is possible to switch the call just to `/usr/bin/aclocal` instead of ``/usr/bin/aclocal-1.14`

## Environment:
- openSUSE 13.1 (Kernel: 3.12.53-40-default)
- cmake (v3.3.2-1.2)
- make (v3.82-160.2.1)
- gnuradio (v3.7.9.2) 

## Changes:
- automake (v1.13) installed for aclocal-1.14 command => set an link to aclocal-1.13. Also change the link to test-drives
- change gflags `git clone git://...` to `git clone http://...` because company firewall is blocking git://

## Attachment:
[cmake_output.txt](https://github.com/gnss-sdr/gnss-sdr/files/401657/cmake_output.txt)
[make_output.txt](https://github.com/gnss-sdr/gnss-sdr/files/401604/make_output.txt)

carlesfernandez, 2016-08-20:
Please check that `/usr/lib64`actually come first in your `$PATH` than `/opt/anaconda/lib`

theGidy, 2016-09-06:
Issue is solved. It was a problem with the path. I removed anaconda from the path to compile it correctly.

closed 2016-09-06 by theGidy