gnss-sdr/gnss-sdr #36: Tutorial "My first position fix" doesn't work

darkstar007, 2016-10-30:
If the config file is cut'n'pasted as is on the web page, and then the filename is fixed to point to the correct file, an error occurs:

> gnss-sdr --config_file=test2.conf 
>linux; GNU C++ version 6.2.0 20160927; Boost_106100; UHD_003.009.005-0-unknown
>
>Initializing GNSS-SDR v0.0.8 ... Please wait.
>Logging will be done at "/tmp"
>Use gnss-sdr --log_dir=/path/to/log to change that.
>./example_capture.dat: No such file or directory
>The configuration file has not been found.
>Please create a configuration file based on the examples at the 'conf/' folder 
>and then generate your own GNSS Software Defined Receiver by doing:
>$ gnss-sdr --config_file=/path/to/my_GNSS_SDR_configuration.conf
>GNSS-SDR program ended.

looking at the log file says:

>Log file created at: 2016/10/30 11:56:11
>Running on machine: scarlett
>Log line format: [IWEF]mmdd hh:mm:ss.uuuuuu threadid file:line] msg
>I1030 11:56:11.010908 13734 file_configuration.cc:196] Configuration file test2.conf opened with no errors
>I1030 11:56:11.011131 13734 gnss_block_factory.cc:148] Getting SignalSource with implementation File_Signal_Source
>I1030 11:56:11.011562 13734 file_signal_source.cc:178] file_signal_source: Unable to open the samples file ./example_capture.dat, exiting the program.

so the output from the command line says the config file couldn't be read but the log file says it was read - **which seems very confusing!!**

So there are **2** issues:

1. There is a [GNSS-SDR] line missing from the example conf file in the tutorial
2. The handling of the conf file needs a little attention.

Thanks,

Matt

carlesfernandez, 2016-10-31:
This has now been fixed in the website. See http://gnss-sdr.org/my-first-fix/ 

Thank you for reporting this issue.

darkstar007, 2016-10-31:
OK, that works. Thanks.

closed 2016-10-31 by carlesfernandez