gnss-sdr/gnss-sdr #37: Another issue with Tutorial "My first position fix"

darkstar007, 2016-10-30:
After fixing the config file as I described in #36 I have now come across another issue. When I run the example I get:

>gnss-sdr --config_file=test2.conf 
>linux; GNU C++ version 6.2.0 20160927; Boost_106100; UHD_003.009.005-0-unknown
>
>Initializing GNSS-SDR v0.0.8 ... Please wait.
>Logging will be done at "/tmp"
>Use gnss-sdr --log_dir=/path/to/log to change that.
>Processing file >/home/matt/gnss/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat, which contains 1600000000 [bytes]
>GNSS signal recorded time to be processed: 99.999 [s]
>E1030 12:43:27.096798 16579 gnss_flowgraph.cc:159] itemsize mismatch: copy1:0 using 8, stream_to_vector0:0 using 4
>E1030 12:43:27.096873 16579 control_thread.cc:110] Unable to connect flowgraph
>Total GNSS-SDR run time 0.000353 [seconds]
>GNSS-SDR program ended.

Unfortunately its difficult to tell where this error message has come from, so I'm unsure where to change the configuration file to make it work.

Thanks,

Matt

carlesfernandez, 2016-10-31:
This has now been fixed in the website. See http://gnss-sdr.org/my-first-fix/ 

Thank you for reporting this issue.

darkstar007, 2016-10-31:
OK, that works now. Thanks.

closed 2016-10-31 by carlesfernandez

ciarz83, 2017-08-10:
I followed the instruction on  http://gnss-sdr.org/my-first-fix/ but when I try to to run the example I get:

Initializing GNSS-SDR v0.0.6.git-- ... Please wait.
Logging will be done at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
Processing file /home/michele/GNSS_work/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat, which contains 1600000000 [bytes]
GNSS signal recorded time to be processed: 99.999 [s]
Using Volk machine: avx_64_mmx_orc
E0810 11:57:04.919533  7730 gnss_flowgraph.cc:355] itemsize mismatch: gps_l1_ca_pvt_cc0:0 using 8, null_sink0:0 using 2
E0810 11:57:04.919651  7730 control_thread.cc:139] Unable to connect flowgraph
Total GNSS-SDR run time 0.001819 [seconds]
GNSS-SDR program ended.

Probably I have to change something in the configuration file.
Thank you in advance 

Michele

yinhonggen, 2018-09-23:
$ gnss-sdr --config_file=./GNSS-WORK/SDR-receiver.conf
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

Initializing GNSS-SDR v0.0.9.git-next-c296617 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
Processing file /home/ucas/yinhonggen/GNSS-WORK/example_capture.dat, which contains 1064566784 [bytes]
GNSS signal recorded time to be processed: 6.65154 [s]
Using Volk machine: sse4_2_64_orc
E0923 22:50:44.569898  9512 gnss_flowgraph.cc:257] itemsize mismatch: valve0:0 using 8, interleaved_short_to_complex0:0 using 2
E0923 22:50:44.569995  9512 control_thread.cc:128] Unable to connect flowgraph
Total GNSS-SDR run time: 0.000395011 [seconds]
GNSS-SDR program ended.

mlinuxguy, 2018-09-23:
same issue here with all new checkout and following gnss.conf guide:
testuser@raspsdr:~ $ gnss-sdr --config_file=/home/testuser/gnss.conf
linux; GNU C++ version 6.2.0 20161010; Boost_106100; UHD_003.009.005-0-unknown

Initializing GNSS-SDR v0.0.9.git-next-c2966176 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
Processing file /home/testuser/gnss_work/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat, which contains 1600000000 [bytes]
GNSS signal recorded time to be processed: 99.999 [s]
WARNING: Implementation 'GPS_L1_CA_Observables' of the Observables block has been replaced by 'Hybrid_Observables'.
Please update your configuration file.
WARNING: Implementation 'GPS_L1_CA_PVT' of the PVT block has been replaced by 'RTKLIB_PVT'.
Please update your configuration file.
WARNING: The global parameter name GNSS-SDR.internal_fs_hz has been DEPRECATED.
WARNING: Please replace it by GNSS-SDR.internal_fs_sps in your configuration file.
Set GNSS-SDR.internal_fs_sps in configuration file
E0923 13:57:24.631060 31021 gnss_flowgraph.cc:338] Set GNSS-SDR.internal_fs_sps in configuration
E0923 13:57:24.631708 31021 control_thread.cc:128] Unable to connect flowgraph
Total GNSS-SDR run time: 0.00397618 [seconds]
GNSS-SDR program ended.