gnss-sdr/gnss-sdr #38: log_dir parameter not found

jaredd, 2016-11-02:
Following the hello world tutorial and running
> gnss-sdr --config_file=hello_world.conf
linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.0.git-28-gc66cb1ba

Initializing GNSS-SDR v0.0.8.git-next-74a23c5 ... Please wait.
Logging will be done at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
...

but running
> gnss-sdr --log_dir=$HOME/temp --config_file=hello_world.conf
linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.0.git-28-gc66cb1ba

ERROR: unknown command line flag 'log_dir'

So it appears that there's no log_dir flag.  Interesting that the GLOG_log_dir env var does work which means some definition is getting through from glog.  More info:

> gnss-sdr --version
linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.0.git-28-gc66cb1ba

gnss-sdr version 0.0.8.git-next-74a23c5

carlesfernandez, 2016-11-03:
I'm trying it on different machines and this flag works fine for me. Did you build GFlags and Glog from source, were they downloaded and built by the CMake script, or you installed them from the .deb packages? I have not seen this error before, and I can't reproduce it.

jaredd, 2016-11-03:
The packages were installed via the cmake script in gnss-sdr.  I can find the source in the <GNSS_SRC_DIR>/thirdparty subdir.  I tried to set ENABLE_OWN_GLOG=ON but still no dice, same error.

jaredd, 2016-11-03:
Looking at the gflags subdir, it appears that it wasn't built.  There was no build subdir and no Makefiles.  Could this cause a problem?

carlesfernandez, 2016-11-04:
This should be fixed in commit cc5cf874770061dc3380c5aa3e6fe5555f11c75 

Could you pull from the next branch and see if it works for you? You might need `libtool`, `autoconf` and `automake` packages installed.

jaredd, 2016-11-07:
Yep, that worked.  Thanks.

closed 2016-11-07 by jaredd