gnss-sdr/gnss-sdr #40: Feature Request: GPX output support

Artoria2e5, 2016-12-24:
[GPX](http://www.topografix.com/gpx.asp) (GPS eXchange) is used by quite a few GPS-related applications, notably OpenStreetMap, for storing and importing waypoints, routes and tracks. It features a more verbose (compared to KML) XML usage in paths which allows various metadata to be added to individual waypoints, paths and/or files. With GPX output, GNSS-SDR would be able to store useful information, including timestamps, DOP, and altitude in a more machine-friendly way.

carlesfernandez, 2018-05-05:
GPX format support added in the `next` branch (see #166)

carlesfernandez, 2018-05-07:
The GPX output available in the 'next' branch now stores position solutions (latitude, longitude) with their associated altitude, timestamp and DOP parameters.

I'm closing this issue since the feature is now implemented in the  'next' branch. Thanks to @Artoria2e5 for the suggestion and to @acebrianjuan for addressing this issue.

closed 2018-05-07 by carlesfernandez