gnss-sdr/gnss-sdr #41: gnss-sdr fails in dynamic environments

hahnpv, 2017-01-06:
As mentioned on the gnss-sdr developers list I am using gnss-sdr as a receiver for a (simulated) rocket trajectory from earth to orbit. gnss-sdr fails to calculate PVT above 50km. I figured out what was causing this - in gps_l1_ca_ls_pvt.cc on line 178:

`            //ToDo: Find an Observables/PVT random bug with some satellite configurations that gives an erratic PVT solution (i.e. height>50 km)
            if (d_height_m > 50000)
                {
                    b_valid_position = false;
                    return false;
                }`

Commenting this block of code out allows for calculations above 50km. I am able to now track a trajectory past 125km but after awhile the PVT starts failing again by generating ridiculously large altitudes (e+15)  and lat/lon positions that dance around. I don't know if altitude or velocity is the culprit. 

I am able to take the observables' pseudoranges and the trackers' prompt_I and decode both ephemeris and valid lat/lon/alt by post-processing in SoftGNSS, a Matlab-based offline GPS code. So I am pretty well convinced that the correct observables are being generated, and a problem exists in the calculation of the PVT. This is alluded to in the comment to the code block above.

I forked gnss-sdr and am willing to assist in debugging this, but any additional information or insights would be greatly appreciated. Thanks.

philip hahn

carlesfernandez, 2017-01-11:
Just for the records (the conversation is going on in the mailing list): the issue seems solved in commit https://github.com/gnss-sdr/gnss-sdr/commit/235aa7735795d8ae32e0d70f820e1cb81097f161

carlesfernandez, 2017-02-03:
Commit https://github.com/gnss-sdr/gnss-sdr/commit/0362476864afca3f1553b365caab740f3bc19f1b adds a more in-deep fix by reworking  the Tracking, Telemetry, Observables and PVT blocks. Now the PVT solution should be numerically stable.

closed 2024-08-30 by carlesfernandez