gnss-sdr/gnss-sdr #42: What's the best way to calculate Doppler?

zhao19851019, 2017-01-22:
Current calculation Doppler occupy too much memory, as well as a good way to calculate the Doppler do?

shaunakdixit, 2017-03-08:
Hi,
I would like to work on this issue.
could you please tell me the algorithm that is used by the GNSS-SDR.
there are few more ways to reduce the memory usage like
1- using of STL vectors 
2- Another fact I discovered- For same program iostream for Input/Output uses more memory than stdio. The difference is small - 2.8M vs 2.6M, but still worth noting.

carlesfernandez, 2017-08-03:
The Acquisition stage is quite demanding in terms of memory and computational load. 
The description of the currently available implementations is at http://gnss-sdr.org/docs/sp-blocks/acquisition/ and the source code is under https://github.com/gnss-sdr/gnss-sdr/tree/next/src/algorithms/acquisition/ 

Feel free to contribute with specific suggestions on how to improve current implementations or proposing new approaches.

closed 2017-08-03 by carlesfernandez