gnss-sdr/gnss-sdr #44: Compilation error in v 0.0.9 when -DENABLE_CUDA=ON

chmeul, 2017-03-22:
Hi,

It looks like in v0.0.9, in the Gnss_Synchro class, the Code_Phase_secs property was renamed  to Rem_code_phase_secs, but this has not been updated in the GPU enabled tracking loop. Because of this, when compiling gnss-sdr with -DENABLE_CUDA=ON, the following error occurs:

` ~/pybombs/src/gnss-sdr/src/algorithms/tracking/gnuradio_blocks/gps_l1_ca_dll_pll_tracking_gpu_cc.cc:442:34: error: ‘class Gnss_Synchro’ has no member named ‘Code_phase_secs’`

Thanks for checking! #

Arribas, 2017-03-29:
Hi,
In principle it is fixed in the next branch. Please test the update and report if it works,
Thank you!
Javi.

chmeul, 2017-03-29:
Hi Javi.

It works now, thanks!
I think it is the same issue reported by Stefan on the mailing list.

Erwan

closed 2017-03-29 by chmeul