gnss-sdr/gnss-sdr #54: issue with Two_Bit_Packed_File_Signal_Source and real data type

chmeul, 2017-06-16:
Dear all,

We used to process 2-bit packed real data by converting them beforehand to bytes. Today we investigated the use of the Two_Bit_Packed_File_Signal_Source with SignalSource.sample_type=real, by using the same parameters as in the configuration file form the repo:
gnss-sdr_GPS_L1_nsr_twobit_packed.conf

and we got the following error:

```
E0616  gnss_flowgraph.cc:129] itemsize mismatch: char_to_float0:0 using 4, valve0:0 using 8
E0616 control_thread.cc:116] Unable to connect flowgraph
```

The error also arises using the conf file from the repo.

Could you please double check and try to locate the issue, as we were not able to do it?

We also tried with another Two_Bit_Packed_File_Signal_Source configuration with iq data (SignalSource.sample_type=iq) and everything worked fine.

Thanks!

odrisci, 2017-06-16:
There's a bug in `Two_Bit_Packed_File_Signal_Source` around [line 214](https://github.com/gnss-sdr/gnss-sdr/blob/32e4d227af032365cfd7f9291d9f2c2fca57d99d/src/algorithms/signal_source/adapters/two_bit_packed_file_signal_source.cc#L214). We need to update to change the first argument to `gnss_sdr_make_valve` to `sizeof(float)` when the source is real.

Similarly for the `file_sink` and `throttle` blocks on the following lines. I'll do this when I get back to my machine

chmeul, 2017-06-19:
Thanks!
We are now testing. For the moment the tracking does not work but we still have to confirm that it does not come from our side, in the way we are repacking the data from our multi-antenna receiver.

carlesfernandez, 2017-08-03:
I'm closing this since this particular issue seems now solved. Please feel free to reopen it if required.

closed 2017-08-03 by carlesfernandez

zoqueton99, 2021-06-08:
Hello everyone, I've encountered the same problem in the last version. I'm working on the next branch, the issue seems to be in line 56 and 71.
Is this right or I'm missing something?
Thanks!