gnss-sdr/gnss-sdr #57: L2C telemetry decoder not found

Peterthegreat, 2017-08-01:
Hello,

I am running this [config](https://github.com/gnss-sdr/gnss-sdr/blob/master/conf/gnss-sdr_GPS_L2C_USRP1_realtime.conf) modified for UHD_Signal_Source, but I get this error:

E0801 06:48:39.465976 13095 gnss_block_factory.cc:1278] TelemetryDecoder_2S.GPS_L2C_Telemetry_Decoder: Undefined implementation for block
Segmentation fault

I looked it up and looks like the source did not compiled successfully, but the software was installed via apt-get install command. Any tips? Also tried a reinstall.

My setup: USRP N210, Linux kali 4.9.0-kali4-amd64 #1 SMP Debian 4.9.30-2kali1 (2017-06-22) x86_64 GNU/Linux; "gnss-sdr is already the newest version (0.0.8-1+b5)" . Is this an old build? (i noticed the latest version is 0.0.9) but even so, the software should work.

Thanks

carlesfernandez, 2017-08-03:
Hi @Peterthegreat ,

packages installed via apt-get tend to get stuck in a version that becomes obsolete quickly. For instance, Ubuntu 17.04 gets gnss-sdr 0.0.9, but 16.10 provides 0.0.8. In Debian, 0.0.9 is only provided by sid/buster. 

The good thing is that you have all dependencies already there, so building from source should run smoothly. I'd recommend this:

```
$ sudo apt-get remove gnss-sdr
$ git clone https://github.com/gnss-sdr/gnss-sdr
$ cd gnss-sdr/build
$ git checkout next
$ cmake ..
$ make 
$ sudo make install
```

The decoding of the CNAV message should then be available with the configuration you mention. Some relevant improvements were made to that block since 0.0.8

Hope this helps,
Carles

Peterthegreat, 2017-08-09:
Thank you for the fast response. I will reopen the issue if there's any problem

closed 2017-08-09 by Peterthegreat