gnss-sdr/gnss-sdr #59: What is copy2:0?

eduardocarrascal, 2017-08-11:
Hello, 

First I want to say thank you for being up to date with all the issues posted. Second, I keep getting an error about item size mismatch...

E0810 20:56:46.229605 13762 gnss_flowgraph.cc:154] itemsize mismatch: copy2:0 using 8, complex_byte_to_float_x20:0 using 2

E0810 20:56:46.229755 13762 control_thread.cc:139] Unable to connect flowgraph

I am using the code from the "My first position fix" portion of the guide and I am trying to process my own data. I understand that there must be some consistency when it comes to item types so I made sure that different parts of the code blocks matched (copy0:0, copy1:0, direct_resampler_make_conditioner_cb0:0 and valve0:0). These correspond to the .item_types of the DataTypeAdapter, the InputFilter, the Resampler and the Acquisition_1C respectively. I figured this out through trial and error. However, I reached this message and I cannot figure out what item_type copy2:0 is referring to. It can't be Tracking_1C can it? That one only accepts gr_complex and therefore it cannot be modified. Then again it would explain why it has a value of 8. Let me know what you think; what should I change to make the item types match?

carlesfernandez, 2017-08-12:
Could you please post your configuration file?

In general, you need to end up with gr_complex at the output of the Signal Conditioner, except in the case of using the ```GPS_L1_CA_DLL_PLL_C_Aid_Tracking``` implementation for Tracking_1C, which is currently the only Tracking implementation that accepts cshort at its input.

BTW, the table at the bottom of http://gnss-sdr.org/docs/tutorials/understanding-data-types/ could help.

eduardocarrascal, 2017-08-13:
[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_hz=2000000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=File_Signal_Source
SignalSource.filename=/home/eduardocarrascal/GNSS_Data/2017-07-31-22-00-06-085783768_A.txt
SignalSource.item_type=cbyte
SignalSource.sampling_frequency=1000000
SignalSource.freq=1575420000
SignalSource.samples=0

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner
DataTypeAdapter.implementation=Pass_Through
DataTypeAdapter.item_type=cbyte
InputFilter.implementation=Pass_Through
InputFilter.item_type=cbyte
Resampler.implementation=Direct_Resampler
Resampler.sample_freq_in=1000000
Resampler.sample_freq_out=2000000
Resampler.item_type=cbyte

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=8
Channels.in_acquisition=1
Channel.signal=1C

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=cbyte
Acquisition_1C.threshold=0.008
Acquisition_1C.doppler_max=10000
Acquisition_1C.doppler_step=250

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.pll_bw_hz=40.0;
Tracking_1C.dll_bw_hz=4.0;

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder

;######### OBSERVABLES CONFIG ############
Observables.implementation=GPS_L1_CA_Observables

;######### PVT CONFIG ############
PVT.implementation=GPS_L1_CA_PVT
PVT.averaging_depth=100
PVT.flag_averaging=true
PVT.output_rate_ms=10
PVT.display_rate_ms=500

eduardocarrascal, 2017-08-13:
> tutorials/understanding-data-types/ could help.

Posted.

eduardocarrascal, 2017-08-13:
Also, if I set the acquisition item.type to gr_complex, it gives me a mismatch error as well. Is that what you meant by the output of the Signal Conditioner being gr_complex?

eduardocarrascal, 2017-08-15:
Also, I was wondering: does it matter that the file is .txt instead of
.dat? Because either way it doesn't work but maybe you have noticed a
difference.

Eduardo C

carlesfernandez, 2017-08-16:
Hi Eduardo,

Tracking implementations for `cbyte` are still not there, so you need to convert your samples to `cshort` or `gr_complex`. You can need a Data Type Adapter for that. Try with:

```
SignalSource.item_type=ibyte
```

```
;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner
DataTypeAdapter.implementation=Ibyte_To_Complex
InputFilter.implementation=Pass_Through
InputFilter.item_type=gr_complex
Resampler.implementation=Direct_Resampler
Resampler.sample_freq_in=1000000
Resampler.sample_freq_out=2000000
Resampler.item_type=gr_complex
```

and then replace ```Acquisition_1C.item_type=cbyte```by ```Acquisition_1C.item_type=gr_complex```.

Hope this helps.

PD: is is ok if the file is .txt, as long as it contains binary data.

eduardocarrascal, 2017-08-16:
So, many things have happened since I posted. I tried building the program from the source, and that has given me more promising results. It locks onto a signal but then loses it. (Good thing is I no longer see the error).

After I read your message and applied the changes, it locks onto less PRN's but loses them again. Here's what I mean,

gnss-sdr --config_file=./my_GPS_receiver.conf
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

Initializing GNSS-SDR v0.0.9.git-next-f8c7f14 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
Processing file /home/eduardocarrascal/GNSS_Data/2017-07-31-22-00-06-085783768_A.txt, which contains 40000000 [bytes]
GNSS signal recorded time to be processed: 4.998 [s]
Using Volk machine: avx_64_mmx_orc
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 01 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 02 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 03 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 04 (Block Unknown)
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 05 (Block IIR-M)
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 06 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 07 (Block IIR-M)
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 09 (Block IIF)
Current receiver time: 1 [s]
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 03 (Block IIF)
Loss of lock in channel 0!
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 01 (Block IIF)
Loss of lock in channel 1!
Loss of lock in channel 3!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 02 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 10 (Block IIF)
[... 17 lines cut ...]
Current receiver time: 3 [s]
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 03 (Block IIF)
Loss of lock in channel 4!
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 05 (Block IIR-M)
Loss of lock in channel 6!
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 07 (Block IIR-M)
Current receiver time: 4 [s]
Loss of lock in channel 0!
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 01 (Block IIF)
Loss of lock in channel 3!
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 10 (Block IIF)
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 02 (Block IIR)
Loss of lock in channel 5!
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 06 (Block IIF)
Loss of lock in channel 7!
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 09 (Block IIF)
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 03 (Block IIF)
Loss of lock in channel 4!
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 05 (Block IIR-M)
Stopping GNSS-SDR, please wait!
Total GNSS-SDR run time: 9.0033 [seconds]
GNSS-SDR program ended.

eduardocarrascal, 2017-08-16:
And after I apply your changes...

gnss-sdr --config_file=./my_GPS_receiver.conf
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

Initializing GNSS-SDR v0.0.9.git-next-f8c7f14 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
Processing file /home/eduardocarrascal/GNSS_Data/2017-07-31-22-00-06-085783768_A.txt, which contains 40000000 [bytes]
GNSS signal recorded time to be processed: 9.999 [s]
Using Volk machine: avx_64_mmx_orc
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 15 (Block IIR-M)
Current receiver time: 1 [s]
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 31 (Block IIR-M)
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 13 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 23 (Block IIR)
Loss of lock in channel 0!
Current receiver time: 2 [s]
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 27 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 06 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 30 (Block IIF)
Loss of lock in channel 1!
Loss of lock in channel 3!
Current receiver time: 3 [s]
Loss of lock in channel 2!
Loss of lock in channel 4!
Current receiver time: 4 [s]
Loss of lock in channel 5!
Loss of lock in channel 0!
Current receiver time: 5 [s]
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 17 (Block IIR-M)
Current receiver time: 6 [s]
Current receiver time: 7 [s]
Loss of lock in channel 6!
Current receiver time: 8 [s]
Current receiver time: 9 [s]
Stopping GNSS-SDR, please wait!
Total GNSS-SDR run time: 6.95793 [seconds]
GNSS-SDR program ended.

eduardocarrascal, 2017-08-22:
A new problem that I'm facing is that the signal is lost during the
tracking and navigation cannot start. I notice that the PRN's that are
displayed are acquired are very inconsistent even though the command is the
same. I already posted what I mean by this. Could you give the thread
another look? Thanks for all your help by the way, it's reassuring having
someone that works on the same field.

Eduardo C

odrisci, 2017-08-30:
What is the actual sampling rate and data type of your recording?

In your original config file you had 

```
SignalSource.item_type=cbyte
SignalSource.sampling_frequency=1000000
```

In the new file, you have:

```
SignalSource.item_type=ibyte
SignalSource.sampling_frequency=2000000
```

So was the sampling rate 1 MHz or 2 MHz?

Also, there was a typo as the `File_Signal_Source` does not recognise `cbyte` so it was (quietly) assuming you meant `gr_complex`. I find it useful to create a local `Log` folder and run `gnss-sdr` with the command: 

```
gnss-sdr --config_file=path_to_config.conf --log_dir=./Log
```

Have a look in the files `gnss-sdr.INFO` `gnss-sdr.WARNING` and `gnss-sdr.ERROR` files in the `Log` folder - lots of useful information in there. (Note if you don't specify the `--log_dir` switch, then the log files will be written to a temporary folder, see the first few lines output by `gnss-sdr` to see where that temp folder is located in your system). The `WARNING` log would show you that the `File_Signal_Source` doesn't recognise the specified `item_type` and so is assuming `gr_complex` (see [here](https://github.com/gnss-sdr/gnss-sdr/blob/d77f48f55cfd3f1fa0fc6ae08686a5a16af4b725/src/algorithms/signal_source/adapters/file_signal_source.cc#L111) ).

I don't know if you are building from source, but if you are then try building in `Debug` mode - this puts lots more information in the `gnss-sdr.INFO` log. This is particularly useful for acquisition as it shows you what the threshold was for each failed and successful acquisition, if every satellite is over the threshold, it means your threshold is too low and should be increased (reduce the `pfa` setting in your acquisition channels).

Hope this helps

eduardocarrascal, 2017-08-30:
Thank you for all that information. The original sampling rate was 5.0 MHz. But even when I tried that it still failed. I will try using the log and yes I built it from source, so should I uninstall and then build it again? Or should I just build it again without uninstalling? Thanks again.

odrisci, 2017-08-30:
Ah, interesting. Could you try setting the following then:

```
GNSS-SDR.internal_fs_hz=5000000
```
and

```
SignalSource.sampling_frequency=5000000
```

The key point is this: the `SignalSource.sampling_frequency` **must** be the sampling frequency of the recording, otherwise nothing works. Secondly, `GNSS-SDR.internal_fs_hz` **must** be the sampling frequency at the output of the signal conditioner (the acquisition and tracking blocks use this property in their processing, again if this is wrong nothing will work).

In your case I think you have configured the `Resampler` to `PassThrough` so the sampling rate at the output of the signal conditioner should be the same as the sampling rate in the file: 5 MHz.

eduardocarrascal, 2017-08-30:
It still fails, maybe frequency in should also be 5 MHz?

eduardocarrascal, 2017-08-30:
No, that still failed

linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

Initializing GNSS-SDR v0.0.9.git-next-f8c7f14 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
Processing file /home/eduardocarrascal/GNSS_Data/2017-08-22-14-20-05-231640176_A.txt, which contains 40000000 [bytes]
GNSS signal recorded time to be processed: 1.999 [s]
Using Volk machine: avx_64_mmx_orc
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 40000. Due to alignment requirements
   64 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
Current receiver time: 1 [s]
Stopping GNSS-SDR, please wait!
Total GNSS-SDR run time: 2.5885 [seconds]
GNSS-SDR program ended.

odrisci, 2017-09-11:
Hi Eduardo

At this point it becomes difficult for me to say what is not working. Looking at the output of `gnss-sdr` above it seems that everything is working fine -- except that you are not acquiring any satellites. So this is probably not a `gnss-sdr` issue, but a data issue.

If you compile `gnss-sdr` in debug mode and re-run the software, then the log file will contain information on the acquisition threshold and test statistics. It may be that the threshold is simply set too high and this is the easiest way to determine that. Here you would look for satellites whose test statistics were much higher than average, but still below the threshold. If all satellite signals are about the same and well below the threshold, you have a different issue.

Now, I don't know anything about your test setup, so the list of things that could possibly be wrong with the data is long (apologies if most of these seem completely obvious):
1. Antenna:
    1.1 Do you have an antenna attached?
    1.2. If you do is it a GPS antenna?
    1.3. If it is a GPS antenna, is it an active antenna? A passive antenna just won't give you sufficient SNR without an external LNA
    1.4. If it is an active antenna, is it powered?
    1.5. If so, does it have a good view of the sky? Its not going work indoors, `gnss-sdr` is not a high sensitivity receiver
2. RF front-end
    2.1 Are you sure the of the centre frequency (1575.42 MHz according to your config file)?
    2.2 Are you sure of the data type (interleaved complex bytes according to the config file)? Is the data 8 bit two's complement? Is it packed 1-bit or 2-bit samples?
    2.3 Are you sure of the sampling rate (5 Msps)?
    2.4 Have you checked the spectrum using Matlab/Octave/Python/your favourite numerical programming environment? You should have a nice noise like spectrum tapering towards the edges. Large spurs may cause problems for acquisition. Are you being jammed?
    2.5 Have you checked the histograms of the samples? Both real and imaginary components should follow a zero mean Gaussian distribution.
    2.6 What is your clock uncertainty? Check the front-end specs. A clock uncertainty of X parts per million contributes (X x 1.57542) kHz of Doppler uncertainty at L1: e.g. 10 ppm ==> 15.7542 kHz Doppler uncertainty. Make sure your `Acquisition_1C.doppler_max` parameter matches this uncertainty
    2.7 You only have 2 seconds of data in your file - this is very short. Even if you acquire the signals you are not going to be able to do much with them

closed 2018-01-18 by carlesfernandez