gnss-sdr/gnss-sdr #60: Question on generating two local PRNs (one of them inverted)

fgallardo, 2017-08-11:
Hi there,

This is not a problem/issue, but a question. Maybe it is quite simple and the question is stupid (sorry for that if this is the case). 

Why are you generating two local versions of the PRN code, one of them inverted?

I mean, if I am no wrongly interpreting the code: You generate the local PRN code at fs, then FFT, then conjugate. So far, so good. 

But why are you generating a second local PRN, this time inverted?

Thanks a lot and sorry for bothering!

BR
Fran

https://github.com/gnss-sdr/gnss-sdr/blob/d77f48f55cfd3f1fa0fc6ae08686a5a16af4b725/src/algorithms/acquisition/gnuradio_blocks/galileo_pcps_8ms_acquisition_cc.cc#L139

fgallardo, 2017-08-18:
Forget about it.

You're taking your Rx I/Q branches and then checking what branch is the one with the greater correlation with the PRN. Then you start to operate on that branch only.

carlesfernandez, 2024-11-15:
We have a collection of different acquisition algorithms. The main one is called `pcps_acquisition`, but there are other experimental implementations for you to try out. In the specific implementation you are mentioning, a second code is generated to determine the sign of the following symbol.

closed 2024-11-15 by carlesfernandez