gnss-sdr/gnss-sdr #62: L1/L2 data collection

sriramyab, 2017-08-22:
Hello, 
       I have a USRP-N210 series. I have both DBSRX2 and WBX daughterboards. I want to collect both L1/L2 GPS signals for my research project. In your readme. txt you mentioned "Configuring the USRP X300 with two front-ends for receiving signals in L1 and L2 bands". I didn't quite understand what sort of setup you used for this purpose? Did you use two different USRPs each equipped with a daughterboard and synced them using an external clock? Or did you accomplish this using a single USRP? If you used only 1 USRP, do you know if I can use USRP-N210 as well for this purpose? Or do I need USRP-X300? What sort of daughterboard needs to be used for this purpose? 

Thank you, 

Cheers
Ramya

yealwang, 2018-03-04:
Hello,
I want to ask if the GNSSSDR is working good for L2CM signal?

antonioramosdet, 2018-03-06:
GNSS-SDR works with GPS L2C only in the next branch. Some bugs have been detected when working with that signal (i.e. incorrect decodification of some parameters in the CNAV navigation message and problems when combining observations coming for other signals). I am working for fixing these issues so please be aware for updates in the source code (next branch) of this repository

shuoyuanxu, 2021-09-22:
Hello,

Is GNSS-SDR works fine with GPS L2 with USRP N210? Or should I buy a different device? Thank you.

carlesfernandez, 2024-11-15:
GPS L2C should work well with USRP N210. Please open a new issue if you find difficulties in getting it working.

closed 2024-11-15 by carlesfernandez