gnss-sdr/gnss-sdr #64: USRP Real Time Config File

kylecavs, 2017-08-24:
Hi,
I am trying to run the gnss-sdr_GPS_L1_USRP_realtime.conf file with a USRP B200. However, the output provides the following error:

linux; GNU C++ version 4.8.4; Boost_105600; UHD_003.009.005-0-g32951af2

Initializing GNSS-SDR v0.0.9.git-next-265caed ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
-- Detected Device: B200
-- Operating over USB 3.
-- Initialize CODEC control...
-- Initialize Radio control...
-- Performing register loopback test... pass
-- Performing CODEC loopback test... pass
-- Asking for clock rate 16.000000 MHz... 
-- Actually got clock rate 16.000000 MHz.
-- Performing timer loopback test... pass
-- Setting master clock rate selection to 'automatic'.
-- Asking for clock rate 32.000000 MHz... 
-- Actually got clock rate 32.000000 MHz.
-- Performing timer loopback test... pass
Sampling Rate for the USRP device: 2000000.000000 [sps]...
UHD RF CHANNEL #0 SETTINGS
Actual USRP center freq.: 1575420000.000297 [Hz]...
PLL Frequency tune error 0.000297 [Hz]...
Actual daughterboard gain set to: 60.000000 dB...
Setting RF bandpass filter bandwidth to: 1000000.000000 [Hz]...
Check for front-end LO: locked ... is Locked
Starting a TCP Server on port 2101
The TCP Server is up and running. Accepting connections ...
E0824 08:23:47.129644 15475 gnss_flowgraph.cc:84] copy(3): insufficient connected output ports (1 needed, 0 connected)
E0824 08:23:47.129838 15475 control_thread.cc:126] Unable to start flowgraph
Total GNSS-SDR run time: 0.000730433 [seconds]
GNSS-SDR program ended.
Stopping TCP Server on port 2101

Can you provide some insight on this? Also is there a preferred config file that I should use for a B200?

Thanks,
Kyle

carlesfernandez, 2017-08-25:
Hi Kyle,

I've just realized that this particular file was still in an old configuration notation. The example has been updated at 4f292bbe 

The configuration part related to the USRP B200 seems to work well according to your log.

HarryHook, 2017-09-22:
@carlesfernandez  @kylecavs 
I am trying to run GNSS-SDR with a non-active antenna with 40dB gain, a USRP N210 and  SBX(400M-4.4GHz) daughter board to receive GPS L1 C/A signals.  
Do I have to use an active antenna? I first prepared to collect data for non-real-time processing and then start real-time processing.**But It seems that the collection is not the correct data**.Here are my GNU Radio Companion GUI parameter settings.Can you help me see if there is any problem, or let me refer to your GUI. Thanks!!
[image]
[image]
[image]

nebukadnezar, 2018-01-31:
Hi Kyle/Carlos, 
I ran into the same problem. Further to this, I'm in the situation that I have multiple B200s on the same system so I need to be able to specify in the config file which B200 to use by giving their device serial number. I also need GNSS-SDR to use RX2 antenna.... does anyone have a clue how to pass those options in the config file?

I don't know what Carlos's responses to your report mean. There do not appear to be any fixed config files. And checking out next results in uncompilable code as per my other issue reported a few minutes ago.

nebukadnezar, 2018-01-31:
This now works using the next build - would consider that fixed.

commit 4f292bbe (2017-08-25): Fix examples with old configuration notation. Fixes #64"
commit bf2e796c (2017-09-10): Fix examples with old configuration notation. Fixes #64"
closed 2019-08-02 by carlesfernandez