gnss-sdr/gnss-sdr #65: Another issue with Tutorial "My first position fix"

vamsib123, 2017-08-25:
After performing all the mentioned steps, I'm getting this error. As a beginner I'm not able to rectify the problem.
>gnss-sdr --config_file=./my-first-GNSS-SDR-receiver.conf
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

Initializing GNSS-SDR v0.0.6.git-- ... Please wait.
Logging will be done at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
Processing file /home/vamsi/work/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat, which contains 1339452928 [bytes]
GNSS signal recorded time to be processed: 83.71480799999999 [s]
Using Volk machine: avx2_64_mmx_orc
E0825 16:48:45.303230 13893 gnss_flowgraph.cc:355] itemsize mismatch: gps_l1_ca_pvt_cc0:0 using 8, null_sink0:0 using 2
E0825 16:48:45.303338 13893 control_thread.cc:139] Unable to connect flowgraph
Total GNSS-SDR run time 0.001826 [seconds]
GNSS-SDR program ended.
gnss-sdr: /usr/include/boost/thread/pthread/condition_variable_fwd.hpp:81: boost::condition_variable::~condition_variable(): Assertion `!ret' failed.
Aborted (core dumped)

caparrag, 2017-08-25:
Have you installed from package manager or did you built from source? last time I tried to install from packet (don't remember if in Ubuntu or Debian) I struggled with some error similar to this, and I solved by building master from source. I ask this because I remember about the 0.0.6 version

vamsib123, 2017-08-25:
I installed from the package on ubuntu 16.04. I'll try building from source. But try fixing this because all the beginners will consider the direct install easier. Thanks!

eduardocarrascal, 2017-08-27:
Have any of you guys had problems with losing the signal? If not, do you know any threads that address this problem? 

Also, to answer your question, I got the same error when I messed with item_types in the configuration file. Are you using cbyte or cshort by any chance? I just kept it as ishort since this is how most USRP's save the data. Then I performed data conversion to gr_complex. Check out my issue "What is copy2:0?" (#59) and you'll see the response Carles posted which got rid of the message that you have. Basically everything should be gr_complex except that first item_type which should be ishort or ibyte. Hope this helps. 

Now please help me...

vamsib123, 2017-08-29:
I'm am a beginner and just started with this repo. I found very less support and the mailing lists have almost stopped. Sorry, there is nothing i can help you with.

CTTCGeoLab, 2017-09-26:
Same problem, installation from package:

gnss-sdr: /usr/include/boost/thread/pthread/condition_variable_fwd.hpp:81: boost::condition_variable::~condition_variable(): Assertion `!ret' failed.

Followed your tutorial but it didn't work (Ubuntu VM 16.04). I will try to compile from source as mentioned.

This is the program output:
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

Initializing GNSS-SDR v0.0.6.git-- ... Please wait.
Logging will be done at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
Processing file /home/cttc/data/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat, which contains 1600000000 [bytes]
GNSS signal recorded time to be processed: 99.999 [s]
Using Volk machine: avx_64_mmx_orc
E0926 13:34:56.886004  2925 gnss_flowgraph.cc:355] itemsize mismatch: gps_l1_ca_pvt_cc0:0 using 8, null_sink0:0 using 2
E0926 13:34:56.886137  2925 control_thread.cc:139] Unable to connect flowgraph
Total GNSS-SDR run time 0.002364 [seconds]
GNSS-SDR program ended.
gnss-sdr: /usr/include/boost/thread/pthread/condition_variable_fwd.hpp:81: boost::condition_variable::~condition_variable(): Assertion `!ret' failed.
Aborted (core dumped)

Thanks

carlesfernandez, 2017-10-06:
A notice has been written at http://gnss-sdr.org/my-first-fix/ warning about this issue. Please use v0.0.9 or, even better, the `next` branch of the source code.

closed 2018-01-18 by carlesfernandez