gnss-sdr/gnss-sdr #69: Proposal for non-blocking acquisition

odrisci, 2017-09-11:
As the sampling rate and/or frequency uncertainty increases the acquisition block becomes the bottleneck in performance. This is a problem for real time operation with Galileo (at least 5 Msps) or with front-ends with poor oscillators (like the HackRF, which has a 10 ppm oscillator, leading to a frequency uncertainty in the +/- 15 kHz range).

The idea is to modify the `pcps_acquisition_cc` block so that the bulk of the work is done in a separate thread. A new parameter `blocking` is passed at construction, if this is `true` the behaviour is as normal, if its `false` then the acquisition block skips over samples that arrive while the processing thread is busy. The default value is `true`, as the existing implementation is blocking.

This I think is what the `pcps_multithread_acquisition` block was designed for, but this never worked for me. This implementation has a couple of advantages  in that it re-uses the existing `pcps_acquisition_cc` block and it uses standard C++11 threading concepts.

The disadvantage of this approach is that we introduce yet another thread, with the associated overhead that comes with that. However, I've found that this enables real-time operation that would otherwise not be possible as all threads are held up waiting for the acquisition block to process.

I've pushed a working implementation to my fork [here](/odrisci/gnss-sdr/tree/non_blocking_acq). Comments are welcome.

carlesfernandez, 2017-09-14:
Dear @odrisci ,

this was exactly the idea behind `pcps_multithread_acquisition`, but we were just unable to make it play well with the GNU Radio schedule. You did, and it is even integrated into `pcps_acquisition_cc`. Very cool!

I'm quite clogged these days with other duties, but I'll try to test it during the weekend with complex configurations. May I have your permission to merge / add it in a new branch in the upstream repository ?

Thank you very much for looking into this.

odrisci, 2017-09-14:
Hi Carles

I've created a pull request for this at #72 - if you have a chance to test it yourself that would be great. I have been using it in real time operation over a number of hours using both a BladeRF and a HackRF and it seems to be fine, but it would be good to have some more tests.

Cillian

carlesfernandez, 2017-09-18:
Ok, I did a new test:

taking as a input the file 2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat, available at https://sourceforge.net/projects/gnss-sdr/files/data/ 

I tested the following configuration:

```
[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=4000000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=File_Signal_Source
SignalSource.filename=/Users/carlesfernandez/Documents/data/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat  ; <- PUT YOUR FILE HERE
SignalSource.item_type=ishort
SignalSource.sampling_frequency=4000000
SignalSource.samples=0
SignalSource.repeat=false
SignalSource.dump=false
SignalSource.dump_filename=../data/signal_source.dat
SignalSource.enable_throttle_control=false

;######### SIGNAL_CONDITIONER CONFIG ############

SignalConditioner.implementation=Signal_Conditioner

DataTypeAdapter.implementation=Ishort_To_Complex
InputFilter.implementation=Pass_Through
InputFilter.input_item_type=gr_complex
InputFilter.output_item_type=gr_complex
Resampler.implementation=Pass_Through
Resampler.item_type=gr_complex
Resampler.sample_freq_in=4000000
Resampler.sample_freq_out=4000000

;######### CHANNELS GLOBAL CONFIG ############
Channels_1B.count=8
Channels.in_acquisition=1
Channel.signal=1B

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1B.implementation=Galileo_E1_PCPS_Ambiguous_Acquisition
Acquisition_1B.item_type=gr_complex
Acquisition_1B.if=0
Acquisition_1B.sampled_ms=4
Acquisition_1B.pfa=0.0001
Acquisition_1B.doppler_max=15000
Acquisition_1B.doppler_step=125
Acquisition_1B.cboc=false
Acquisition_1B.dump=false
Acquisition_1B.dump_filename=./acq_dump.dat
; Acquisition_1B.blocking=false

;######### TRACKING GLOBAL CONFIG ############
Tracking_1B.implementation=Galileo_E1_DLL_PLL_VEML_Tracking
Tracking_1B.item_type=gr_complex
Tracking_1B.if=0
Tracking_1B.pll_bw_hz=20.0;
Tracking_1B.dll_bw_hz=3.0;
Tracking_1B.early_late_space_chips=0.15;
Tracking_1B.very_early_late_space_chips=0.6;
Tracking_1B.dump=false
Tracking_1B.dump_filename=../data/veml_tracking_ch_

;######### TELEMETRY DECODER CONFIG ############
TelemetryDecoder_1B.implementation=Galileo_E1B_Telemetry_Decoder
TelemetryDecoder_1B.dump=false

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=false
Observables.dump_filename=./observables.dat

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=PPP_Static  ; options: Single, Static, Kinematic, PPP_Static, PPP_Kinematic
PVT.iono_model=Broadcast ; options: OFF, Broadcast, SBAS, Iono-Free-LC, Estimate_STEC, IONEX
PVT.trop_model=Saastamoinen ; options: OFF, Saastamoinen, Estimate_ZTD, Estimate_ZTD_Grad
PVT.output_rate_ms=100;
PVT.display_rate_ms=500;
PVT.dump=false
PVT.dump_filename=./PVT
PVT.nmea_dump_filename=./gnss_sdr_pvt.nmea
PVT.flag_nmea_tty_port=false
PVT.nmea_dump_devname=/dev/pts/4
PVT.flag_rtcm_server=false;
PVT.rtcm_tcp_port=2101
PVT.rtcm_MT1045_rate_ms=5000
PVT.rtcm_MSM_rate_ms=1000
PVT.flag_rtcm_tty_port=false;
PVT.rtcm_dump_devname=/dev/pts/1
```

The results are the following:

1. With the default `Acquisition_1B.blocking=true`, the processing of that file always gets PVT fixes starting from second 43.
2. With `Acquisition_1B.blocking=false`, the processing is much faster but most of the executions can’t get position fixes. I just got them in one execution out of ten.
       NOTE: This is solved by setting `SignalSource.enable_throttle_control=true`, but then we lose the performance gain.

In my opinion, the reason is that, when setting `Acquisition_1B.blocking=false`, the number of samples that the flow graph accepts is only a matter of how fast the host machine is. When using RF front-ends, the samples enter the receiver at the pace the front-end delivers, so the receiver has time to perform acquisition and jump into tracking after few samples the acquisition succeed. 

However, in the case of using a file as a signal input, the ingestion of samples is much faster than with a front-end. Hence, the receiver “consumes” an indeterminate number of samples while the thread is acquiring, so the results of the acquisition may be applied far beyond the instant in which acquisition was computed. 

As you have found, there could be a sweet spot in which, depending on the input sample rate delivered by the front-end and the capacity of the host computer, the whole mechanism actually works, and then everything runs much faster than with `blocking=true`. But this will be always dependant on the processing speed of the machine that is executing acquisition. For a given sampling rate and a given host machine, `blocking=false` could then be the best option.

In summary, I’m totally in favor of merging this branch into `next` (we should work on the 16sc version before that), since the default `blocking=true` still works the same in every machine (I've tried in just two, so more testing is needed, but i think it does play well with the GNU Radio scheduler) but we should clearly document this option stating that the results are machine-dependant if it is set to `false`.

Do you agree with this analysis?

odrisci, 2017-09-19:
I agree with you here. In fact I would never use `blocking=false` for a file signal source, this is something we should add to the documentation. I use it exclusively for real-time operation. I think the documentation should also make it very clear that the use of `blocking=false` means that the receiver operation is non-deterministic. It can be very annoying to try and track down bugs when you can't get your software into a consistent state.

One of the nice things with the acquisition block is that it records the number of samples processed at the time that the acquisition was performed and stores it in the output `GnssSynchro` object - this allows the tracking loop to propagate the code phase forward in time based on the Doppler and time that has elapsed since the acquisition. Of course if many seconds have elapsed (as could happen with a file source) then the resultant error may be outside the tracking loop pull-in range - another reason not to use the non-blocking approach in a file sample source. However, I have found that even for very large sampling rates (say 40 Msps) the acquisition block still provides outputs that are useable for tracking, even if the computation takes hundreds of milliseconds to compute.

I've added a quick test to `GpsL1CaPcpsAcquisitionTest` and `GalileoE1PcpsAmbiguousAcquisitionTest` in my local repo. Each is a repetition of the `ValidationOfResults` test case, only with `blocking=false` added to the config. The tests pass just fine, but this does not really capture the fact that the acquisition block is non-blocking (a future modification could revert `pcps_acquisition_cc` to its previous implementation and the test would still pass). I don't see an easy way of doing this other than exposing the `d_blocking` and `d_worker_thread_active` member variables in the block's interface. What do you think?

I've been thinking a bit about the implementation of the `pcps_acquistion_sc` - it really pains me to have two classes with almost identical implementation, it makes for a real code maintenance headache. One possible solution would be to make the current implementation a generic `pcps_acquisition` block that would take an `item_type` parameter as input. From what I can see the current `pcps_acquisition_sc` simply converts the input from `lv_16sc_t` to `gr_complex` using a volk kernel and then implements precisely the same functionality as `pcps_acquisition_cc`. I'm hacking together a solution along these lines for my personal use, do you think this would be a good idea? Or is the separation of blocks into specific implementations for specific data types a key design decision within `gnss-sdr`?

Perhaps this last point should be a separate issue

carlesfernandez, 2017-09-20:
I've created a new issue at https://github.com/gnss-sdr/gnss-sdr/issues/75 for this last point, which is an important design choice.

carlesfernandez, 2017-09-20:
About:

> I've added a quick test to `GpsL1CaPcpsAcquisitionTest` and `GalileoE1PcpsAmbiguousAcquisitionTest` in my local repo. Each is a repetition of the `ValidationOfResults` test case, only with [...]

I did just the same tests locally and came to the same conclusions :smile:

In a first thought, I'm not comfortable with exposing `d_worker_thread_active` in the public interface. Testing concurrent threads is really tricky stuff. I think that ensuring that everything works as expected if `blocking=true`, and clearly documenting what are the implications of `blocking=false`, is a neat way to go.

carlesfernandez, 2017-10-03:
Implemented for `pcps_acquisition_sc` in e352978777f88f3dd7ff9539d6a872277799d92f (no templates, just ugly code repetition)

I think it's ready to be merged to `next`.

carlesfernandez, 2017-10-03:
Merged to `next` and documented at http://gnss-sdr.org/docs/sp-blocks/acquisition/

carlesfernandez, 2017-10-05:
I'm closing this issue, since the non-blocking acquisition option is already in `next`.

closed 2017-10-05 by carlesfernandez