gnss-sdr/gnss-sdr #70: GPS Satellite ID out of range

BlairFonville, 2017-09-12:
HI,
I ran the Next branch overnight, and found two issues. These are probably unrelated, so I've titled this post to reflect just one:
1. The _warning_ log shows ~1.5 minutes of 

> W0912 [timestamp] 9444 rtcm.cc:2836] GPS satellite ID must be between 0 and 31, but PRN 32 was found

2. The second issue is that the receiver was stalled. I believe this is unrelated because the time at which it stalled (i.e. ceased to provide a PVT solution) was ~4 hours after the timestamp of the warnings in item 1 above.

Are these known issues?

Thanks,
Blair

carlesfernandez, 2017-09-14:
Thank you for the report. None of those problems should happen. Can I ask for the receiver configuration you are using, so we can try to reproduce this behavior? 

We have been running the receiver for days without seeing this stall, but we saw it in the past and  maybe a hidden bug is still there. Looking into it.

BlairFonville, 2017-09-18:
Hi Carles,

Sorry for the delay; I've was out of town for a bit. I am working with real-time tracking using a USRP B210. I'm running on Ubuntu 14.04. 

The reason I tried the _Next_ build, was because I was having similar issues with  _Main_. Though, with _Main_ the symptoms of my failures were different: 
Typically (with  _Main_)  I would see **NaN** values reported as the Lat/Long/El, after otherwise tracking successfully for some period of time (maybe several hours). Additionally, I would receive reports of large time synchronization violations (I don't recall the exact syntax of the messages) - and the time synch errors would build slowly in time offset. I don't see these messages/errors with the _Next_ build, but maybe it's related - I don't know.

Thanks,
Blair

BlairFonville, 2017-09-18:
Another note, I'm running Ubuntu in a VM on Windows. I've allocated 16GB RAM and 6 cores, so I don't think there should be any resource problems, but I wonder if it could be related.

Blair

BlairFonville, 2017-09-19:
One correction to item 2 from the OP - the receiver does not actually stall; it only fails to track and compute a PVT. These some of the messages I receive from the INFO log after the failure has occurred:

> 00:05:38.24: point pos error (chi-square error nv=9 vv=14.5 cs=13.8)
00:07:13.24: point pos error (chi-square error nv=9 vv=14.0 cs=13.8)
00:09:24.24: point pos error (chi-square error nv=9 vv=16.7 cs=13.8)
00:11:51.24: point pos error (chi-square error nv=9 vv=21434038178.8 cs=13.8)
00:14:01.24: point pos error (chi-square error nv=9 vv=17.3 cs=13.8)
00:14:47.24: point pos error (chi-square error nv=9 vv=14.3 cs=13.8)
00:15:32.24: point pos error (chi-square error nv=9 vv=14.4 cs=13.8)
00:16:16.24: point pos error (chi-square error nv=9 vv=17.3 cs=13.8)
00:17:02.24: point pos error (chi-square error nv=9 vv=19.1 cs=13.8)
00:17:07.24: point pos error (chi-square error nv=9 vv=14.0 cs=13.8)
00:18:56.24: point pos error (chi-square error nv=9 vv=37.6 cs=13.8)
00:18:57.24: point pos error (chi-square error nv=9 vv=142.7 cs=13
I0919 10:27:50.402086  3164 gps_l1_ca_dll_pll_tracking_cc.cc:422] Loss of lock in channel 5!
I0919 10:27:50.402142  3134 gnss_flowgraph.cc:424] Channel 5 TRK FAILED satellite GPS PRN 03 (Block IIF)
I0919 10:27:50.404589  3153 gps_l1_ca_dll_pll_tracking_cc.cc:257] Starting tracking of satellite GPS PRN 03 (Block IIF) on channel 5
I0919 10:27:50.404595  3153 gps_l1_ca_dll_pll_tracking_cc.cc:263] PULL-IN Doppler [Hz]=2500 Code Phase correction [samples]=-0.00738849 PULL-IN Code Phase [samples]=658.007
I0919 10:27:50.404613  3134 gnss_flowgraph.cc:396] Channel 5 ACQ SUCCESS satellite GPS PRN 03 (Block IIF)
I0919 10:27:50.681671  3193 rtklib_solver.cc:300] RTKLIB rtkpos error message: 22:59:17.83: point pos error (lack of valid sats ns=4)
22:59:18.83: point pos error (lack of valid sats ns=6)
22:59:19.83: point pos error (lack of valid sats ns=6)
22:59:20.83: point pos error (lack of valid sats ns=6)
22:59:21.83: point pos error (lack of valid sats ns=6)
22:59:22.83: point pos error (lack of valid sats ns=6)
22:59:23.83: point pos error (lack of valid sats ns=6)
22:59:24.83: point pos error (lack of valid sats ns=6)
22:59:25.83: point pos error (lack of valid sats ns=6)
22:59:26.83: point pos error (lack of valid sats ns=6)
22:59:27.83: point pos error (lack of valid sats ns=6)
22:59:28.83: point pos error (lack of valid sats ns=6)
22:59:29.83: point pos error (lack of valid sats ns=6)
22:59:30.83: point pos error (lack of valid sats ns=6)
22:59:31.83: point pos error (lack of valid sats ns=6)
22:59:32.83: point pos error (lack of valid sats ns=6)
22:59:33.83: point pos error (lack of valid sats ns=6)
22:59:34.83: point pos error (lack of valid sats ns=6)
22:59:35.83: point pos error (lack of valid sats ns=6)
22:59:36.83: point pos error (lack of valid sats ns=6)
22:59:37.83: point pos error (lack of valid sats ns=6)
22:59:38.83: point pos error (lack of valid sats ns=6)
22:59:39.83: point pos error (lack of valid sats ns=6)
22:59:40.83: point pos error (lack of valid sats ns=6)
22:59:41.83: point pos error (lack of valid sats ns=6)
22:59:42.83: point pos error (lack of valid sats ns=6)
22:59:43.83: point pos error (lack of valid sats ns=6)
22:59:44.83: point pos error (lack of valid sats ns=6)
22:59:45.83: point pos error (lack of valid sats ns=6)
22:59:46.83: point pos error (lack of valid sats ns=6)
23:03:11.83: point pos error (chi-square error nv=9 vv=16.1 cs=13.8)
23:04:18.83: point pos error (chi-square error nv=9 vv=33.0 cs=13.8)
23:04:19.83: point pos error (chi-square error nv=9 vv=339.1 cs=13.8)
23:04:20.83: point pos error (chi-square error nv=9 vv=999.4 cs=13.8)
23:04:21.83: point pos error (chi-square error nv=9 vv=1923.9 cs=13.8)
23:04:22.83: point pos error (chi-square error nv=9 vv=3305.2 cs=13.8)
23:04:23.83: point pos error (chi-square error nv=9 vv=4709.2 cs=13.8)
23:04:24.83: point pos error (chi-square error nv=9 vv=6801.9 cs=13.8)
23:04:25.83: point pos error (chi-square error nv=9 vv=9105.4 cs=13.8)
23:04:26.83: point pos error (chi-square error nv=9 vv=11760.8 cs=13.8)

BlairFonville, 2017-09-19:
From the log, it appears that the problem may lie with a day-boundary bug. The sdr repeatedly logs the above messages with timestamps jumping backwards once the time approaches midnight UTC (00:00:00). The logging continues this way indefinately (until I kill the process).

Blair Fonville

carlesfernandez, 2018-01-19:
Dear Blair,

we have rewritten the internal finite state machine that manages the acquisition/tracking switch within channels. The receiver has been working now for more than 12 hours (including a midnight passing) delivering continuous PVT fixes. We will leave it working for some more days in order to check that it does not get stalled anymore.

Thank you for reporting this bug.

closed 2018-12-14 by carlesfernandez