gnss-sdr/gnss-sdr #75: Implementation of GNU Radio blocks for different data types

carlesfernandez, 2017-09-20:
Quoting @odrisci from https://github.com/gnss-sdr/gnss-sdr/issues/69:

> I've been thinking a bit about the implementation of the `pcps_acquistion_sc` - it really pains me to have two classes with almost identical implementation, it makes for a real code maintenance headache. One possible solution would be to make the current implementation a generic `pcps_acquisition` block that would take an `item_type `parameter as input. From what I can see the current `pcps_acquisition_sc` simply converts the input from `lv_16sc_t` to `gr_complex` using a volk kernel and then implements precisely the same functionality as `pcps_acquisition_cc`. I'm hacking together a solution along these lines for my personal use, do you think this would be a good idea? Or is the separation of blocks into specific implementations for specific data types a key design decision within `gnss-sdr`?

carlesfernandez, 2017-09-20:
This is an interesting issue that we have been discussing for ages. Obviously the current solution is not the most elegant one, but there are reasons for it to be this way.

If the GNU Radio block takes the input type as a parameter, you need to instantiate resources for all implementations, with the consequent waste of unused memory buffers. Having conditional initializers in the constructor does not seem a good design choice, and in fact it is a practice not recommended by almost all C++ code guidelines. However, it can be done, see this question at Stack Overflow: https://stackoverflow.com/questions/1014518/right-way-to-conditionally-initialize-a-c-member-variable You possibly already did that, but I wonder if such a solution scales well. I’m curious about how you managed the different block's io_signature.

I remember a discussion about that in the GNU Radio mailing list, but now I can’t find it in a quick search. The conclusion was that this is an unsafe choice, and they opted for keeping block implementations for different data types in different files. Their solution is actually quite nice: using templates (e.g., https://github.com/gnuradio/gnuradio/blob/master/gr-blocks/lib/multiply_const_vXX_impl.cc.t ) and then using some CMake macros, as the ones defined here: https://github.com/gnuradio/gnuradio/blob/next/gr-utils/python/modtool/gr-newmod/cmake/Modules/GrMiscUtils.cmake to generate the actual _cc and _ss files. This avoids the ugly code repetition, but it still provides separate implementations.

I’m happy to spend time on things that already work if there is a good design reason for it :-) Would you agree that this “template solution” is the best way to go? This is an interesting design choice, longly debated within @Arribas and me, with no obvious solution.

odrisci, 2017-09-21:
The templating idea is interesting, I'll have a look at the gnuradio source to get a better handle on how it works.

As regards the general idea of managing the input item type, I think that we have 3 cases in `gnss-sdr`:

1. Blocks that convert streams of samples to streams of samples (mostly in the `SignalConditioner`)
2. Blocks that convert streams of samples to `GnssSynchro` objects (acquisition and tracking)
3. Blocks that process streams of `GnssSynchro` objects

The item type problem does not arise for the 3rd case. For the first case, this is essentially the same as the situation in most `gnuradio` applications, and I think the templating approach would work particularly well here.

The issue that I see is mostly around the 2nd case, and even here we have two sub cases:

2a. Acquisition
2b. Tracking

I think it is appropriate to differentiate between these cases as acquisition is an infrequent process (compared to tracking) and we are often willing to use the FFT approach here, even if it means that we lose true real-time capability. My approach for acquisition has been to provide a set of helper functions that call `volk` kernels to do the conversion within the acquisition block.

Pros:
* Acquisition blocks are usually inactive, therefore the conversion does not usually take place, saving computations
* Re-use of the core FFT block functionality whatever the input type

Cons:
* If more than one acquisition block is active then the conversion is performed multiple times

For the tracking blocks I think we have a very interesting case. Essentially most of the tracking blocks follow precisely the same sequence of steps:

* Generate local replicas
* Perform carrier wipe-off
* Perform correlation (multiple taps)
* Generate instantaneous error estimates based on the correlator outputs (using discriminators)
* Update state (code phase, carrier phase, Doppler etc)
* Pass state on to telemetry etc using the `GnssSynchro`

The first 3 stages are currently all implemented in the `multicorrelator` class. If we could abstract the multicorrelator interface and encapsulate it in a pure abstract base class, then the tracking blocks could be initialised with a shared pointer to a `multicorrelator` object.

Pros:
* Tracking block does not need to worry about the input item type, only the correlator dump item type
    * For simplicity I would suggest that all correlator outputs be fixed as `gr_complex`
* New correlation strategies can be implemented without modifying/reproducing the core tracking blocks, e.g.
    * 1,2,3,4, 8 bit correlators
    * Table look up for codes rather than on-the-fly resampling
    * Table look up for carrier,
    * Etc???
* No change to the channel interfaces, so all existing tracking blocks continue to work just fine

Cons:
* It ain't broke, so why fix it? :)

odrisci, 2017-09-21:
On handling the io signatures, I have a helper function that provides a `get_item_type_size( item_type)` interface. So we can do something like:

```c++
gr::io_signature::make(1, 1, get_item_type_size( item_type) * samples_per_ms );
```

At the moment its very rough and I'm not 100 % sure of it, but it does allow me to get super fast processing from a 16-bit front-end in real time.

I'll push these changes to my fork in the next few weeks, there's a lot of mess in there at the moment that I need to clean up first

carlesfernandez, 2018-02-07:
Hi, 

we have an update here: we have merged `pcps_acquisition_cc.*` and `pcps_acquisition_sc.*` into a single GNU Radio block `pcps_acquisition.*`. This new block accepts both `gr_complex` and `cshort` as input data types, so no more duplication of code here :-). The block just have a new parameter specifying the input data type. 

This change does not affect the API for this block, so no changes are required in the configuration. 

For the Tracking blocks, we have proposed an idea in our application to the Google Summer of Code Program for this year. Please check http://gnss-sdr.org/google-summer-code-2018-ideas-list/ for more details.

closed 2019-08-02 by carlesfernandez