gnss-sdr/gnss-sdr #77: front-end-cal is turning off internal bias-t of rtl-sdr.com v3

sv1, 2017-09-23:
Hello I am using an RTL-SDR.com V3  which has an internal bias-t for powering my active gps antenna. The gnss apps (front-end-call, gnss-sdr) upon startup, they disable the bias-t, causing no gps signals to be found.

carlesfernandez, 2017-10-17:
Hi,

thanks for pointing this out. At https://www.rtl-sdr.com/rtl-sdr-blog-v-3-dongles-user-guide/ I’ve found some code that activates the bias-t. I’m pasting it below for the records. Since the current version of osmosdr does not seem to implement this feature, we have some options here:
  1.  Maybe it makes sense to write a new signal source for rtl-sdr dongles, based on librtlsdr-dev (which installs /usr/include/rtl-sdr.h and the library). The rtlsdr API provides ```int rtlsdr_set_bias_tee(rtlsdr_dev_t *dev, int on);```, at least at its latest version.
  2. Ok, this is a long shot: Could you try to use the [RtlTcp_Signal_Source](http://gnss-sdr.org/docs/sp-blocks/signal-source/#implementation-rtltcp_signal_source) and call rtl_tcp with the ```-T``` flag? Example:

<pre>
$ rtl_tcp -a 127.0.0.1 -f 1575420000 -g 0 -s 2000000 <b>-T</b>
</pre>

### Just for the records
The code that actually implements the activation is:

<pre>
#include <libusb.h>
//...

enum sys_reg {
	DEMOD_CTL = 0x3000,
	<b>GPO = 0x3001,</b>
	GPI = 0x3002,
	<b>GPOE = 0x3003,</b>
	<b>GPD = 0x3004,</b>
	SYSINTE = 0x3005,
	SYSINTS = 0x3006,
	GP_CFG0 = 0x3007,
	GP_CFG1 = 0x3008,
	SYSINTE_1 = 0x3009,
	SYSINTS_1 = 0x300a,
	DEMOD_CTL_1 = 0x300b,
	IR_SUSPEND = 0x300c,
};

enum blocks {
	DEMODB = 0,
	USBB = 1,
	<b>SYSB = 2,</b>
	TUNB = 3,
	ROMB = 4,
	IRB = 5,
	IICB = 6,
};

int rtlsdr_set_bias_tee(rtlsdr_dev_t *dev, int on)
{
	if (!dev)
		return -1;

	rtlsdr_set_gpio_output(dev, 0);
	rtlsdr_set_gpio_bit(dev, 0, on);

	return 1;
}

void rtlsdr_set_gpio_output(rtlsdr_dev_t *dev, uint8_t gpio)
{
	int r;
	gpio = 1 << gpio;

	r = rtlsdr_read_reg(dev, SYSB, GPD, 1);
	rtlsdr_write_reg(dev, SYSB, GPD, r & ~gpio, 1);
	r = rtlsdr_read_reg(dev, SYSB, GPOE, 1);
	rtlsdr_write_reg(dev, SYSB, GPOE, r | gpio, 1);
}

void rtlsdr_set_gpio_bit(rtlsdr_dev_t *dev, uint8_t gpio, int val)
{
	uint16_t r;

	gpio = 1 << gpio;
	r = rtlsdr_read_reg(dev, SYSB, GPO, 1);
	r = val ? (r | gpio) : (r & ~gpio);
	rtlsdr_write_reg(dev, SYSB, GPO, r, 1);
}

uint16_t rtlsdr_read_reg(rtlsdr_dev_t *dev, uint8_t block, uint16_t addr, uint8_t len)
{
	int r;
	unsigned char data[2];
	uint16_t index = (block << 8);
	uint16_t reg;

	r = libusb_control_transfer(dev->devh, CTRL_IN, 0, addr, index, data, len, CTRL_TIMEOUT);

	if (r < 0)
		fprintf(stderr, "%s failed with %d\n", __FUNCTION__, r);

	reg = (data[1] << 8) | data[0];

	return reg;
}

int rtlsdr_write_reg(rtlsdr_dev_t *dev, uint8_t block, uint16_t addr, uint16_t val, uint8_t len)
{
	int r;
	unsigned char data[2];

	uint16_t index = (block << 8) | 0x10;

	if (len == 1)
		data[0] = val & 0xff;
	else
		data[0] = val >> 8;

	data[1] = val & 0xff;

	r = libusb_control_transfer(dev->devh, CTRL_OUT, 0, addr, index, data, len, CTRL_TIMEOUT);

	if (r < 0)
		fprintf(stderr, "%s failed with %d\n", __FUNCTION__, r);

	return r;
}
</pre>

sv1, 2017-10-17:
> -v-3-dongles-user-guide/ I’ve found some code that activates the
> $ rtl_tcp -a 127.0.0.1 -f 1575420000 -g 0 -s 2000000 -T
> 5493e4bb","name":"GitHub"},"entity":{"external_key":"github/gnss-
> sdr/gnss-sdr","title":"gnss-sdr/gnss-sdr","subtitle":"GitHub
> repository","main_image_url":"https://cloud.githubusercontent.com/ass
> ets/143418/17495839/a5054eac-5d88-11e6-95fc-
> 7290892c7bb5.png","avatar_image_url":"https://cloud.githubusercontent
> .com/assets/143418/15842166/7c72db34-2c0b-11e6-9aed-
> b52498112777.png","action":{"name":"Open in
> GitHub","url":"https://github.com/gnss-sdr/gnss-
> sdr"}},"updates":{"snippets":[{"icon":"PERSON","message":"@carlesfern
> andez in #77: Hi,\r\n\r\nthanks for pointing this out. At
> implement this feature, we have some options here:\r\n 1. Maybe it
> latest version.\r\n 2. Ok, this is a long shot: Could you try to use
> source/#implementation-rtltcp_signal_source) and call rtl_tcp with
> the ```-T``` flag? Example:\r\n\r\n\u003cpre\u003e\r\n$ rtl_tcp -a
> 127.0.0.1 -f 1575420000 -g 0 -s 2000000 \u003cb\u003e-
> T\u003c/b\u003e\r\n\u003c/pre\u003e\r\n\r\n\r\n### Just for the
> records\r\nThe code that actually implements the activation
> is:\r\n\r\n\u003cpre\u003e\r\n#include
> \u003clibusb.h\u003e\r\n//...\r\n\r\nenum sys_reg {\r\n\tDEMOD_CTL =
> 0x3000,\r\n\t\u003cb\u003eGPO = 0x3001,\u003c/b\u003e\r\n\tGPI =
> 0x3002,\r\n\t\u003cb\u003eGPOE =
> 0x3003,\u003c/b\u003e\r\n\t\u003cb\u003eGPD =
> 0x3004,\u003c/b\u003e\r\n\tSYSINTE = 0x3005,\r\n\tSYSINTS =
> [... 34 lines cut ...]
> CTRL_TIMEOUT);\r\n\r\n\tif (r \u003c 0)\r\n\t\tfprintf(stderr, \"%s
> failed with %d\\n\", __FUNCTION__, r);\r\n\r\n\treturn
> r;\r\n}\r\n\u003c/pre\u003e"}],"action":{"name":"View
> Issue","url":"https://github.com/gnss-sdr/gnss-
> sdr/issues/77#issuecomment-337322433"}}}

Hello Charles,

the actual code for turning on the internal bias-t is here:

https://github.com/rtlsdrblog/rtl_biast

which is the one I use to turn on/off the bias-t.

In this commit:

https://github.com/osmocom/gr-osmosdr/commit/3c7d3f166401b1af6afd903b8d
2b9729e8897189

on June 11th, the bias-t is implemented in osmosdr so it is a matter of
time to hit the linux distros.

BR
Apostolos

carlesfernandez, 2017-10-17:
Oh, that's awesome. So a simple addition to Osmosdr_Signal_Source would be enough. Thanks for the pointers!

carlesfernandez, 2017-10-17:
Looking at the code, it seems that using [Osmosdr_Signal_Source](http://gnss-sdr.org/docs/sp-blocks/signal-source/#implementation-osmosdr_signal_source) with ```SignalSource.osmosdr_args=bias=1``` would do the job.

OrbitTheSun, 2017-12-05:
Whether the bias-tee is turned off before reception depends on which version of gr-osmosdr was used when compiling GNSS-SDR.

With an older version (for example, v0.1.4.8), the utility [rtl_bias](https://github.com/OrbitTheSun/rtl_biast) may be used to switch the bias-tee. Front-end-cal can then be called with the bias-tee turned on. Same applies to gnss-sdr. After reception the bias-tee is switched off automatically by the programs.

However with a newer version of [gr-osmosdr](https://github.com/osmocom/gr-osmosdr), i.e. the current v0.1.4-98, the bias-tee is turned off before reception, unless the `SignalSource.osmosdr_args` has been given the correct parameter. It took me a while to figure out what to do. Add the following line to the configuration files:

    SignalSource.osmosdr_args=rtl,bias=1

Without 'rtl,', the bias-tee will be turned off!

carlesfernandez, 2017-12-05:
Awesome! Thanks for sharing.

carlesfernandez, 2017-12-06:
Added to the documentation at http://gnss-sdr.org/docs/sp-blocks/signal-source/#implementation-osmosdr_signal_source

OrbitTheSun, 2017-12-06:
Thank you, Carles! - Perhaps you may add the line with a short note to the file `gnss-sdr_GPS_L1_rtlsdr_realtime.conf` and to `front-end-cal.conf` commented out.

carlesfernandez, 2018-01-18:
Done!

closed 2018-12-14 by carlesfernandez