gnss-sdr/gnss-sdr #79: GNSS-SDR build error

Peterthegreat, 2017-10-05:
Hello,
I am trying to build the source code, since the previous install with apt-get didn't had all of the required functions. I made it compile until 96%, but now I am stuck.[ Here is the log](https://pastebin.com/BYH6wv2R)

Any tips?
Thanks

carlesfernandez, 2017-10-05:
From the logs, it seems you are building the software with clang and libc++ but linking against libraries built by GCC and using libstdc++. Those two libraries have an incompatible ABI, so it will not compile.

Assuming that you are under Linux, you can either use GCC, or use Clang without forcing the usage of libc++. This is handled automatically by CMake in the 'next' branch.

Hope this helps,
Carles

Peterthegreat, 2017-10-05:
Thank you for the fast reply! Building appears successful.

closed 2017-10-05 by Peterthegreat