gnss-sdr/gnss-sdr #98: how to use antenna

robellebor, 2017-11-24:
i have installed and tried GNSS SDR following the steps in "http://gnss-sdr.org/my-first-fix/" and it works ....

but i want to use antenna instead of the raw file ... so how can configure the conf file to connect with my antenna ???

carlesfernandez, 2017-11-24:
You will need to use a SignalSource for a RF front-end. They are documented here: http://gnss-sdr.org/docs/sp-blocks/signal-source/#radio-frequency-front-ends 

A working example with a USRP is at http://gnss-sdr.org/conf/

robellebor, 2017-11-29:
thanks
but today i updated my OS to ubuntu 17.10 and am trying to install gnss-sdr again , but after i did ... when i execute "gnss-sdr --version" i get:

gnss-sdr: error while loading shared libraries: libboost_system.so.1.58.0: cannot open shared object file: No such file or directory

what is happening ??

carlesfernandez, 2018-01-18:
Did you uninstall the previous version and rebuild gnss-sdr from the scratch?

baguzajja, 2018-01-30:
how about this... 
linux; GNU C++ version 7.2.0; Boost_106200; UHD_3.11.0.git-774-g8f5f38e4

Exception in static block reg_basic_and_lf_dboards
  LookupError: KeyError: The dboard id 0x0000 is already registered to Basic TX.
Exception in static block reg_rfx_dboards
  LookupError: KeyError: The dboard id pair [0x0024, 0x0028] is already registered to RFX400.
Exception in static block reg_xcvr2450_dboard
  LookupError: KeyError: The dboard id pair [0x0061, 0x0060] is already registered to XCVR2450.
Exception in static block reg_sbx_dboards
  LookupError: KeyError: The dboard id pair [0x0054, 0x0055] is already registered to SBX.
Exception in static block reg_ubx_dboards
  LookupError: KeyError: The dboard id pair [0x0074, 0x0073] is already registered to UBX v0.3.
Exception in static block reg_wbx_simple_dboards
  LookupError: KeyError: The dboard id pair [0x0053, 0x0052] is already registered to WBX.
Exception in static block reg_dbsrx_dboard
  LookupError: KeyError: The dboard id 0x000d is already registered to DBSRX.
Exception in static block reg_unknown_dboards
  LookupError: KeyError: The dboard id 0xfff0 is already registered to Unknown TX.
Exception in static block reg_tvrx_dboard
  LookupError: KeyError: The dboard id 0x0040 is already registered to TVRX.
Exception in static block reg_dbsrx2_dboard
  LookupError: KeyError: The dboard id 0x0012 is already registered to DBSRX2.
Exception in static block reg_tvrx2_dboard
  LookupError: KeyError: The dboard id 0x0046 is already registered to TVRX2.
Exception in static block reg_twinrx_dboards
  LookupError: KeyError: The dboard id 0x0091 is already registered to TwinRX v1.0.
gnss-sdr version 0.0.9.git-next-b6326f5b
Segmentation fault

what happent

closed 2019-08-02 by carlesfernandez