gnss-sdr/gnss-sdr #108: gnss-sdr-next wont compile....get the following error

herbsims, 2018-01-10:
Building py27-pyqt5 I get the following error:

--->  Configuring py27-pyqt5
Error: Failed to configure py27-pyqt5: configure failure: command execution failed
Error: See /opt/local/var/macports/logs/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_python_py-pyqt5/py27-pyqt5/main.log for details.
Error: Follow https://guide.macports.org/#project.tickets to report a bug.
Error: Processing of port gnss-sdr-next failed

What do I need to do to fix this?????

carlesfernandez, 2018-01-19:
Hi, 

you will need to:
```
$ sudo port selfupdate
$ sudo port uninstall volk-gnss-sdr gnss-sdr
```
(or their -devel counterparts)
```
$ sudo port deactivate gnuradio gr-fcdproplus gr-osmosdr qwt52 py27-pyqwt
```
and 
```
$ sudo port install qwt61 +qt5
```
Then, ```sudo port install gnss-sdr-next``` works for me (macOS 10.13.2). However, take into account that the gr-osmosdr port does not work with gnuradio-next, so the OsmoSDR-based signal sources won’t be available. The best solution there is to use Macports to install gnuradio-next and then building and installing gr-osmosdr and gnss-sdr from source.

carlesfernandez, 2018-01-25:
Good news: the `gr-osmosdr` port has been updated and now it can be built against `gnuradio-next`. 

The `+osmosdr` variant has been included by default when installing the `gnss-sdr-next` port, so the [Osmosdr_Signal_Source](http://gnss-sdr.org/docs/sp-blocks/signal-source/#implementation-osmosdr_signal_source) implementation will be available.

Please do not forget to update your port tree:
```
$ sudo port selfupdate
```

closed 2018-12-14 by carlesfernandez