gnss-sdr/gnss-sdr #109: Errors when stopping gnss-sdr.

mark90054mark, 2018-01-12:
I'm running GNSS-SDR with Ubuntu 16.04.3 LTS (32-bit). When I press q-enter to stop gnss-sdr, I always get the following error message: 

> lctsai@lctsai:~/bladerf_test$ gnss-sdr --config_file=bladerf_false.conf
linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.010.002.000-release

Initializing GNSS-SDR v0.0.9.git-next-6256f15 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
OsmoSdr arguments: bladerf
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11.1
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy redpitaya 
Opening nuand bladeRF with device identifier string: ""
 Serial # 68a0...dae2
 FW v2.0.0 FPGA v0.7.1
Actual RX Rate: 2000000.000000 [SPS]...
Actual RX Freq: 1575420000.000000 [Hz]...
PLL Frequency tune error 0.000000 [Hz]...Actual RX Gain: 6.000000 dB...
gr::buffer::allocate_buffer: warning: tried to allocate
   496 items of size 132. Due to alignment requirements
   1024 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   496 items of size 132. Due to alignment requirements
   1024 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   496 items of size 132. Due to alignment requirements
   1024 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   496 items of size 132. Due to alignment requirements
   1024 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   496 items of size 132. Due to alignment requirements
   1024 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   496 items of size 132. Due to alignment requirements
   1024 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   496 items of size 132. Due to alignment requirements
   1024 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   496 items of size 132. Due to alignment requirements
   1024 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   496 items of size 132. Due to alignment requirements
   1024 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   496 items of size 132. Due to alignment requirements
   1024 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   496 items of size 132. Due to alignment requirements
   1024 were allocated.  If this isn't OK, consider padding
[... 613 lines cut ...]
b732a000-b7331000 r--p 000d3000 08:01 8920514    /usr/local/lib/libgnuradio-filter-3.7.11.1.so.0.0.0
b7331000-b7332000 rw-p 000da000 08:01 8920514    /usr/local/lib/libgnuradio-filter-3.7.11.1.so.0.0.0
b7332000-b7375000 r-xp 00000000 08:01 8920850    /usr/local/lib/libgnuradio-uhd-3.7.11.1.so.0.0.0
b7375000-b7376000 ---p 00043000 08:01 8920850    /usr/local/lib/libgnuradio-uhd-3.7.11.1.so.0.0.0
b7376000-b7379000 r--p 00043000 08:01 8920850    /usr/local/lib/libgnuradio-uhd-3.7.11.1.so.0.0.0
b7379000-b737a000 rw-p 00046000 08:01 8920850    /usr/local/lib/libgnuradio-uhd-3.7.11.1.so.0.0.0
b737a000-b7d90000 r-xp 00000000 08:01 8531749    /usr/lib/i386-linux-gnu/libuhd.so.003.010
b7d90000-b7d91000 ---p 00a16000 08:01 8531749    /usr/lib/i386-linux-gnu/libuhd.so.003.010
b7d91000-b7daa000 r--p 00a16000 08:01 8531749    /usr/lib/i386-linux-gnu/libuhd.so.003.010
b7daa000-b7dac000 rw-p 00a2f000 08:01 8531749    /usr/lib/i386-linux-gnu/libuhd.so.003.010
b7dac000-b7db0000 rw-p 00000000 00:00 0 
b7db0000-b7e9d000 r-xp 00000000 08:01 8535776    /usr/lib/i386-linux-gnu/libgnuradio-osmosdr.so.0.1.4
b7e9d000-b7ea1000 r--p 000ec000 08:01 8535776    /usr/lib/i386-linux-gnu/libgnuradio-osmosdr.so.0.1.4
b7ea1000-b7ea2000 rw-p 000f0000 08:01 8535776    /usr/lib/i386-linux-gnu/libgnuradio-osmosdr.so.0.1.4
b7ea2000-b7ea3000 rw-p 00000000 00:00 0 
b7ea3000-b7eac000 r-xp 00000000 08:01 8522347    /usr/lib/i386-linux-gnu/libgnutls-openssl.so.27.0.2
b7eac000-b7ead000 r--p 00008000 08:01 8522347    /usr/lib/i386-linux-gnu/libgnutls-openssl.so.27.0.2
b7ead000-b7eae000 rw-p 00009000 08:01 8522347    /usr/lib/i386-linux-gnu/libgnutls-openssl.so.27.0.2
b7eae000-b7eb7000 r-xp 00000000 08:01 8535790    /usr/lib/libarmadillo.so.6.500.5
b7eb7000-b7eb8000 ---p 00009000 08:01 8535790    /usr/lib/libarmadillo.so.6.500.5
b7eb8000-b7eb9000 r--p 00009000 08:01 8535790    /usr/lib/libarmadillo.so.6.500.5
b7eb9000-b7eba000 rw-p 0000a000 08:01 8535790    /usr/lib/libarmadillo.so.6.500.5 terminated (core dumped)
lctsai@lctsai:~/bladerf_test$ 

How can I fix this problem?

carlesfernandez, 2018-01-23:
Thanks for your report.

It seems that a destructor is not working well. Could you share the config
file you are using, so we can narrow down the search?

Best regards,
Carles

mark90054mark, 2018-02-08:
Dear Carles,
Thanks for your reply. I solved the problem later following the instruction "Install dependencies using PyBOMBS".
My config file is attached below,
[bladerf_github_mark.txt](https://github.com/gnss-sdr/gnss-sdr/files/1705934/bladerf_github_mark.txt)

closed 2018-04-03 by mark90054mark