gnss-sdr/gnss-sdr #117: next won't compile - missing lib4cpp

nebukadnezar, 2018-01-31:
inside freshly cloned git repo:

`git checkout next && cd build && cmake ../`

and this happens:

```
-- The CXX compiler identification is GNU 5.4.0
-- The C compiler identification is GNU 5.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Configuring GNSS-SDR v0.0.9.git-next-b6326f5 to be built on Ubuntu GNU/Linux Release 16.04 (64 bits)
-- Build type not specified: defaulting to Release.
-- Checking for ARM
-- System is not ARM
-- Testing for SIMD extensions
-- Detected SSE4A
-- Detected SSE4.2
-- Detected SSE4.1
-- Detected SSSE3
-- Detected SSE3
-- Detected SSE2
-- Detected SSE1
-- Detected MMX
-- Detected AES
-- Detected AVX
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Googletest (libgtest-dev package) has been found.
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   date_time
--   system
--   filesystem
--   thread
--   serialization
--   chrono
--   unit_test_framework
--   program_options
--   atomic
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for GNU Radio Module: RUNTIME
-- Checking for module 'gnuradio-runtime'
--   Found gnuradio-runtime, version 3.7.11
--  * INCLUDES=/usr/local/include
--  * LIBS=/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so
-- Found GNURADIO_RUNTIME: /usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so  
-- GNURADIO_RUNTIME_FOUND = TRUE
-- Checking for GNU Radio Module: ANALOG
-- Checking for module 'gnuradio-analog'
--   Found gnuradio-analog, version 3.7.11
--  * INCLUDES=/usr/local/include
--  * LIBS=/usr/local/lib/libgnuradio-analog.so;/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so;/usr/local/lib/libvolk.so
-- Found GNURADIO_ANALOG: /usr/local/lib/libgnuradio-analog.so;/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so;/usr/local/lib/libvolk.so  
-- GNURADIO_ANALOG_FOUND = TRUE
-- Checking for GNU Radio Module: BLOCKS
-- Checking for module 'gnuradio-blocks'
--   Found gnuradio-blocks, version 3.7.11
--  * INCLUDES=/usr/local/include
--  * LIBS=/usr/local/lib/libgnuradio-blocks.so;/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so
-- Found GNURADIO_BLOCKS: /usr/local/lib/libgnuradio-blocks.so;/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so  
-- GNURADIO_BLOCKS_FOUND = TRUE
-- Checking for GNU Radio Module: FFT
-- Checking for module 'gnuradio-fft'
--   Found gnuradio-fft, version 3.7.11
--  * INCLUDES=/usr/local/include
--  * LIBS=/usr/local/lib/libgnuradio-fft.so;/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so
-- Found GNURADIO_FFT: /usr/local/lib/libgnuradio-fft.so;/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so  
-- GNURADIO_FFT_FOUND = TRUE
-- Checking for GNU Radio Module: FILTER
-- Checking for module 'gnuradio-filter'
--   Found gnuradio-filter, version 3.7.11
--  * INCLUDES=/usr/local/include
--  * LIBS=/usr/local/lib/libgnuradio-filter.so;/usr/local/lib/libgnuradio-fft.so;/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so
-- Found GNURADIO_FILTER: /usr/local/lib/libgnuradio-filter.so;/usr/local/lib/libgnuradio-fft.so;/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so  
-- GNURADIO_FILTER_FOUND = TRUE
-- Checking for GNU Radio Module: PMT
-- Checking for module 'gnuradio-runtime'
--   Found gnuradio-runtime, version 3.7.11
--  * INCLUDES=/usr/local/include
--  * LIBS=/usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so
-- Found GNURADIO_PMT: /usr/local/lib/libgnuradio-runtime.so;/usr/local/lib/libgnuradio-pmt.so  
-- GNURADIO_PMT_FOUND = TRUE
-- Checking for module 'volk'
--   Found volk, version 1.3
-- Found VOLK: /usr/local/lib/libvolk.so  
CMake Error at CMakeLists.txt:550 (message):
  *** Lib4cpp is required to build gnss-sdr
```

Lib4cpp doesn't seem to be a thing... google returning next to nothing. Any clues anyone?

nebukadnezar, 2018-01-31:
great, that fixed it thanks! It's now compiling, and I can run the GPS L1 UHD sample. Provides the expected output and position fix. Any other systems however don't work, Galileo E1 keeps losing lock, and GLONASS dies with an error message (more on that later - let's get GPS working first).

The RTL-SDR example config is not producing anything at all. It just sits on a black screen. The RF input is connected to the exact same signal source as the UHD via the same splitter, and the RTL-SDR dongle (a new v3 from RTL-SDR.com) is working very well with other SDR decoders like ADS-B and AIS. Tried varying the gains, but no luck. Any suggestions on what else to try?

agent00pi, 2018-07-23:
Did you enable RTL-SDR support at build time? You would have had to call `DENABLE_OSMOSDR=ON` with your make flags.

It's set to on by default in the version that is in the repositories, but you have to manually enable it if you are building it yourself.

closed 2018-12-14 by carlesfernandez