gnss-sdr/gnss-sdr #120: My first position fix

kasmf, 2018-02-05:
what is My first position fix?

carlesfernandez, 2018-02-05:
It is a [page](http://gnss-sdr.org/my-first-fix/) describing the first steps using GNSS-SDR.

kasmf, 2018-02-06:
how to correct the error, when the command in a terminal is like $ gnss-sdr --config_file=./my-first-GNSS-SDR-receiver.conf the error it comes like bellow:
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

Initializing GNSS-SDR v0.0.6.git-- ... Please wait.
Logging will be done at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
Processing file /home/kas/GNSS-raw-signal/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat, which contains 1600000000 [bytes]
GNSS signal recorded time to be processed: 99.999 [s]
Using Volk machine: avx2_64_mmx_orc
E0206 09:15:36.066758  3136 gnss_flowgraph.cc:355] itemsize mismatch: gps_l1_ca_pvt_cc0:0 using 8, null_sink0:0 using 2
E0206 09:15:36.066821  3136 control_thread.cc:139] Unable to connect flowgraph
Total GNSS-SDR run time 0.001682 [seconds]
GNSS-SDR program ended.
gnss-sdr: /usr/include/boost/thread/pthread/condition_variable_fwd.hpp:81: boost::condition_variable::~condition_variable(): Assertion `!ret' failed.
Aborted (core dumped)

carlesfernandez, 2018-02-06:
You are using an old version of GNSS-SDR (v0.0.6 according to your log). The latest stable release is 0.0.9, and the `next` branch of the repository contains a number of bug fixes and the addition of new features.

Please build the software from source, as explained [here](http://gnss-sdr.org/build-and-install/).

kasmf, 2018-02-06:
i remove the gnss-sdr using $ sudo apt-get remove gnss-sdr and install using $ sudo apt-get install gnss-sdr to get the latest gnss-sdr version (gnss-sdr version 0.0.9). still the gnss-sdr version is gnss-sdr version 0.0.6,... how to remove the old gnss-sdr version and install the latest version (gnss-sdr version 0.0.9)?

carlesfernandez, 2018-02-06:
Please do this:

 * Remove the package:
```
$ sudo apt-get remove gnss-sdr
```

* Add dependencies not required in previous versions:
```
$ sudo apt-get install libhdf5-dev libmatio-dev
```
 * Build GNSS-SDR from source:

```
$ git clone https://github.com/gnss-sdr/gnss-sdr.git
$ cd gnss-sdr/build
$ git checkout next
$ cmake ..
$ make -j4
$ sudo make install
```

acebrianjuan, 2018-02-06:
Hi @kasmf 

The version of GNSS-SDR that is available in the Ubuntu repositories was frozen at v0.0.6, which is **not** the latest version. Therefore, installing GNSS-SDR using  `sudo apt install gnss-sdr` will **not** upgrade you to the latest version (v0.0.9), because it is not available yet to the package manager.

You can check which version of a package is available to the package manager with the command: 

    $ apt policy <package>

In the case of GNSS-SDR, you will see something like this:
```
$ apt policy gnss-sdr 
gnss-sdr:
  Installed: (none)
  Candidate: 0.0.6-1
  Version table:
     0.0.6-1 500
        500 http://es.archive.ubuntu.com/ubuntu xenial/universe amd64 Packages
```
Suggested read: [Why don't the Ubuntu repositories have the latest versions of software?](https://askubuntu.com/q/151283)

If you want the latest version you will have to build GNSS-SDR from source following [the instructions](https://github.com/gnss-sdr/gnss-sdr/issues/120#issuecomment-363437732) provided by @carlesfernandez 

On a side note, you are welcome to ask questions like this one in the GNSS-SDR [mailing list](https://lists.sourceforge.net/lists/listinfo/gnss-sdr-developers).

kasmf, 2018-02-07:
i follow the instructions provided  by @carlesfernandez, but  it comes the error on the terminal as:
kas@kas-HP-ProDesk-490-G3-MT-Business-PC:~/gnss-sdr/build$ make -j4
make: *** No targets specified and no makefile found.  Stop.
how to correct please?

carlesfernandez, 2018-02-07:
Did you get any error in the configuration step (that is, `cmake ..`)?

If cmake complained, could you post here the message?

kasmf, 2018-02-07:
no error in the configuration step of   cmake .., but the error is in the configuration step of make -j4

kasmf, 2018-02-07:
sorry the error is in the configuration step of cmake ..              as:
kas@kas-HP-ProDesk-490-G3-MT-Business-PC:~/gnss-sdr/build$ cmake ..
-- Configuring GNSS-SDR v0.0.9.git-next-413ab8c to be built on Ubuntu GNU/Linux Release 16.04 (64 bits)
-- Build type not specified: defaulting to Release.
-- Checking for ARM
-- System is not ARM
-- Testing for SIMD extensions
--  Googletest has not been found.
--  Googletest will be downloaded and built automatically 
--  when doing 'make'. 
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   date_time
--   system
--   filesystem
--   thread
--   serialization
--   chrono
--   unit_test_framework
--   program_options
--   atomic
-- Checking for GNU Radio Module: RUNTIME
-- Checking for module 'gnuradio-runtime'
--   No package 'gnuradio-runtime' found
--  * INCLUDES=GNURADIO_RUNTIME_INCLUDE_DIRS-NOTFOUND
--  * LIBS=GNURADIO_RUNTIME_LIBRARIES_gnuradio-runtime-NOTFOUND
-- Could NOT find GNURADIO_RUNTIME (missing:  GNURADIO_RUNTIME_LIBRARIES GNURADIO_RUNTIME_INCLUDE_DIRS) 
-- GNURADIO_RUNTIME_FOUND = FALSE
-- Required GNU Radio Component: RUNTIME missing!
-- Checking for GNU Radio Module: ANALOG
-- Checking for module 'gnuradio-analog'
--   No package 'gnuradio-analog' found
--  * INCLUDES=GNURADIO_ANALOG_INCLUDE_DIRS-NOTFOUND
--  * LIBS=GNURADIO_ANALOG_LIBRARIES_gnuradio-analog-NOTFOUND
-- Could NOT find GNURADIO_ANALOG (missing:  GNURADIO_ANALOG_LIBRARIES GNURADIO_ANALOG_INCLUDE_DIRS) 
-- GNURADIO_ANALOG_FOUND = FALSE
-- Required GNU Radio Component: ANALOG missing!
-- Checking for GNU Radio Module: BLOCKS
-- Checking for module 'gnuradio-blocks'
--   No package 'gnuradio-blocks' found
--  * INCLUDES=GNURADIO_BLOCKS_INCLUDE_DIRS-NOTFOUND
--  * LIBS=GNURADIO_BLOCKS_LIBRARIES_gnuradio-blocks-NOTFOUND
-- Could NOT find GNURADIO_BLOCKS (missing:  GNURADIO_BLOCKS_LIBRARIES GNURADIO_BLOCKS_INCLUDE_DIRS) 
-- GNURADIO_BLOCKS_FOUND = FALSE
-- Required GNU Radio Component: BLOCKS missing!
-- Checking for GNU Radio Module: FFT
-- Checking for module 'gnuradio-fft'
--   No package 'gnuradio-fft' found
--  * INCLUDES=GNURADIO_FFT_INCLUDE_DIRS-NOTFOUND
--  * LIBS=GNURADIO_FFT_LIBRARIES_gnuradio-fft-NOTFOUND
-- Could NOT find GNURADIO_FFT (missing:  GNURADIO_FFT_LIBRARIES GNURADIO_FFT_INCLUDE_DIRS) 
-- GNURADIO_FFT_FOUND = FALSE
-- Required GNU Radio Component: FFT missing!
-- Checking for GNU Radio Module: FILTER
-- Checking for module 'gnuradio-filter'
--   No package 'gnuradio-filter' found
--  * INCLUDES=GNURADIO_FILTER_INCLUDE_DIRS-NOTFOUND
--  * LIBS=GNURADIO_FILTER_LIBRARIES_gnuradio-filter-NOTFOUND
-- Could NOT find GNURADIO_FILTER (missing:  GNURADIO_FILTER_LIBRARIES GNURADIO_FILTER_INCLUDE_DIRS) 
-- GNURADIO_FILTER_FOUND = FALSE
-- Required GNU Radio Component: FILTER missing!
-- Checking for GNU Radio Module: PMT
-- Checking for module 'gnuradio-runtime'
--   No package 'gnuradio-runtime' found
--  * INCLUDES=GNURADIO_PMT_INCLUDE_DIRS-NOTFOUND
--  * LIBS=GNURADIO_PMT_LIBRARIES_gnuradio-pmt-NOTFOUND
-- Could NOT find GNURADIO_PMT (missing:  GNURADIO_PMT_LIBRARIES GNURADIO_PMT_INCLUDE_DIRS) 
-- GNURADIO_PMT_FOUND = FALSE
-- Required GNU Radio Component: PMT missing!
-- CMake cannot find GNU Radio >= 3.7.3
Go to https://github.com/gnuradio/pybombs
and follow the instructions to install GNU Radio in your system.
CMake Error at CMakeLists.txt:505 (message):
  GNU Radio 3.7.3 or later is required to build gnss-sdr

-- Configuring incomplete, errors occurred!
See also "/home/kas/gnss-sdr/build/CMakeFiles/CMakeOutput.log".
See also "/home/kas/gnss-sdr/build/CMakeFiles/CMakeError.log".

acebrianjuan, 2018-02-07:
As far as I know, that error message indicates that some dependencies are missing ([source](https://askubuntu.com/q/103348)).

Did you install all [the dependencies](url) that are required by GNSS-SDR?

Try following these steps:

1. Install all the dependencies:

```
$ sudo apt-get install build-essential cmake git libboost-dev \
   libboost-date-time-dev libboost-system-dev libboost-filesystem-dev \
   libboost-thread-dev libboost-chrono-dev libboost-serialization-dev \
   libboost-program-options-dev libboost-test-dev liblog4cpp5-dev \
   libuhd-dev gnuradio-dev gr-osmosdr libblas-dev liblapack-dev \
   libarmadillo-dev libgflags-dev libgoogle-glog-dev libhdf5-dev \
   libgnutls-openssl-dev libmatio-dev python-mako python-six \
   googletest
```

2. Run `$ ./configure` inside the` gnss-sdr/build` directory

3. Then continue with the installation

```
$ make -j4
$ sudo make install
```

acebrianjuan, 2018-02-07:
I have updated the instructions on my previous comment. The `$ ./configure` script had nothing to do in this scenario.

carlesfernandez, 2018-02-07:
The important info from the CMake message is:

```
-- CMake cannot find GNU Radio >= 3.7.3
Go to https://github.com/gnuradio/pybombs
and follow the instructions to install GNU Radio in your system.
CMake Error at CMakeLists.txt:505 (message):
GNU Radio 3.7.3 or later is required to build gnss-sdr
```

You need to install a version of GNU Radio equal or greater than 3.7.3

According to https://packages.ubuntu.com/search?keywords=gnuradio, the package that comes with Ubuntu 16.04 is version 3.7.9, so you can use the package for installing it. Please follow the steps described by @acebrianjuan

kasmf, 2018-02-07:
i follow the instruction of acebrianjuan, but it comes the error as:
E: Unable to locate package googletest
how to set the error please?

acebrianjuan, 2018-02-07:
This error means that the package manager is unable to find a package called `googletest` in the official repositories. Never mind, if you pay attention to the [cmake log](https://github.com/gnss-sdr/gnss-sdr/issues/120#issuecomment-363752949) that you posted previously, `make` will take care of that:

```
-- Googletest has not been found.
-- Googletest will be downloaded and built automatically
-- when doing 'make'.
```

So proceed with the building process ([step #2](https://github.com/gnss-sdr/gnss-sdr/issues/120#issuecomment-363753613)) and let us know about your progress.

carlesfernandez, 2018-02-07:
The package `googletest` was named `libgtest-dev` in Ubuntu 16.04. Anyway, CMake will automatically download the source code and build it for you if it does not find that package in your system. So please proceed with doing `make` and let us know.

kasmf, 2018-02-08:
thank you very much for all of you help me ...  i follow the acebrianjuan instructions, it is work fine.

vasu471, 2018-02-08:
2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat
can some body provide the IF intermediate frequency and Sampling frequency for the above data

acebrianjuan, 2018-02-08:
Hi @vasu471 

According to the information available in the [GNSS-SDR website](http://gnss-sdr.org/my-first-fix/#step-2-download-a-file-of-raw-signal-samples), the file `2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat`

> contains 100 seconds of raw GNSS signal samples collected by a RF front-end centered at 1,575.42 MHz, that was delivering baseband samples at 4 MS/s, in an interleaved I&Q 16-bit integer format.

kasmf, 2018-02-09:
is it possible position fix with gnss-sdr by receiving the real time of gps signal ? that means by configuring 
the rf front end instead of the downloaded a file of raw signal samples. if possible, how we follow the instructions  ?

acebrianjuan, 2018-02-09:
Glad to help you @kasmf 

The installation instructions were taken from [gnss-sdr.org](http://gnss-sdr.org/), so credit should go [its contributors](https://github.com/gnss-sdr/geniuss-place/graphs/contributors). All the inner workings of GNSS-SDR are documented there!

> is it possible position fix with gnss-sdr by receiving the real time of gps signal ?

Yes!

> if possible, how we follow the instructions ?

A very popular and inexpensive front-end to get started with is the RTL2832U. You will need an active antenna too. Instructions can be found at: [GNSS-SDR operation with a Realtek RTL2832U USB dongle DVB-T receiver](http://gnss-sdr.org/docs/tutorials/gnss-sdr-operation-realtek-rtl2832u-usb-dongle-dvb-t-receiver/)

kasmf, 2018-02-09:
thank you acebrianjuan !

ssciancalepore, 2018-12-01:
Dear all,

I have just installed the last version of the gnss-sdr package using pybombs, and the version is gnss-sdr version _0.0.9.git-next-6dd47336._

I also downloaded the real traces of the 2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat experiment, and ran the gnss-sdr tool with the command:

`gnss-sdr --config-file=./conf/gnss-sdr.conf --log_dir=./logs/`

Note that, in the config file, I have just modified the path of the source file to indicate the path of the local copy of the ".dat" file.

However, the tool does not generate any kml output, but only a log file with many messages, showing acquisition and immediate loss of lock to the satellites.

I have tried different combinations of the `GNSS-SDR.internal_fs_sps` and the `SignalSource.sampling_frequency parameters` (Default: both 4000000), but nothing changes.

Has anyone already solved this issue?

Thanks a lot for your suggestions

carlesfernandez, 2019-07-03:
Please update your repo and feel free to reopen the issue if you still find problems.

closed 2019-07-03 by carlesfernandez

wamlord, 2020-03-07:
> Please do this:
>
> * Remove the package:
>
> ```
> $ sudo apt-get remove gnss-sdr
> ```
>
> * Add dependencies not required in previous versions:
>
> ```
> $ sudo apt-get install libhdf5-dev libmatio-dev
> ```
>
> * [...]
"

When I get "make -j4" i get
make: *** No targets specified and no makefile found.  Stop.
whats's the deal?

acebrianjuan, 2020-03-08:
Hi @wamlord,

This same error was addressed in this issue.
Please read from [this comment](https://github.com/gnss-sdr/gnss-sdr/issues/120#issuecomment-363707768) onwards.

Hopefully this will be enough to solve your problem.

Regards,
Álvaro

wamlord, 2020-03-08:
> Hi @wamlord,
> 
> This same error was addressed in this issue.
> Please read from [this comment](https://github.com/gnss-sdr/gnss-sdr/issues/120#issuecomment-363707768) onwards.
> 
> Hopefully this will be enough to solve your problem.
> 
> Regards,
> Álvaro

[image]
ok now what?

pyjhky1315, 2024-09-19:
> It is a [page](http://gnss-sdr.org/my-first-fix/) describing the first steps using GNSS-SDR.

I'm sorry to bother you, could you help me understand why this is happening？

eugene@ubuntu:~/gnss-sdr/conf$ sudo gnss-sdr --config_file=./hackrf.conf
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Initializing GNSS-SDR v0.0.19.git-main-87fcfd237 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
OsmoSdr arguments: hackrf,bias=1
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
Using HackRF One with firmware local-79baef7
Enabled antenna bias voltage
Actual RX Rate: 2e+06 [SPS]...
Actual RX Freq: 1.57542e+09 [Hz]...
PLL Frequency tune error: 0 [Hz]...
Actual RX Gain: 14 dB...
Actual Bandwidth: 6e+06 [Hz]...
RF Channels: 1
Starting a TCP/IP server of RTCM messages on port 2101
The TCP/IP server of RTCM messages is up and running. Accepting connections ...
E0919 05:52:24.634222 118200 gnss_flowgraph.cc:1140] Can't connect signal conditioner 1 to channel 10: vector::_M_range_check: __n (which is 1) >= this->size() (which is 1)
Unable to connect flowgraph
Stopping TCP/IP server on port 2101
Total GNSS-SDR run time: 0.528808 [seconds]
GNSS-SDR program ended.
eugene@ubuntu:~/gnss-sdr/conf$

and it is  the conf

[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=8000000
GNSS-SDR.Beidou_banned_prns=56,57,58

SignalSource.implementation=Osmosdr_Signal_Source
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=2000000
SignalSource.freq=1575420000
SignalSource.gain=40
SignalSource.rf_gain=40
SignalSource.if_gain=30
SignalSource.AGC_enabled=false
SignalSource.samples=0
SignalSource.repeat=false
;# Next line enables the internal HackRF One bias (3.3 VDC)
SignalSource.osmosdr_args=hackrf,bias=1
SignalSource.enable_throttle_control=false
SignalSource.dump=false
;######### SIGNAL_SOURCE CONFIG ############

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner0.implementation=Signal_Conditioner
;SignalConditioner1.implementation=Signal_Conditioner
;SignalConditioner2.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter0.implementation=Pass_Through
DataTypeAdapter1.implementation=Pass_Through
DataTypeAdapter2.implementation=Pass_Through

;######### INPUT_FILTER CONFIG ############
InputFilter0.implementation=Freq_Xlating_Fir_Filter
InputFilter0.decimation_factor=7
InputFilter0.input_item_type=gr_complex
InputFilter0.output_item_type=gr_complex
InputFilter0.taps_item_type=float
[... 196 lines cut ...]
PVT.rtcm_MT1019_rate_ms=5000
PVT.rtcm_MT1077_rate_ms=1000
PVT.rinex_version=2
PVT.flag_nmea_tty_port=true
PVT.nmea_dump_devname=/dev/pts/3

; To use gpsd with GNSS-SDR
; 1. run socat -d -d pty,raw,echo=0 pty,raw,echo=0
; 2. Set PVT.nmea_dump_devname to first PTY from socat output
; 3. run gpsd -b -n -N /dev/pts/4
; where /dev/pts/4 is the second PTY from socat output
; 4. run some gpsd client (xgps or other)

PVT.enable_monitor=true
PVT.monitor_client_addresses=127.0.0.1
PVT.monitor_udp_port=1111

Monitor.enable_monitor=true
Monitor.decimation_factor=4
Monitor.client_addresses=127.0.0.1
Monitor.udp_port=1112

;AcquisitionMonitor.enable_monitor=true
AcquisitionMonitor.client_addresses=127.0.0.1
AcquisitionMonitor.udp_port=1112