gnss-sdr/gnss-sdr #124: Tracking of GPS all the time.

mark90054mark, 2018-02-08:
I have installed GNSS-SDR v0.0.9.git-next-199b1a9 on Ubuntu 16.04LTS 64-bit. 
I'm using BladeRF and various GPS antennas (NovAtel, u-blox, Leica, etc.) connected with bias tee. Those antennas have been tested with corresponding receiver and can get position fixed within few minutes. 

When I try to get position fixed, the GNSS-SDR keeps tracking all the time but cannot receive any navigation message. According to my observation, it tracks GPS PRN 24 mostly.

It seems that I should change/add some parameters in my config file attached as below,
[bladerf_github_mark.conf.txt](https://github.com/gnss-sdr/gnss-sdr/files/1706018/bladerf_github_mark.conf.txt)

Can anyone share your config file that works with BladeRF please? Any advice and suggestions will be greatly appreciated.

osqzss, 2018-02-24:
Here's my working configuration file for bladeRF:
https://github.com/osqzss/gnss-sdr/blob/bladerf/conf/gnss-sdr_GPS_L1_bladeRF.conf

The bladeRF gain mapping  of GrOsmoSDR  is a bit confusing.

wxhahxw, 2018-05-30:
Hi osqzss,
Thanks for your configuration file. I tried it on bladeRF, but it still doesn't work. I wait for several minutes, the GNSS-SDR keeps tracking all the time but cannot receive any navigation message. Looking for your help.

osqzss, 2018-05-30:
Taoglas have a good collection of low cost GNSS antennas with reasonable performance.
https://www.digikey.com/product-detail/en/taoglas-limited/AA.108.301111/931-1135-ND/2690282

I'm using GPIO Labs' bias-tee to provide +5V to the antenna LNA.
https://www.tindie.com/products/gpio/1575-mhz-gps-l1-bandpass-filter-with-bias-tee/

closed 2019-08-02 by carlesfernandez