gnss-sdr/gnss-sdr #135: Segment fault when run ‘My first position fix’ in ‘next’ branch

samwiseyu, 2018-03-09:
Hello，I downloaded the whole project of gnss-sdr several days ago. I tried ‘My first position fix’ in ‘master’ branch ，and it worked well. However when I ran 'My first position fix' in 'next' branch this morning，a segment fault appeared.  Before running 'gnss-sdr', I have deleted all the files in 'build' folder , and rebuilt it in 'next' branch.

Please help me, thank you very much!

antonioramosdet, 2018-03-10:
If you are using the same configuration file when running the master branch
version and the next version maybe you are finding some troubles due to
deprecated lines in the config. file.
You can find some updated configuration files of reference in the
repository 'conf' directory.
In any case, can you please send the config. file you are using and the log
file (/tmp) in order to detect exactly where the problem is?
Thank you for the collaboration.

Antonio

samwiseyu, 2018-03-12:
Dear Antonio，
Thank you for your reply！
I downloaded the codes in 'next' branch and build it again today, and found the error information  disappeared and the project worked normally again. By checking the building log, I think the reason is that the codes I downloaded last time were not integrated and some files were missing.

Thank you very much!

closed 2018-03-12 by samwiseyu