gnss-sdr/gnss-sdr #150: Loss of lock in channel

kot-baun, 2018-04-10:
I'm trying to get coordinates with GNSS-SDR (next branch) and LimeSDR. Sometimes I get them, but I usually get the message "Loss of lock in the channel". What is the reason for this and how to fix it?

acebrianjuan, 2018-04-10:
Hi @kot-baun 

The low-level cause of a loss of lock is explained in detail in the documentation page of the [Tracking](http://gnss-sdr.org/docs/sp-blocks/tracking/) block:

> When either the code or the carrier detectors are below given thresholds during a consecutive number of code periods ϑ, the Tracking block informs to control plane through the message queue.

From my own experience, it is commonplace to get _sporadic_ "Loss of lock in the channel" messages during a normal operation of the receiver. This might happen, for example, whenever a satellite stops being in line-of-sight either because it is being shadowed/blocked by a nearby building or because it is below the horizon.

However, if you are _constantly_ seeing this messages for all the channels, this might indicate an anomaly in your receiver configuration. For example, this could be caused by a GPS antenna with a very limited/poor sky visibility or by an incorrect parameter in the receiver configuration file, among many others.

I hope this will help you.

Álvaro

carlesfernandez, 2018-04-10:
Hi,

tracking loops have some metrics that monitor “how well” the block is tracking a GNSS signal. I’ve added to the [Tracking blocks page](http://gnss-sdr.org/docs/sp-blocks/tracking/) a description of those metrics and how they are implemented. If either the code lock or the carrier lock indicators are worse than some given thresholds, and this happens during a given number of integration periods, the tracking is declared out of lock and the satellite is dropped.

If the loss of lock is caused by a severe signal fading, you cannot do too much. However, for mild scenarios, there are some parameters that you can play with in order to mitigate such losses. Typically, this is solved by changing the bandwidth of the tracking loop filters (that is, `dll_bw_hz` and `pll_bw_hz`). If they are too wide, you are accepting to much noise power and the loop will not be able to track the signal; if they are too narrow, rapid changes in the signal parameters will also cause a loss of lock.

In addition, you can also play with the detection thresholds, the number of correlation outputs to perform the C/N0 estimation, or the number of failures before a signal is declared out of lock. Please check the mentioned page for details.

Best regards,
Carles

nycnewman, 2018-05-13:
I'm seeing the same problem with next branch and RTLSDR.org dongle. Works with master branch and get GPS coordinates but get permanent "Loss of lock in channel X" when trying to use next branch. Have looked at document mentioned earlier and tried varying PLL / DLL bandwidth, samples, etc without success. Channel lock lost almost immediately.

Any thoughts on best to diagnose this?

wumouyan, 2018-09-05:
Hi~
I also meet the same problem with next branch and the provided sampled data **2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat** 
I used the provided default ** gnss-sdr.conf** file. 
Works with master branch and get GPS positioning results but get permanent "Loss of lock in channel X" when trying to use next branch.
Channel lock lost almost immediately. 
Is there any problem?
Mouyan

iknow34languages, 2019-01-23:
HI,
This problems happens also when running "Hello world": After get the first position ,and the lock always  loss all the time 

First position fix at 2013-Apr-04 06:24:02.150000 UTC is Lat = 41.2748 [deg], Long = 1.98762 [deg], Height= 85.3143 [m]
Current receiver time: 57 s
### **Loss of lock in channel 5!**
Current receiver time: 58 s
Current receiver time: 59 s
Current receiver time: 1 min 0 s
Current receiver time: 1 min 1 s
Current receiver time: 1 min 2 s
New GPS NAV message received in channel 4: subframe 4 from satellite GPS PRN 20 (Block IIR)
New GPS NAV message received in channel 0: subframe 4 from satellite GPS PRN 01 (Block IIF)
New GPS NAV message received in channel 6: subframe 4 from satellite GPS PRN 32 (Block IIF)
New GPS NAV message received in channel 3: subframe 4 from satellite GPS PRN 11 (Block IIR)
New GPS NAV message received in channel 7: subframe 4 from satellite GPS PRN 17 (Block IIR-M)
Current receiver time: 1 min 3 s
Current receiver time: 1 min 4 s
Current receiver time: 1 min 5 s
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 21 (Block IIR)
Current receiver time: 1 min 6 s
### Loss of lock in channel 5!
Current receiver time: 1 min 7 s
Current receiver time: 1 min 8 s
New GPS NAV message received in channel 4: subframe 5 from satellite GPS PRN 20 (Block IIR)
New GPS NAV message received in channel 0: subframe 5 from satellite GPS PRN 01 (Block IIF)
New GPS NAV message received in channel 6: subframe 5 from satellite GPS PRN 32 (Block IIF)
New GPS NAV message received in channel 3: subframe 5 from satellite GPS PRN 11 (Block IIR)
New GPS NAV message received in channel 7: subframe 5 from satellite GPS PRN 17 (Block IIR-M)
### Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 30 (Block IIF)
Current receiver time: 1 min 9 s
Current receiver time: 1 min 10 s
Current receiver time: 1 min 11 s
Current receiver time: 1 min 12 s
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 21 (Block IIR)
Current receiver time: 1 min 13 s
### Loss of lock in channel 5!
Current receiver time: 1 min 14 s
New GPS NAV message received in channel 4: subframe 1 from satellite GPS PRN 20 (Block IIR)
New GPS NAV message received in channel 0: subframe 1 from satellite GPS PRN 01 (Block IIF)
New GPS NAV message received in channel 6: subframe 1 from satellite GPS PRN 32 (Block IIF)
New GPS NAV message received in channel 3: subframe 1 from satellite GPS PRN 11 (Block IIR)
New GPS NAV message received in channel 7: subframe 1 from satellite GPS PRN 17 (Block IIR-M)
Current receiver time: 1 min 15 s
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 14 (Block IIR)
Current receiver time: 1 min 16 s
Loss of lock in channel 5!
Current receiver time: 1 min 17 s
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 21 (Block IIR)
Current receiver time: 1 min 18 s
Current receiver time: 1 min 19 s
### Loss of lock in channel 5!
Current receiver time: 1 min 20 s
New GPS NAV message received in channel 4: subframe 2 from satellite GPS PRN 20 (Block IIR)
New GPS NAV message received in channel 0: subframe 2 from satellite GPS PRN 01 (Block IIF)
New GPS NAV message received in channel 6: subframe 2 from satellite GPS PRN 32 (Block IIF)
New GPS NAV message received in channel 3: subframe 2 from satellite GPS PRN 11 (Block IIR)
New GPS NAV message received in channel 7: subframe 2 from satellite GPS PRN 17 (Block IIR-M)
Current receiver time: 1 min 21 s
Current receiver time: 1 min 22 s
Current receiver time: 1 min 23 s
Current receiver time: 1 min 24 s
Current receiver time: 1 min 25 s
Current receiver time: 1 min 26 s
New GPS NAV message received in channel 4: subframe 3 from satellite GPS PRN 20 (Block IIR)
New GPS NAV message received in channel 0: subframe 3 from satellite GPS PRN 01 (Block IIF)
New GPS NAV message received in channel 6: subframe 3 from satellite GPS PRN 32 (Block IIF)
New GPS NAV message received in channel 3: subframe 3 from satellite GPS PRN 11 (Block IIR)
New GPS NAV message received in channel 7: subframe 3 from satellite GPS PRN 17 (Block IIR-M)
Current receiver time: 1 min 27 s
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 21 (Block IIR)
Current receiver time: 1 min 28 s
Loss of lock in channel 5!
Current receiver time: 1 min 29 s
Current receiver time: 1 min 30 s
Current receiver time: 1 min 31 s
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 27 (Block IIF)
Current receiver time: 1 min 32 s
New GPS NAV message received in channel 4: subframe 4 from satellite GPS PRN 20 (Block IIR)
New GPS NAV message received in channel 0: subframe 4 from satellite GPS PRN 01 (Block IIF)
New GPS NAV message received in channel 6: subframe 4 from satellite GPS PRN 32 (Block IIF)
New GPS NAV message received in channel 3: subframe 4 from satellite GPS PRN 11 (Block IIR)
New GPS NAV message received in channel 7: subframe 4 from satellite GPS PRN 17 (Block IIR-M)
Current receiver time: 1 min 33 s
Current receiver time: 1 min 34 s
Current receiver time: 1 min 35 s
Current receiver time: 1 min 36 s
Current receiver time: 1 min 37 s
Current receiver time: 1 min 38 s
New GPS NAV message received in channel 4: subframe 5 from satellite GPS PRN 20 (Block IIR)
New GPS NAV message received in channel 0: subframe 5 from satellite GPS PRN 01 (Block IIF)
New GPS NAV message received in channel 6: subframe 5 from satellite GPS PRN 32 (Block IIF)
New GPS NAV message received in channel 3: subframe 5 from satellite GPS PRN 11 (Block IIR)
New GPS NAV message received in channel 7: subframe 5 from satellite GPS PRN 17 (Block IIR-M)
Current receiver time: 1 min 39 s
Current receiver time: 1 min 40 s
Loss of lock in channel 5!
Stopping GNSS-SDR, please wait!

closed 2019-08-02 by carlesfernandez

khangrolean, 2024-02-16:
Hi, I'm trying to test Galileo but below message is executed every time. Please Guide

Galileo E1C secondary code locked in channel 7 for satellite Galileo PRN E21 (Block FOC-FM15)
Current receiver time: 9 min 39 s
Current receiver time: 9 min 40 s
Current receiver time: 9 min 41 s
Current receiver time: 9 min 42 s
Current receiver time: 9 min 43 s
Current receiver time: 9 min 44 s
Current receiver time: 9 min 45 s
Galileo E1C secondary code locked in channel 1 for satellite Galileo PRN E27 (Block FOC-FM17)
Current receiver time: 9 min 46 s
Current receiver time: 9 min 47 s
Current receiver time: 9 min 48 s
Current receiver time: 9 min 49 s
Current receiver time: 9 min 50 s
Current receiver time: 9 min 51 s
Current receiver time: 9 min 52 s
Current receiver time: 9 min 53 s
Current receiver time: 9 min 54 s
Loss of lock in channel 5!
Tracking of Galileo E1C signal started on channel 5 for satellite Galileo PRN E18 (Block FOC-FM1*)
Current receiver time: 9 min 55 s
Current receiver time: 9 min 56 s
Current receiver time: 9 min 57 s
Current receiver time: 9 min 58 s
Current receiver time: 9 min 59 s
Current receiver time: 10 min 0 s
Stopping GNSS-SDR, please wait!
Stopping TCP/IP server on port 2101
Total GNSS-SDR run time: 1199.73 [seconds]
GNSS-SDR program ended.