gnss-sdr/gnss-sdr #184: matlab

desterpol, 2018-06-14:
Hi,

I got the position with gnss

Now I'm trying these matlab scripts.

Can I execute them in post-processing?

The description says:
Read GNSS-SDR Tracking dump binary file

Where can I find the dump files?

Also this path must be modified?
path = '/ home / javier / workspace / gnss-sdr / trunk / install /';

agent00pi, 2018-07-20:
Hi desterpol,

You will need to enable the dump files in your gnss-sdr.conf files, for example:
`;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=true`

In this configuration, an observables.dat and/or observables.mat will be placed in the directory you ran gnss-sdr from. You can specify where this file is places using the `dump_filename` option.

You will also need to modify the path to wherever you have the files stored.

desterpol, 2018-07-25:
and then what can i do with this file?

And in which format thsi file will be saved? If i open it what i can see? 0
and 1?

agent00pi, 2018-08-03:
The .mat files will have arrays of double precision integers. I haven't had a play with the .dat files.

In the *.mat files, you'll see the parameters relevant to that block, so for example in the observables.mat, I can see carrier doppler and phase shift information, as well as information about which PRN code was being tracked at the time.

desterpol, 2018-08-03:
Oh thanks
Do you know how can i change the rate of acquisition of the position? I
mean change the number of position get in one minute!

agent00pi, 2018-08-08:
In your configuration file, under your PVT configuration, you can set the PVT update rate (defaults to 500ms) and terminal output (500ms).