gnss-sdr/gnss-sdr #185: real time receiving issue

larisam22, 2018-06-27:
I tried to receive GPS L1 C/A signal in real time using HackRF One and a GPS antenna ANT555 and GNSS-SDR sotware. My first pozition fix ( https://gnss-sdr.org/my-first-fix/ ) using the downloaded file from the gnss-sdr site worked, decodind location from recorded data.

I use the following configuration:
~~~~~~
[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=2000000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=Osmosdr_Signal_Source
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=2000000
SignalSource.freq=1575420000
;# RF Gain: LNA Gain {0, 3, 6}
SignalSource.gain=6
;# IF Gain: N/A
SignalSource.rf_gain=40
;# BB Gain: RX VGA1 + VGA2 [5, 60]
SignalSource.if_gain=48
SignalSource.AGC_enabled=false
SignalSource.samples=0
SignalSource.repeat=false
SignalSource.osmosdr_args=hackrf,bias=1
SignalSource.enable_throttle_control=false
SignalSource.dump=false
SignalSource.dump_filename=./signal_source.dat

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Pass_Through

;######### INPUT_FILTER CONFIG ############
InputFilter.implementation=Freq_Xlating_Fir_Filter
InputFilter.decimation_factor=1
InputFilter.input_item_type=gr_complex
InputFilter.output_item_type=gr_complex
InputFilter.taps_item_type=float
InputFilter.number_of_taps=5
InputFilter.number_of_bands=2
InputFilter.band1_begin=0.0
InputFilter.band1_end=0.85
InputFilter.band2_begin=0.9
InputFilter.band2_end=1.0
InputFilter.ampl1_begin=1.0
InputFilter.ampl1_end=1.0
InputFilter.ampl2_begin=0.0
InputFilter.ampl2_end=0.0
InputFilter.band1_error=1.0
InputFilter.band2_error=1.0
InputFilter.filter_type=bandpass
InputFilter.grid_density=16
InputFilter.dump=false
InputFilter.dump_filename=../data/input_filter.dat

;######### RESAMPLER CONFIG ############
Resampler.implementation=Pass_Through

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=8
Channels.in_acquisition=1
Channel.signal=1C

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition_Fine_Doppler
Acquisition_1C.item_type=gr_complex
Acquisition_1C.coherent_integration_time_ms=1
Acquisition_1C.threshold=0.015
Acquisition_1C.doppler_max=10000
Acquisition_1C.doppler_step=500
Acquisition_1C.max_dwells=15
Acquisition_1C.dump=false
Acquisition_1C.dump_filename=./acq_dump.dat

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.pll_bw_hz=40.0;
Tracking_1C.dll_bw_hz=2.0;
Tracking_1C.order=3;
Tracking_1C.early_late_space_chips=0.5;
Tracking_1C.dump=false
Tracking_1C.dump_filename=./tracking_ch_

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder
TelemetryDecoder_1C.dump=false

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=false
Observables.dump_filename=./observables.dat

;######### PVT CONFIG ############
;PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.iono_model=Broadcast
PVT.trop_model=Saastamoinen
PVT.flag_rtcm_server=false
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.rtcm_tcp_port=2101
PVT.rtcm_MT1019_rate_ms=5000
PVT.rtcm_MT1077_rate_ms=1000
PVT.rinex_version=2

~~~~~~~~~
After 2 minutes of traking I received some navigation mesage:
~~~~~
Current receiver time: 2 min 25 s
New GPS NAV message received in channel 1: subframe 2 from satellite GPS PRN 15 (Block IIR-M)
Current receiver time: 2 min 26 s
Current receiver time: 2 min 27 s
Current receiver time: 2 min 28 s
Current receiver time: 2 min 29 s
Current receiver time: 2 min 30 s
Current receiver time: 2 min 31 s
New GPS NAV message received in channel 1: subframe 3 from satellite GPS PRN 15 (Block IIR-M)
New GPS NAV message received in channel 0: subframe 3 from satellite GPS PRN 13 (Block IIR)
Current receiver time: 2 min 32 s
Current receiver time: 2 min 33 s
Current receiver time: 2 min 34 s
Current receiver time: 2 min 35 s
Current receiver time: 2 min 36 s
Current receiver time: 2 min 37 s
New GPS NAV message received in channel 1: subframe 4 from satellite GPS PRN 15 (Block IIR-M)
Current receiver time: 2 min 38 s
Current receiver time: 2 min 39 s
Current receiver time: 2 min 40 s
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 27 (Block IIF)
Current receiver time: 2 min 41 s
Current receiver time: 2 min 42 s
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 27 (Block IIF)
Current receiver time: 2 min 43 s
New GPS NAV message received in channel 1: subframe 5 from satellite GPS PRN 15 (Block IIR-M)
Current receiver time: 2 min 44 s
Loss of lock in channel 3!
Current receiver time: 2 min 45 s
Current receiver time: 2 min 46 s
Current receiver time: 2 min 47 s
Current receiver time: 2 min 48 s
Current receiver time: 2 min 49 s
New GPS NAV message received in channel 1: subframe 1 from satellite GPS PRN 15 (Block IIR-M)
New GPS NAV message received in channel 0: subframe 1 from satellite GPS PRN 13 (Block IIR)
New GPS NAV message received in channel 5: subframe 1 from satellite GPS PRN 07 (Block IIR-M)
New GPS NAV message received in channel 7: subframe 1 from satellite GPS PRN 07 (Block IIR-M)
Current receiver time: 2 min 50 s
Current receiver time: 2 min 51 s
Current receiver time: 2 min 52 s
Current receiver time: 2 min 53 s
Current receiver time: 2 min 54 s
Current receiver time: 2 min 55 s
New GPS NAV message received in channel 1: subframe 2 from satellite GPS PRN 15 (Block IIR-M)
New GPS NAV message received in channel 0: subframe 2 from satellite GPS PRN 13 (Block IIR)
New GPS NAV message received in channel 5: subframe 2 from satellite GPS PRN 07 (Block IIR-M)
New GPS NAV message received in channel 7: subframe 2 from satellite GPS PRN 07 (Block IIR-M)
Current receiver time: 2 min 56 s
~~~~~~~

And after 8 minutes:
~~~~~
Current receiver time: 8 min 24 s
Current receiver time: 8 min 25 s
New GPS NAV message received in channel 3: subframe 2 from satellite New GPS NAV message received in channel 2: subframe 2 from satellite GPS PRN GPS27 PRN  (Block IIF)
27 (Block IIF)
New GPS NAV message received in channel 4: subframe 2 from satellite GPS PRN 27 (Block IIF)
New GPS NAV message received in channel 6: subframe 2 from satellite GPS PRN 27 (Block IIF)
New GPS NAV message received in channel 0: subframe 2 from satellite GPS PRN 27 (Block IIF)
New GPS NAV message received in channel 1: subframe 2 from satellite GPS PRN 27 (Block IIF)
New GPS NAV message received in channel 7: subframe 2 from satellite GPS PRN 07 (Block IIR-M)
New GPS NAV message received in channel 5: subframe 2 from satellite GPS PRN 07 (Block IIR-M)
Current receiver time: 8 min 26 s
Current receiver time: 8 min 27 s
Current receiver time: 8 min 28 s
Current receiver time: 8 min 29 s
Current receiver time: 8 min 30 s
Current receiver time: 8 min 31 s
New GPS NAV message received in channel 3: subframe 3 from satellite GPS PRN 27 (Block IIF)
New GPS NAV message received in channel 4: subframe 3 from satellite GPS PRN 27 (Block IIF)
New GPS NAV message received in channel 1: subframe 3 from satellite GPS PRN 27 (Block IIF)
New GPS NAV message received in channel 2: subframe 3 from satellite GPS PRN 27 (Block IIF)
New GPS NAV message received in channel 0: subframe 3 from satellite GPS PRN 27 (Block IIF)
New GPS NAV message received in channel 6: subframe 3 from satellite GPS PRN 27 (Block IIF)
New GPS NAV message received in channel 5: subframe 3 from satellite GPS PRN 07 (Block IIR-M)
New GPS NAV message received in channel 7: subframe 3 from satellite GPS PRN 07 (Block IIR-M)
Current receiver time: 8 min 32 s
Current receiver time: 8 min 33 s
Current receiver time: 8 min 34 s
Current receiver time: 8 min 35 s
Current receiver time: 8 min 36 s
Current receiver time: 8 min 37 s
New GPS NAV message received in channel 7: subframe 4 from satellite GPS PRN 07 (Block IIR-M)
New GPS NAV message received in channel 5: subframe 4 from satellite GPS PRN 07 (Block IIR-M)
~~~~~~~~~

Why I can not decode my current location? Is a configuration issue or I can not detect satellites?I've been trying to get a signal on the roof of 10 storey building

odrisci, 2018-06-28:
That's odd, it appears to be tracking the same signal in multiple channels. I haven't seen that before. What version are you running? 
```bash
$ gnss-sdr --version
```
 prints out the version number when you run it.

Also, try widening the Doppler search space (I usually use 15000) and reducing the Doppler step size (go for 250 Hz). The HackRF is very nice, but can have a very large frequency offset, although the offset is reasonably stable.

larisam22, 2018-06-28:
gnss-sdr --version
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

gnss-sdr version 0.0.9.git-next-8b72c8f9

larisam22, 2018-06-29:
After I reduced the Doppler step size I was able to receive data and decode the location. 
Thank you!

closed 2019-08-02 by carlesfernandez