gnss-sdr/gnss-sdr #189: My first position fix - GNSS-SDR can not create message queues!

alecianfe, 2018-07-04:
Hi,
I'm trying to get through the "My first position fix" tutorial, but I'm facing the following error:
`
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown                                                                                                                                                                                                                                                                  Initializing GNSS-SDR v0.0.9 ... Please wait.                                                                                                                           Logging will be done at "/tmp"                                                                                                                                          Use gnss-sdr --log_dir=/path/to/log to change that.                                                                                                                     Processing file /home/alecianfe/GAD/gnss-sdr/thirdparty/signal_samples/Galileo_E1_ID_1_Fs_4Msps_8ms.dat, which contains 256000 [bytes]                                  
GNSS signal recorded time to be processed: 0.015 [s]                                                                                                                    GNSS-SDR can not create message queues!                                                                                                                                                                                                                                                                                               terminate called after throwing an instance of 'std::exception*'                                                                                                        Aborted (core dumped)    `

Right now I'm using the branch master.
In particular, the message _GNSS-SDR can not create message queues!_ is printed after checking the return value of the function _msgget_. In addition, when returning value -1, which corresponds to an error, it writes the corresponding error value in the variable errno.
Hence, I've modified the code (classes galileo_e1_pvt_cc.cc, gps_l1_ca_pvt_cc.cc, hybrid_pvt_cc.cc and ttff_gps_l1.cc) to read this value and it ended up with error 38, which corresponds to ENOSYS          Function not implemented (POSIX.1-2001).

What am I missing? 

 Thanks

alecianfe, 2018-07-04:
It seems the problem is the environment I'm working on, i.e. Windows Subsystem Linux.
I've found this topic [https://stackoverflow.com/questions/47582180/creating-a-message-queue-using-msgget-in-linux-in-a-c-file-receiving-function-n](url)
I'll trying on native Linux and I'll let you know if it's working

carlesfernandez, 2018-07-04:
Thanks @alecianfe, I've never tried in that environment. Please let me know if it works for you on native Linux (I strongly recommend the next branch), but anyway it's good to know that it fails in that environment. We should disable that feature in that case.

alecianfe, 2018-07-16:
Hi @carlesfernandez. Thank you for your reply. 
I'm now working on Linux Ubuntu 16.04 on GNSS-SDR next branch. I'm still following the "My first position fix" tutorial using the default dataset and conf file I got this output containing some warnings and an error:

```
GNSS signal recorded time to be processed: 99.999 [s]
WARNING: Implementation 'GPS_L1_CA_Observables' of the Observables block has been replaced by 'Hybrid_Observables'.
Please update your configuration file.
WARNING: Implementation 'GPS_L1_CA_PVT' of the PVT block has been replaced by 'RTKLIB_PVT'.
Please update your configuration file.
Using Volk machine: avx2_64_mmx_orc
WARNING: The global parameter name GNSS-SDR.internal_fs_hz has been DEPRECATED.
WARNING: Please replace it by GNSS-SDR.internal_fs_sps in your configuration file.
Set GNSS-SDR.internal_fs_sps in configuration file
E0716 10:00:51.104707  8798 gnss_flowgraph.cc:333] Set GNSS-SDR.internal_fs_sps in configuration
E0716 10:00:51.104769  8798 control_thread.cc:128] Unable to connect flowgraph
Total GNSS-SDR run time: 0.000988782 [seconds]
GNSS-SDR program ended.
```

After fixing the conf file following the traceback I got the following:
 ```
GNSS signal recorded time to be processed: 99.999 [s]
Using Volk machine: avx2_64_mmx_orc
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
gr::buffer::allocate_buffer: warning: tried to allocate
   4 items of size 16000. Due to alignment requirements
   32 were allocated.  If this isn't OK, consider padding
   your structure to a power-of-two bytes.
   On this platform, our allocation granularity is 4096 bytes.
Current receiver time: 1 s
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 01 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 11 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 20 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 32 (Block IIF)
```
Any ideas on why this is not working?

closed 2019-08-02 by carlesfernandez

ZiglioUK, 2021-06-09:
> Thanks @alecianfe, I've never tried in that environment. Please let me know if it works for you on native Linux (I strongly recommend the next branch), but anyway it's good to know that it fails in that environment. We should disable that feature in that case.

For the record: this problem occurs with WSL 1 (legacy), not WSL 2 (recommended)