gnss-sdr/gnss-sdr #190: Math_kernel.cl

DennisCh1989, 2018-07-06:
I read  pcps_opencl_acquisition_cc.cc code  . I've tried to find out where's "math_kernel.cl"   in the project. How can i get this file?. I'm new in OpenCL.

carlesfernandez, 2018-07-06:
Hi @DennisCh1989  

You will need the [OpenCL library](https://www.khronos.org/opencl/), a compatible graphic card and building GNSS-SDR with:

```
$ cmake -DENABLE_OPENCL=ON ..
```
and then, in your configuration file:

```
...
Acquisition_1C.implementation=GPS_L1_CA_PCPS_OpenCl_Acquisition
...
```

Hope this helps.

DennisCh1989, 2018-07-06:
Thanks a lot for your previous answer. I beg you pardon , but I did this and built the project.  The matter is  the piece of code 
// build the program from the source in the file
    std::ifstream kernel_file(kernel_filename, std::ifstream::in);
    std::string kernel_code(std::istreambuf_iterator<char>(kernel_file),
        (std::istreambuf_iterator<char>()));
    kernel_file.close();
in  pcps_opencl_acquisition_cc.cc. It needs  text file "math_kernel.cl" which contains several sources of kernels
My a little more experience colleagues  in OpenCL field said me that this text files must contain  in repo of the project.  I didn't find this file in any branch.Maybe you have forgotten to add "math_kernel.cl"  due to  an openCl acceleration is  a rarely used feature of gnss-sdr project??? My own tests with run_tests utility give positive results with openCl implementations of forward and inverse FFTs, but not with vector operations like "conj_vector" ( see acq code with the opencl acceleration). 
I think you will find my  final purpose interesting for you.
I try to implement  passive radar  based on  gps. There  are several steps of  its algorithm:
1) regenerate direct signal paths of the satellites
2) subtract regenerated signals in order to reduce interference noise , which  is important for the next step
3)perform convolutional on a long time period (2 seconds )  to get peaks from landscapes and airplanes from different satellites. Obviously, this step needs acceleration with openCL.
Many features were already added to the fork of gnss-sdr. Testing is in progress.

DennisCh1989, 2018-07-06:
I have USRP B200 and run OpenCL at  CPU now(I haven't  GPU now).

closed 2018-07-07 by DennisCh1989