gnss-sdr/gnss-sdr #192: Tracking issue: Loss of lock in every Channel

matteocutugno, 2018-07-09:
Hello,
I have a problem with bladerf signal. I've downloaded it from Taroz's repository (http://www.taroz.net/gnsssdrlib/data/bladerf_l1.zip) 
and i tried to process it in postprocessing.
The issue is that th tracking block doesn't work.
Here is my .conf file:

[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=20000000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=File_Signal_Source
SignalSource.filename=/media/matteo/Archivio/20180708-blade/bladerf_l1.bin
SignalSource.item_type=ishort
SignalSource.sampling_frequency=4000000
SignalSource.freq=1575420000
;SignalSource.gain=40
;SignalSource.rf_gain=40
;SignalSource.if_gain=30
SignalSource.samples=0

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner
DataTypeAdapter.implementation=Ishort_To_Complex
InputFilter.implementation=Pass_Through
InputFilter.item_type=gr_complex
Resampler.implementation=Direct_Resampler
Resampler.sample_freq_in=2000000
Resampler.sample_freq_out=20000000
Resampler.item_type=gr_complex

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=8
Channels.in_acquisition=1
Channel.signal=1C

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.dump=true
Acquisition_1C.dump_filename=./acquisition_ch_
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.if=0
Acquisition_1C.threshold=0.005
Acquisition_1C.sampled_ms=1
Acquisition_1C.doppler_max=10000
Acquisition_1C.doppler_min=-10000
Acquisition_1C.doppler_step=250
;Acquisition_1C.max_dwells=5

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.if=0
Tracking_1C.dump=true
Tracking_1C.dump_filename=./tracking_ch_
Tracking_1C.pll_bw_hz=70.0;
Tracking_1C.dll_bw_hz=7.0;
Tracking_1C.order=3;

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder
TelemetryDecoder_1C.dump=true
TelemetryDecoder_1C.dump_filename=./telemetry_ch_

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=true
Observables.dump_filename=./obs_

[... 130 lines cut ...]
I0708 17:31:03.461503 13288 gnss_flowgraph.cc:898] Channel 5 ACQ SUCCESS satellite GPS PRN 20 (Block IIR)
I0708 17:31:03.684963 13348 dll_pll_veml_tracking.cc:650] Loss of lock in channel 0!
I0708 17:31:03.685575 13288 gnss_flowgraph.cc:928] Channel 0 TRK FAILED satellite GPS PRN 01 (Block IIF)
I0708 17:31:03.685600 13288 gnss_flowgraph.cc:935] Channel 0 Starting acquisition GPS PRN 01 (Block IIF), Signal 1C
I0708 17:31:03.784517 13305 dll_pll_veml_tracking.cc:519] Starting tracking of satellite GPS PRN 01 (Block IIF) on channel 0
I0708 17:31:03.784535 13305 dll_pll_veml_tracking.cc:527] PULL-IN Doppler [Hz] = -7500. Code Phase correction [samples] = 0.433548. PULL-IN Code Phase [samples] = 5490.57
I0708 17:31:03.784607 13288 gnss_flowgraph.cc:898] Channel 0 ACQ SUCCESS satellite GPS PRN 01 (Block IIF)
I0708 17:31:04.373543 13318 dll_pll_veml_tracking.cc:650] Loss of lock in channel 6!
I0708 17:31:04.373654 13288 gnss_flowgraph.cc:928] Channel 6 TRK FAILED satellite GPS PRN 16 (Block IIR)
I0708 17:31:04.373677 13288 gnss_flowgraph.cc:935] Channel 6 Starting acquisition GPS PRN 16 (Block IIR), Signal 1C
I0708 17:31:04.594424 13299 dll_pll_veml_tracking.cc:519] Starting tracking of satellite GPS PRN 16 (Block IIR) on channel 6
I0708 17:31:04.594440 13299 dll_pll_veml_tracking.cc:527] PULL-IN Doppler [Hz] = -7500. Code Phase correction [samples] = 1.55087. PULL-IN Code Phase [samples] = 12986.4
I0708 17:31:04.594501 13288 gnss_flowgraph.cc:898] Channel 6 ACQ SUCCESS satellite GPS PRN 16 (Block IIR)
I0708 17:31:09.189141 13338 dll_pll_veml_tracking.cc:650] Loss of lock in channel 2!
I0708 17:31:09.189479 13288 gnss_flowgraph.cc:928] Channel 2 TRK FAILED satellite GPS PRN 10 (Block IIF)
I0708 17:31:09.189508 13288 gnss_flowgraph.cc:935] Channel 2 Starting acquisition GPS PRN 10 (Block IIF), Signal 1C
I0708 17:31:09.318686 13303 dll_pll_veml_tracking.cc:519] Starting tracking of satellite GPS PRN 10 (Block IIF) on channel 2
....... and so on

It seems that DLL and PLL loops can't estimate doppler shift or something else. I tried with different parameters in acquisition block( Doppler.max, Doppler.step and Doppler.Threshold)
and also in Tracking block (bandwidth of dll and pll)

Thank you in advance for your help

Matteo

Paulth18, 2018-07-18:
Are you sure that this " SignalSource.sampling_frequency=4000000" is correct? Try with 2

And then, why you are using bw so large for filter?
Tracking_1C.pll_bw_hz=70.0;
Tracking_1C.dll_bw_hz=7.0;
Try with more narrow value.

Anyway try with some other file, if the problem remain comment here again.

matteocutugno, 2018-07-19:
Thank you for your answer.

I tried with SignalSource.sampling_frequency=2000000
And I also tried to narrowing bw.
It continues to loss lock.
With other signals like Spain sample (my first position fix) it works.
I was interested in testing bladerf x40 signal sample because i've just ordered one and next week i hope they will deliver to me. Soi'm afraid i'm not able to make it work.

Anyway i dont't understand how i have to set some paremeters of acquisition and tracking loop that depends on the signal like:
Acquisition_1C.threshold
Acquisition_1C.doppler_max
Acquisition_1C.doppler_min
Acquisition_1C.doppler_step
Tracking_1C.pll_bw_hz
Tracking_1C.dll_bw_hz

Thank you

Matteo

Paulth18, 2018-07-19:
Ma sei italiano?

matteocutugno, 2018-07-19:
si

closed 2020-09-18 by matteocutugno