gnss-sdr/gnss-sdr #194: Trying to capture realtime signal using RTL_SDR Dongle and sdr stops at 'Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 30 (Block IIF)'

shujjahyamin, 2018-07-12:
Dear All,
Currently,I have installed gnss-sdr on Ubuntu 16.04 LTS version and using Windows 10 on same system. 

I am trying to get realtime GPS Signal using gnss-sdr. Tried to install gnss-sdr on multiple systems multiple times but getting loss of lock errors.

Previously i was able to capture realtime signal, but don't know what's happeing now.

After every thing setup correctly using osmo-sdr here's my log:

```
shujjah@shujjah-Y11C:/media/shujjah/Data/Work/SDR$ sudo gnss-sdr --config_file=gnss-sdr_GPS_L1_rtlsdr_realtime.conf
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

Initializing GNSS-SDR v0.0.9.git-next-15e86c8 ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
OsmoSdr arguments: rtl,bias=1
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy redpitaya 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Exact sample rate is: 2000009.059947 Hz
[R82XX] PLL not locked!
Actual RX Rate: 2000009.000000 [SPS]...
Actual RX Freq: 1575420000.000000 [Hz]...
PLL Frequency tune error 0.000000 [Hz]...
AGC enabled
Actual Bandwidth: 0.000000 [Hz]...
Using Volk machine: avx2_64_mmx_orc
Current receiver time: 1 s
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 01 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 02 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 03 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 04 (Block Unknown)
Current receiver time: 2 s
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 01 (Block IIF)
Loss of lock in channel 2!
Loss of lock in channel 0!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 03 (Block IIF)
Loss of lock in channel 3!
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 05 (Block IIR-M)
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 06 (Block IIF)
Current receiver time: 3 s
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 01 (Block IIF)
Loss of lock in channel 0!
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 05 (Block IIR-M)
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 07 (Block IIR-M)
Loss of lock in channel 3!
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 06 (Block IIF)
Current receiver time: 4 s
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 01 (Block IIF)
Loss of lock in channel 0!
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 05 (Block IIR-M)
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 07 (Block IIR-M)
Loss of lock in channel 3!
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 06 (Block IIF)
Current receiver time: 5 s
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 01 (Block IIF)
Loss of lock in channel 0!
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 05 (Block IIR-M)
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 07 (Block IIR-M)
Loss of lock in channel 3!
[... 508 lines cut ...]
;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=false
Observables.dump_filename=./observables.dat

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=PPP_Static  ; options: Single, Static, Kinematic, PPP_Static, PPP_Kinematic
PVT.iono_model=Broadcast ; options: OFF, Broadcast, SBAS, Iono-Free-LC, Estimate_STEC, IONEX
PVT.trop_model=Saastamoinen ; options: OFF, Saastamoinen, SBAS, Estimate_ZTD, Estimate_ZTD_Grad
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.dump_filename=./PVT
PVT.nmea_dump_filename=./gnss_sdr_pvt.nmea;
PVT.flag_nmea_tty_port=false;
PVT.nmea_dump_devname=/dev/pts/4
PVT.dump=false
PVT.flag_rtcm_server=false
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1

Please, help me with this i'll be very thankful.

Regards,
Shujjah

rommar, 2018-07-24:
I had similar problem with my old laptop. As it turned out after several days it just didn't have enough processing power to handle samples in real time. I realized that when I saw that "Current receiver time: .." messages actually go slower than real time.
Could this be the case for you?

shujjahyamin, 2018-07-25:
Don't know sir, i am using three laptops but same problem.

shujjahyamin, 2018-07-27:
> Perhaps you could start gnss-sdr, open some clock widget in Linux and just observe whether "current receiver time" reported by gnss-sdr goes with the same speed as computer clock.
>
> Is gnss-sdr working ok with example RF samples file (mentioned in the manual or quick start)?
>
> What type of SDR frontend do you actually use?
> Do you have active antenna and bias-t?
> Why do you have such strange sampling frequency? Are you sure it is correct value?

Dear Rommar,

I am using RTL SDR Dongle 2838U as RF Frontend. Yes it has bias-tee, i am connecting an external LNA also with it. I am running front-end calibration file to check what sampling frequency is and then use this calibrated value in my gnss-sdr-realtime-rtl configration file.

Regards,

Shujjah

klyg666, 2022-02-23:
Sir,I also encountered this problem（Loss of lock in channel 2！） with RTLSDR Dongle 2838U. How do you solve this problem？

YHao29, 2023-05-23:
I have the same issue when I use USRP. Is there any solutions?

carlesfernandez, 2025-03-14:
Non-integer sampling frequencies are not allowed.

closed 2025-03-14 by carlesfernandez