gnss-sdr/gnss-sdr #207: Receiver stops logging but does not terminate

dmiralles2009, 2018-08-29:
This seems to be a bug in the platform given the conf file I am using. It is as if the software freezes in a section at the middle of processing. See receiver output below. As you can see the total receiver time is 364 vs 5min 13 secs of actual processing. Also noticed that I end up pressing 'q' to force termination. Otherwise will loop forever there.  I was unable to check where the software actually freezes using gdb so at the moment I am clueless. 

Receiver output
'''
Position at 2016-Sep-23 23:10:44.000000 UTC using 4 observations is Lat = 40.008150329 [deg], Long = -105.262698061 [deg], Height = 1653.663 [m]
Position at 2016-Sep-23 23:10:44.500000 UTC using 4 observations is Lat = 40.008050290 [deg], Long = -105.262721250 [deg], Height = 1641.758 [m]
Current receiver time: 5 min 10 s
Position at 2016-Sep-23 23:10:45.000000 UTC using 4 observations is Lat = 40.008171581 [deg], Long = -105.262667973 [deg], Height = 1662.022 [m]
Position at 2016-Sep-23 23:10:45.500000 UTC using 4 observations is Lat = 40.008124513 [deg], Long = -105.262709682 [deg], Height = 1652.998 [m]
Current receiver time: 5 min 11 s
Position at 2016-Sep-23 23:10:46.000000 UTC using 4 observations is Lat = 40.008034266 [deg], Long = -105.262659440 [deg], Height = 1635.572 [m]
Position at 2016-Sep-23 23:10:46.500000 UTC using 4 observations is Lat = 40.008064821 [deg], Long = -105.262694238 [deg], Height = 1636.014 [m]
Current receiver time: 5 min 12 s
Position at 2016-Sep-23 23:10:47.000000 UTC using 4 observations is Lat = 40.007986724 [deg], Long = -105.262703902 [deg], Height = 1631.842 [m]
Position at 2016-Sep-23 23:10:47.500000 UTC using 4 observations is Lat = 40.008186608 [deg], Long = -105.262682831 [deg], Height = 1657.917 [m]
Current receiver time: 5 min 13 s
Position at 2016-Sep-23 23:10:48.000000 UTC using 4 observations is Lat = 40.008030423 [deg], Long = -105.262717771 [deg], Height = 1636.301 [m]
q
Quit keystroke order received, stopping GNSS-SDR !!
Stopping GNSS-SDR, please wait!
Total GNSS-SDR run time: 364.126862 [seconds]
GNSS-SDR program ended.
Writing observables .mat files ... done.
'''
Sample data, configuration files, receiver logs gnss-sdr.INFO and gnss-sdr.WARNING files can be found here: https://drive.google.com/drive/folders/1wh8jPr2aedZQOD8kGNliqjiRwqn66Tag?usp=sharing

Thanks in advance for any help
Damian

dmiralles2009, 2018-08-30:
After some debugging, I have some additional information to share. 
1. This issue is happening in Ubuntu 16.04 (`Linux ratel-pro 4.4.0-134-generic #160-Ubuntu SMP Wed Aug 15 14:58:00 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux`)
2. Running `next` branch
3. If running with gdb and building with `cmake -DCMAKE_BUILD_TYPE=Coverage ../` I get the following output
```
Position at 2016-Sep-23 23:10:47.500000 UTC using 4 observations is Lat = 40.008186608 [deg], Long = -105.262682831 [deg], Height = 1657.917 [m]
Current receiver time: 5 min 13 s
Position at 2016-Sep-23 23:10:48.000000 UTC using 4 observations is Lat = 40.008030508 [deg], Long = -105.262717668 [deg], Height = 1636.273 [m]
[Thread 0x7fffe8e73700 (LWP 11000) exited]
[Thread 0x7fffe3fff700 (LWP 11001) exited]
[Thread 0x7fffe37fe700 (LWP 11002) exited]
[Thread 0x7fffe2676700 (LWP 11003) exited]
[Thread 0x7fff8ffff700 (LWP 11014) exited]
[Thread 0x7fff6f7fe700 (LWP 11022) exited]
[Thread 0x7fff8dffb700 (LWP 11018) exited]
[Thread 0x7fff6dffb700 (LWP 11025) exited]
[Thread 0x7fff6effd700 (LWP 11023) exited]
[Thread 0x7fff6ffff700 (LWP 11021) exited]
[Thread 0x7fff6e7fc700 (LWP 11024) exited]
[Thread 0x7fff8cff9700 (LWP 11020) exited]
[Thread 0x7fff8d7fa700 (LWP 11019) exited]
[Thread 0x7fff8e7fc700 (LWP 11017) exited]
[Thread 0x7fff8effd700 (LWP 11016) exited]
[Thread 0x7fff8f7fe700 (LWP 11015) exited]
[Thread 0x7fffb0ff9700 (LWP 11013) exited]
[Thread 0x7fffb17fa700 (LWP 11012) exited]
[Thread 0x7fffb1ffb700 (LWP 11011) exited]
[Thread 0x7fffb27fc700 (LWP 11010) exited]
[Thread 0x7fffb2ffd700 (LWP 11009) exited]
[Thread 0x7fffb37fe700 (LWP 11008) exited]
[Thread 0x7fffb3fff700 (LWP 11007) exited]
[Thread 0x7fffe0e73700 (LWP 11006) exited]
[Thread 0x7fffe1674700 (LWP 11005) exited]
[Thread 0x7fffe1e75700 (LWP 11004) exited]
[Thread 0x7fff4effd700 (LWP 11030) exited]
[Thread 0x7fff4f7fe700 (LWP 11029) exited]
[Thread 0x7fff4e7fc700 (LWP 11031) exited]
[Thread 0x7fff4ffff700 (LWP 11028) exited]
[Thread 0x7fff6cff9700 (LWP 11027) exited]
[Thread 0x7fff6d7fa700 (LWP 11026) exited]
[Thread 0x7fffe9e75700 (LWP 10998) exited]
[Thread 0x7fffea676700 (LWP 10997) exited]
[Thread 0x7fffeae77700 (LWP 10996) exited]
[Thread 0x7fffe2e77700 (LWP 10995) exited]
[Thread 0x7fffeb678700 (LWP 10994) exited]
[Thread 0x7fffe9674700 (LWP 10999) exited]
^C
Thread 1 "gnss-sdr" received signal SIGINT, Interrupt.
pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
185	../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S: No such file or directory.
(gdb) where
#0  pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
#1  0x00007ffff4280efb in gr::msg_queue::delete_head() ()
   from /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.7.9
#2  0x00000000004f2106 in ControlThread::read_control_messages (this=0x2d46720)
    at /home/gnss/Documents/gnss-sdr/src/core/receiver/control_thread.cc:470
#3  0x00000000004eb0b3 in ControlThread::run (this=0x2d46720)
    at /home/gnss/Documents/gnss-sdr/src/core/receiver/control_thread.cc:160
#4  0x00000000004d204d in main (argc=1, argv=0x7fffffffdcc0) at /home/gnss/Documents/gnss-sdr/src/main/main.cc:140
```
Not sure if this is a thread issue on the platform, curious if someone else has been experiencing something similar

dmiralles2009, 2018-08-31:
Further testing and debugging leads me to think that the issue at hand here is that the receiver process all the data available but it simply does not exits programs execution. It is like it gets stuck when writing the data to file after `gnss_sdr_valve` parses all contents of the file

carlesfernandez, 2018-09-07:
Dear @dmiralles2009 

Please add the following line to your configuration file:

```
SignalSource.sampling_frequency=6625000
```

dmiralles2009, 2018-09-07:
Thanks, @carlesfernandez  that fixed the issue. Silly mistake on my part. I will close the issue then. Hopefully, this could help other as well.

Thanks again

closed 2018-09-07 by dmiralles2009