gnss-sdr/gnss-sdr #209: How to configure a .conf file for SignalSource.implementation=File_Signal_Source, SignalSource.item_type=byte and IF is not zero?

wumouyan, 2018-09-06:
Hi~
I would like to configure a .conf file for the GNSS-SDR, which SignalSource.implementation=File_Signal_Source, SignalSource.item_type=byte and IF is not zero? 
I post my .conf file as follows. The results of the GNSS-SDR are only some signals tracked but not stable while no any positioning result.
The data was tested successfully in Matlab with good tracking and positioning results.

I am in a puzzle about the INPUT_FILTER CONFIG, can anyone tell me  how to understand and configure it?

Best~
Mouyan
----------------------------------------------------
[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
;internal_fs_hz: Internal signal sampling frequency after the signal conditioning stage [Hz].
GNSS-SDR.internal_fs_hz=5455889

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=File_Signal_Source
SignalSource.filename=/home/wu/work/gnss-sdr/data/GPSca-fs16.367667-if4.123968.bin
SignalSource.item_type=byte
SignalSource.sampling_frequency=16367667
SignalSource.freq=1575420000
SignalSource.samples=0
SignalSource.repeat=false
SignalSource.dump=false
SignalSource.dump_filename=../data/signal_source.dat
SignalSource.enable_throttle_control=false

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Pass_Through
DataTypeAdapter.item_type=byte

;######### INPUT_FILTER CONFIG ############
InputFilter.implementation=Freq_Xlating_Fir_Filter
InputFilter.dump=false
InputFilter.dump_filename=../data/input_filter.dat
InputFilter.input_item_type=byte
InputFilter.output_item_type=gr_complex
InputFilter.taps_item_type=float
InputFilter.number_of_taps=5
InputFilter.number_of_bands=2
InputFilter.band1_begin=0
InputFilter.band1_end=0.45
InputFilter.band2_begin=0.55
InputFilter.band2_end=1.0
InputFilter.ampl1_begin=1.0
InputFilter.ampl1_end=1.0
InputFilter.ampl2_begin=0.0
InputFilter.ampl2_end=0.0
InputFilter.band1_error=1.0
InputFilter.band2_error=1.0
InputFilter.filter_type=bandpass
InputFilter.grid_density=16
InputFilter.sampling_frequency=16367667
InputFilter.IF=4123968
InputFilter.decimation_factor=3

;######### RESAMPLER CONFIG ############
Resampler.implementation=Pass_Through
Resampler.dump=false
Resampler.dump_filename=../data/resampler.dat
Resampler.item_type=gr_complex

;######### CHANNELS GLOBAL CONFIG ############
;#count: Number of available GPS satellite channels.
Channels_1C.count=8
Channels.in_acquisition=1
Channel.signal=1C

;######### GPS ACQUISITION CONFIG ############
Acquisition_1C.dump=false
Acquisition_1C.dump_filename=./acq_dump.dat
Acquisition_1C.item_type=gr_complex
Acquisition_1C.if=0
Acquisition_1C.sampled_ms=1
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.threshold=0.008
Acquisition_1C.pfa=0.001
Acquisition_1C.doppler_max=10000
Acquisition_1C.doppler_step=250

;######### TRACKING GPS CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.if=0
Tracking_1C.dump=false
Tracking_1C.dump_filename=../data/epl_tracking_ch_
Tracking_1C.pll_bw_hz=30.0;
Tracking_1C.dll_bw_hz=2.0;
Tracking_1C.order=3;

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder
TelemetryDecoder_1C.dump=false
;TelemetryDecoder_1C.decimation_factor=1;

;######### OBSERVABLES CONFIG ############
Observables.implementation=GPS_L1_CA_Observables
Observables.dump=false
Observables.dump_filename=./observables.dat

;######### PVT CONFIG ############
PVT.implementation=GPS_L1_CA_PVT
PVT.averaging_depth=100
PVT.flag_averaging=false
PVT.output_rate_ms=10
PVT.display_rate_ms=500
PVT.dump_filename=./PVT
PVT.nmea_dump_filename=./gnss_sdr_pvt.nmea;
PVT.flag_nmea_tty_port=false;
PVT.nmea_dump_devname=/dev/pts/4
PVT.flag_rtcm_server=false
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.dump=true

carlesfernandez, 2018-09-07:
This configuration looks good to me. ¿Could you try to use the 'next' branch? Configuration remains the same but except for the last two blocks:

```
;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=false
Observables.dump_filename=./observables.dat

;######### PVT CONFIG if using the next branch ####
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.iono_model=false
PVT.trop_model=false
PVT.flag_rtcm_server=false
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.rinex_version=2
```

wumouyan, 2018-09-08:
@carlesfernandez  I try it, but it still not work well.
Part of tracking results are as follows:
----------------
启动：/home/wu/work/gnss-sdr/gnss-sdr-next/build/src/main/gnss-sdr --config_file=/home/wu/work/gnss-sdr/mysdr_conf/gnss-sdr-next-byteinput.conf --signal_source=/home/wu/work/gnss-sdr/data/GPSca-fs16.367667-if4.123968-16House.bin
linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

Initializing GNSS-SDR v0.0.9.git-- ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
Processing file /home/wu/work/gnss-sdr/data/GPSca-fs16.367667-if4.123968-16House.bin, which contains 3273533400 [bytes]
GNSS signal recorded time to be processed: 199.998 [s]
Using Volk machine: avx_64_mmx_orc
WARNING: Bad specification of ionospheric model.
iono_model possible values: OFF / Broadcast / SBAS / Iono-Free-LC / Estimate_STEC / IONEX
iono_model specified value: false
Setting iono_model to OFF
WARNING: Bad specification of tropospheric model.
trop_model possible values: OFF / Saastamoinen / SBAS / Estimate_ZTD / Estimate_ZTD_Grad
trop_model specified value: false
Setting trop_model to OFF
Current receiver time: 1 s
Observables clock step samples set to 5456
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 01 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 09 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 10 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 11 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 12 (Block IIR-M)
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 13 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 14 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 15 (Block IIR-M)
Current receiver time: 2 s
Loss of lock in channel 0!
Loss of lock in channel 2!
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 01 (Block IIF)
Loss of lock in channel 4!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 16 (Block IIR)
Loss of lock in channel 3!
Loss of lock in channel 5!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 17 (Block IIR-M)
Loss of lock in channel 7!
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 18 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 19 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 20 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 21 (Block IIR)
Current receiver time: 3 s
Loss of lock in channel 0!
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 01 (Block IIF)
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 16 (Block IIR)
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 17 (Block IIR-M)
Loss of lock in channel 5!
Loss of lock in channel 4!
Loss of lock in channel 3!
Loss of lock in channel 7!
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 20 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 22 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 23 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 24 (Block IIF)
Current receiver time: 4 s
Loss of lock in channel 0!
Loss of lock in channel 1!
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 01 (Block IIF)
Loss of lock in channel 2!
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 25 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 26 (Block IIF)
Loss of lock in channel 3!
Loss of lock in channel 4!
Loss of lock in channel 5!
[... 372 lines cut ...]
Loss of lock in channel 5!
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 28 (Block IIR)
Current receiver time: 1 min 10 s
Loss of lock in channel 0!
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 01 (Block IIF)
Loss of lock in channel 5!
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 28 (Block IIR)
Current receiver time: 1 min 11 s
Loss of lock in channel 0!
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 01 (Block IIF)
New GPS NAV message received in channel 4: subframe 1 from satellite GPS PRN 31 (Block IIR-M)
New GPS NAV message received in channel 6: subframe 1 from satellite GPS PRN 14 (Block IIR)
New GPS NAV message received in channel 1: subframe 1 from satellite GPS PRN 25 (Block IIF)
New GPS NAV message received in channel 7: subframe 1 from satellite GPS PRN 29 (Block IIR-M)
New GPS NAV message received in channel 2: subframe 1 from satellite GPS PRN 26 (Block IIF)
New GPS NAV message received in channel 3: subframe 1 from satellite GPS PRN 22 (Block IIR)
Current receiver time: 1 min 12 s
Loss of lock in channel 5!
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 28 (Block IIR)
Loss of lock in channel 0!
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 01 (Block IIF)
Current receiver time: 1 min 13 s
Loss of lock in channel 5!
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 28 (Block IIR)
Loss of lock in channel 0!

wumouyan, 2018-09-08:
Also part of the log messages are posted as follows:
------------------------------------------------------------------------------------------------------------------------------------------
Log file created at: 2018/09/08 10:54:22
Running on machine: wu
Log line format: [IWEF]mmdd hh:mm:ss.uuuuuu threadid file:line] msg
I0908 10:54:22.573416  4173 file_configuration.cc:191] Configuration file /home/wu/work/gnss-sdr/mysdr_conf/gnss-sdr-next-byteinput.conf opened with no errors
I0908 10:54:22.606052  4173 gnss_block_factory.cc:197] Getting SignalSource with implementation File_Signal_Source
I0908 10:54:22.721129  4173 file_signal_source.cc:174] file_source(0)
I0908 10:54:22.721238  4173 file_signal_source.cc:189] Total samples in the file= 3.27353e+09
I0908 10:54:22.721359  4173 file_signal_source.cc:218] Total number samples to be processed= 3273500664 GNSS signal duration= 199.998 [s]
I0908 10:54:22.721608  4173 file_signal_source.cc:222] valve(1)
I0908 10:54:22.721633  4173 file_signal_source.cc:235] File source filename /home/wu/work/gnss-sdr/data/GPSca-fs16.367667-if4.123968-16House.bin
I0908 10:54:22.721644  4173 file_signal_source.cc:236] Samples 3273500664
I0908 10:54:22.721664  4173 file_signal_source.cc:237] Sampling frequency 16367667
I0908 10:54:22.721675  4173 file_signal_source.cc:238] Item type byte
I0908 10:54:22.721686  4173 file_signal_source.cc:239] Item size 1
I0908 10:54:22.721698  4173 file_signal_source.cc:240] Repeat 0
I0908 10:54:22.721719  4173 file_signal_source.cc:241] Dump 0
I0908 10:54:22.721730  4173 file_signal_source.cc:242] Dump filename ../data/signal_source.dat
I0908 10:54:22.721829  4173 gnss_block_factory.cc:243] Getting SignalConditioner with DataTypeAdapter implementation: Pass_Through, InputFilter implementation: Freq_Xlating_Fir_Filter, and Resampler implementation: Pass_Through
I0908 10:54:22.722105  4173 pass_through.cc:111] kludge_copy(2)
I0908 10:54:22.722154  4173 freq_xlating_fir_filter.cc:61] role InputFilter
I0908 10:54:22.783627  4173 freq_xlating_fir_filter.cc:164] input_filter(4)
I0908 10:54:22.783908  4173 pass_through.cc:111] kludge_copy(5)
I0908 10:54:22.783974  4173 gnss_block_factory.cc:275] Getting Observables with implementation Hybrid_Observables
I0908 10:54:22.784065  4173 hybrid_observables.cc:45] role Observables
I0908 10:54:22.784741  4173 hybrid_observables.cc:50] Observables block ID (6)
I0908 10:54:22.784775  4173 gnss_block_factory.cc:298] Getting PVT with implementation RTKLIB_PVT
I0908 10:54:22.784827  4173 rtklib_pvt.cc:62] role PVT
I0908 10:54:22.785626  4173 kml_printer.cc:90] KML printer writing on ./pvt.dat
I0908 10:54:22.786041  4173 gpx_printer.cc:91] GPX printer writing on ./pvt.dat
I0908 10:54:22.786249  4173 geojson_printer.cc:105] GeoJSON printer writing on ./pvt.dat
I0908 10:54:22.786368  4173 nmea_printer.cc:53] NMEA printer writing on ./nmea_pvt.nmea
I0908 10:54:22.786447  4173 rtcm_printer.cc:96] RTCM printer writing on ./pvt.dat_180908_105422.rtcm
I0908 10:54:22.835047  4173 rtklib_pvt.cc:494] pvt(7)
I0908 10:54:22.835158  4173 gnss_block_factory.cc:816] Getting 8 GPS L1 C/A channels
I0908 10:54:22.835239  4173 gnss_block_factory.cc:318] Instantiating Channel 0 with Acquisition Implementation: GPS_L1_CA_PCPS_Acquisition, Tracking Implementation: GPS_L1_CA_DLL_PLL_Tracking, Telemetry Decoder implementation: GPS_L1_CA_Telemetry_Decoder
I0908 10:54:22.835479  4173 pass_through.cc:111] kludge_copy(8)
I0908 10:54:22.835501  4173 gps_l1_ca_pcps_acquisition.cc:57] role Acquisition_1C
I0908 10:54:23.224527  4173 gps_l1_ca_pcps_acquisition.cc:106] acquisition(9)
I0908 10:54:23.224572  4173 gps_l1_ca_dll_pll_tracking.cc:53] role Tracking_1C
I0908 10:54:23.224876  4173 gps_l1_ca_dll_pll_tracking.cc:133] tracking(10)
I0908 10:54:23.224910  4173 gps_l1_ca_telemetry_decoder.cc:53] role TelemetryDecoder_1C
I0908 10:54:23.225087  4173 gnss_satellite.cc:124] System  is not defined {GPS, Glonass, SBAS, Galileo, Beidou}. Initialization?
I0908 10:54:23.225095  4173 gnss_satellite.cc:158] Trying to define PRN while system is not defined
I0908 10:54:23.225100  4173 gnss_satellite.cc:227] System  is not defined
I0908 10:54:23.225124  4173 gps_l1_ca_telemetry_decoder.cc:58] telemetry_decoder(11)
I0908 10:54:23.225162  4173 dll_pll_veml_tracking.cc:1207] Tracking Channel set to 0
I0908 10:54:23.225168  4173 gps_l1_ca_telemetry_decoder_cc.cc:159] Navigation channel set to 0
I0908 10:54:23.225194  4173 channel.cc:82] Channel 0 Doppler_step = 250
I0908 10:54:23.225221  4173 gps_l1_ca_pcps_acquisition.cc:234] Channel 0  Pfa = 0.001
I0908 10:54:23.225255  4173 gps_l1_ca_pcps_acquisition.cc:155] Channel 0 Threshold = 0.00364913
I0908 10:54:23.299301  4173 channel.cc:94] Channel 0 satellite repeat = 0
I0908 10:54:23.299520  4173 gnss_block_factory.cc:318] Instantiating Channel 1 with Acquisition Implementation: GPS_L1_CA_PCPS_Acquisition, Tracking Implementation: GPS_L1_CA_DLL_PLL_Tracking, Telemetry Decoder implementation: GPS_L1_CA_Telemetry_Decoder
I0908 10:54:23.299643  4173 pass_through.cc:111] kludge_copy(13)
I0908 10:54:23.299657  4173 gps_l1_ca_pcps_acquisition.cc:57] role Acquisition_1C
I0908 10:54:23.305426  4173 gps_l1_ca_pcps_acquisition.cc:106] acquisition(14)
I0908 10:54:23.305456  4173 gps_l1_ca_dll_pll_tracking.cc:53] role Tracking_1C
I0908 10:54:23.305706  4173 gps_l1_ca_dll_pll_tracking.cc:133] tracking(15)
I0908 10:54:23.305728  4173 gps_l1_ca_telemetry_decoder.cc:53] role TelemetryDecoder_1C
[... cut ...]
I0908 10:54:51.690043  4181 dll_pll_veml_tracking.cc:547] Starting tracking of satellite GPS PRN 01 (Block IIF) on channel 0
I0908 10:54:51.690055  4181 dll_pll_veml_tracking.cc:554] PULL-IN Doppler [Hz] = -8500. Code Phase correction [samples] = 0.0460553. PULL-IN Code Phase [samples] = 3408.95
I0908 10:54:51.690081  4181 channel_fsm.cc:83] CH = 0. Ev valid acquisition
I0908 10:54:51.690105  4173 gnss_flowgraph.cc:821] Received 1 from 0. Number of applied actions = 91
I0908 10:54:51.690148  4173 gnss_flowgraph.cc:905] Channel 0 ACQ SUCCESS satellite GPS PRN 01 (Block IIF)

closed 2019-08-02 by carlesfernandez