gnss-sdr/gnss-sdr #214: Freq_Xlating_Fir_Filter with filter_type=bandpass causes an error: munmap_chunk(): invalid pointer.

sergey-nik, 2018-09-13:
Hi.

I use Freq_Xlating_Fir_Filter to work with the signal modulated at a given intermediate frequency.
When I set *filter_type=bandpass*, the program exits with an "invalid pointer" error.

The program output:

```
$ gnss-sdr --config_file=./my-test-GNSS-SDR-receiver.conf --log_dir=.
Initializing GNSS-SDR v0.0.9.git-next-bd1bd468 ... Please wait.
Logging will be written at .
Processing file ./raw/l1rx_GPSGAL.dat, which contains 1604911104 [bytes]
GNSS signal recorded time to be processed: 61.7254 [s]
munmap_chunk(): invalid pointer
Aborted (core dumped)
```

The log contains only general information (no errors):

```
Log file created at: 2018/09/13 16:18:26
Running on machine: serg-virt
Log line format: [IWEF]mmdd hh:mm:ss.uuuuuu threadid file:line] msg
I0913 16:18:26.233398  5120 gnss_block_factory.cc:198] Getting SignalSource with implementation File_Signal_Source
```

This happens in the very beginning at *src/algorithms/input_filter/adapters/freq_xlating_fir_filter.cc*, line 108:

    taps_d = gr::filter::pm_remez(number_of_taps - 1, bands, ampl, error_w, filter_type, grid_density);

In my particular case, the arguments are the following:

```
number_of_taps = 6
bands = {..some data..}
ampl = {..some data..}
error_w = {..some data..}
filter_type = "bandpass"
grid_density = 16
```

My OS: Kubuntu 18.04.
GNSS-SDR is installed using PyBOMBS, I'm using branch *next*.
The gnuradio version 3.7.12.

My receiver configuration (the source and conditioner parts):

```
;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=File_Signal_Source
SignalSource.filename=./raw/my_test.dat
SignalSource.item_type=byte
SignalSource.sampling_frequency=26000000
SignalSource.samples=0
SignalSource.repeat=false
SignalSource.dump=false
SignalSource.enable_throttle_control=false

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

DataTypeAdapter.implementation=Pass_Through
DataTypeAdapter.item_type=byte

InputFilter.implementation=Freq_Xlating_Fir_Filter
InputFilter.filter_type=bandpass
InputFilter.sampling_frequency=26000000
InputFilter.IF=6390000
InputFilter.input_item_type=byte
InputFilter.output_item_type=gr_complex
InputFilter.dump=false
InputFilter.taps_item_type=float

Resampler.implementation=Pass_Through
```

The same happens in Kubuntu 17.04 for package-installation of the dependencies, GNSS-SDR branch *master*, and *gnuradio* 3.7.10.

Kind regards.
S.

sergey-nik, 2018-09-24:
Issue has been solved by specifying a band. Basically, at least one band is necessary for the method **pm_remez** in this case, otherwise the method fails without any messages.

BTW, if I want to add a check for the validity of the arguments passed to this method, is this a right place to do so? (*freq_xlating_fir_filter.cc*, before the call of *pm_remez(...)*)

carlesfernandez, 2018-10-03:
Thank you for raising this point. Those validity checks should be implemented in the corresponding adapter, so in https://github.com/gnss-sdr/gnss-sdr/blob/next/src/algorithms/input_filter/adapters/freq_xlating_fir_filter.cc

closed 2018-10-03 by sergey-nik