gnss-sdr/gnss-sdr #218: Hack RF and GNSS-SDR - lot of loss of lock and no PVT

antoinebarre, 2018-10-13:
Hello everybody,

First thanks you very much for this wonderfull and amazing project in order to analyze in detail the GNSS functions.

I tried to do a real time PVT calculus with one HackRF and this antenna (https://www.amazon.fr/gp/product/B074V12W73/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1)
After various test, I recorded a lot of data but GNSS SDR did not provide PVT or FIX.
It provided this information:

Loss of lock in channel 1!
Current input signal time = 343 [s]
OTracking start on channel 2 for satellite GPS PRN 17 (Block IIR-M)
OOOLoss of lock in channel 3!
OTracking start on channel 1 for satellite GPS PRN 22 (Block IIR)
OOOOTracking start on channel 3 for satellite GPS PRN 31 (Block IIR-M)
OOOOOTracking start on channel 4 for satellite GPS PRN 06 (Block IIF)
Current input signal time = 344 [s]
OOOLoss of lock in channel 6!
OOLoss of lock in channel 0!
OLoss of lock in channel 2!
OOOTracking start on channel 5 for satellite GPS PRN 19 (Block IIR)
OTracking start on channel 0 for satellite GPS PRN 03 (Block IIF)
OCurrent input signal time = 345 [s]
Loss of lock in channel 3!
OLoss of lock in channel 1!
Tracking start on channel 2 for satellite GPS PRN 17 (Block IIR-M)
OLoss of lock in channel 4!
OOTracking start on channel 1 for satellite GPS PRN 31 (Block IIR-M)
OTracking start on channel 3 for satellite GPS PRN 22 (Block IIR)
OTracking start on channel 4 for satellite GPS PRN 06 (Block IIF)

I waited 45 minutes but it was the same result. I tried to change the antenna location = same results. I used this configuration from the GNSS SDR website:
_[GNSS-SDR]`

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_hz=2000000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=Osmosdr_Signal_Source
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=2000000
SignalSource.freq=1575420000
SignalSource.gain=40
SignalSource.rf_gain=40
SignalSource.if_gain=30
SignalSource.AGC_enabled=false
SignalSource.samples=0
SignalSource.repeat=false
;# Next line enables the internal HackRF One bias (3.3 VDC)
SignalSource.osmosdr_args=hackrf,bias=1
SignalSource.enable_throttle_control=false
SignalSource.dump=false
SignalSource.dump_filename=./signal_source.dat

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Pass_Through

;######### INPUT_FILTER CONFIG ############
InputFilter.implementation=Freq_Xlating_Fir_Filter
InputFilter.decimation_factor=1
InputFilter.input_item_type=gr_complex
InputFilter.output_item_type=gr_complex
InputFilter.taps_item_type=float
InputFilter.number_of_taps=5
InputFilter.number_of_bands=2
InputFilter.band1_begin=0.0
InputFilter.band1_end=0.85
InputFilter.band2_begin=0.9
InputFilter.band2_end=1.0
InputFilter.ampl1_begin=1.0
InputFilter.ampl1_end=1.0
InputFilter.ampl2_begin=0.0
InputFilter.ampl2_end=0.0
InputFilter.band1_error=1.0
InputFilter.band2_error=1.0
InputFilter.filter_type=bandpass
InputFilter.grid_density=16
InputFilter.dump=false
InputFilter.dump_filename=../data/input_filter.dat

;######### RESAMPLER CONFIG ############
Resampler.implementation=Pass_Through

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=8
Channels.in_acquisition=1
Channel.signal=1C

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition_Fine_Doppler
Acquisition_1C.item_type=gr_complex
Acquisition_1C.if=0
Acquisition_1C.sampled_ms=1
Acquisition_1C.threshold=0.015
Acquisition_1C.doppler_max=10000
Acquisition_1C.doppler_min=-10000
Acquisition_1C.doppler_step=500
Acquisition_1C.max_dwells=15
Acquisition_1C.dump=false
Acquisition_1C.dump_filename=./acq_dump.dat

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.if=0
Tracking_1C.pll_bw_hz=40.0;
Tracking_1C.dll_bw_hz=2.0;
Tracking_1C.order=3;
Tracking_1C.early_late_space_chips=0.5;
Tracking_1C.dump=false
Tracking_1C.dump_filename=./tracking_ch_

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder
TelemetryDecoder_1C.dump=false
TelemetryDecoder_1C.decimation_factor=1;

;######### OBSERVABLES CONFIG ############
Observables.implementation=GPS_L1_CA_Observables
Observables.dump=true
Observables.dump_filename=./observables.dat

;######### PVT CONFIG ############
PVT.implementation=GPS_L1_CA_PVT
PVT.flag_averaging=true
PVT.averaging_depth=5
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.flag_nmea_tty_port=false;
PVT.nmea_dump_devname=/dev/pts/4
PVT.nmea_dump_filename=./gnss_sdr_pvt.nmea;
PVT.flag_rtcm_server=false
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.dump=false
PVT.dump_filename=./PVT

;######### PVT CONFIG if using the next branch ####
;PVT.implementation=RTKLIB_PVT
;PVT.positioning_mode=Single
;PVT.output_rate_ms=100
;PVT.display_rate_ms=500
;PVT.iono_model=Broadcast
;PVT.trop_model=Saastamoinen
;PVT.flag_rtcm_server=true
;PVT.flag_rtcm_tty_port=false
;PVT.rtcm_dump_devname=/dev/pts/1
;PVT.rtcm_tcp_port=2101
;PVT.rtcm_MT1019_rate_ms=5000
;PVT.rtcm_MT1077_rate_ms=1000
;PVT.rinex_version=2_

Do you have ideas or answers to this problem?
Hardware or software problem?
Thanks you in advance

Antoine

odrisci, 2018-10-13:
The 'O' in the output shows that the driver is dropping samples. Your host
can't keep up with the sampling rate. Try adding

```
Acquisition_1C.blocking=false
```

To the conf file. Probably the issue is the fft in the acquisition block

antoinebarre, 2018-10-13:
thanks a lot for your quick answer.
i added "Acquisition_1C.blocking=false"
results : less O but no fix

I tried also to decrease to 250 the doppler step without good results
_________________________________________________________________________________
Tracking start on channel 2 for satellite GPS PRN 23 (Block IIR)
Tracking start on channel 0 for satellite GPS PRN 02 (Block IIR)
Current input signal time = 111 [s]
Current input signal time = 112 [s]
Current input signal time = 113 [s]
Current input signal time = 114 [s]
Tracking start on channel 1 for satellite GPS PRN 20 (Block IIR)
OCurrent input signal time = 115 [s]
OOOCurrent input signal time = 116 [s]
Loss of lock in channel 1!
Loss of lock in channel 0!
Loss of lock in channel 2!
Tracking start on channel 3 for satellite GPS PRN 21 (Block IIR)
OTracking start on channel 0 for satellite GPS PRN 02 (Block IIR)
OCurrent input signal time = 117 [s]
OTracking start on channel 1 for satellite GPS PRN 23 (Block IIR)
OOOOOOTracking start on channel 2 for satellite GPS PRN 09 (Block IIF)
OOCurrent input signal time = 118 [s]
OLoss of lock in channel 3!
Tracking start on channel 4 for satellite GPS PRN 12 (Block IIR-M)
Loss of lock in channel 0!
OLoss of lock in channel 1!
OCurrent input signal time = 119 [s]
OTracking start on channel 3 for satellite GPS PRN 02 (Block IIR)
OTracking start on channel 0 for satellite GPS PRN 23 (Block IIR)
OOLoss of lock in channel 2!
OOCurrent input signal time = 120 [s]
OLoss of lock in channel 4!
OOOOOLoss of lock in channel 0!
OOOOLoss of lock in channel 3!
OCurrent input signal time = 121 [s]
OOTracking start on channel 1 for satellite GPS PRN 23 (Block IIR)
OOOOTracking start on channel 0 for satellite GPS PRN 02 (Block IIR)
OOOOCurrent input signal time = 122 [s]
OOOOOOOOLoss of lock in channel 1!
Current input signal time = 123 [s]
OLoss of lock in channel 0!
OOTracking start on channel 2 for satellite GPS PRN 23 (Block IIR)
Tracking start on channel 0 for satellite GPS PRN 02 (Block IIR)
OOCurrent input signal time = 124 [s]
OOOOOOOCurrent input signal time = 125 [s]
OOLoss of lock in channel 2!
Loss of lock in channel 0!
OCurrent input signal time = 126 [s]
Tracking start on channel 1 for satellite GPS PRN 23 (Block IIR)
Tracking start on channel 0 for satellite GPS PRN 02 (Block IIR)
OOCurrent input signal time = 127 [s]
OOLoss of lock in channel 1!
Current input signal time = 128 [s]
Loss of lock in channel 0!
Tracking start on channel 2 for satellite GPS PRN 23 (Block IIR)
Tracking start on channel 0 for satellite GPS PRN 02 (Block IIR)
Current input signal time = 129 [s]
OOCurrent input signal time = 130 [s]
OOLoss of lock in channel 2!
Loss of lock in channel 0!
Current input signal time = 131 [s]
Tracking start on channel 1 for satellite GPS PRN 23 (Block IIR)
Tracking start on channel 0 for satellite GPS PRN 02 (Block IIR)
OTracking start on channel 2 for satellite GPS PRN 09 (Block IIF)
OCurrent input signal time = 132 [s]
OOOOOOOCurrent input signal time = 133 [s]
OLoss of lock in channel 1!
Loss of lock in channel 2!
OOLoss of lock in channel 0!
OTracking start on channel 3 for satellite GPS PRN 23 (Block IIR)
Current input signal time = 134 [s]
OOTracking start on channel 0 for satellite GPS PRN 02 (Block IIR)
OOCurrent input signal time = 135 [s]
Tracking start on channel 1 for satellite GPS PRN 06 (Block IIF)
OOLoss of lock in channel 3!
Current input signal time = 136 [s]
Loss of lock in channel 0!
OTracking start on channel 2 for satellite GPS PRN 23 (Block IIR)
Loss of lock in channel 1!
Tracking start on channel 0 for satellite GPS PRN 02 (Block IIR)
Current input signal time = 137 [s]
OLoss of lock in channel 2!
OCurrent input signal time = 138 [s]
Tracking start on channel 1 for satellite GPS PRN 06 (Block IIF)

odrisci, 2018-10-13:
Ok, your processing can't keep up, which shouldn't happen at 2Msps. Possible causes:

  1. You are using a very slow processor, or microcontroller
  2. You are running in a VM with slow transfer from USB
  3. You're running a debug build

The last option is the most likely in my experience, forgetting to switch to a release build after debugging has stung me on many occasions

odrisci, 2018-10-13:
Oh, and make sure to run volk_profile and volk_gnsssdr_profile. This will ensure the processing is optimized for your platform

antoinebarre, 2018-10-13:
How do you active volk options?
If it is a configuration problem, it will be better to record RF and replay with GNSS SDR?
Concerning the possible causes :+1: i used this configuration (thanks screenfish :) )
                          ./+o+-       
                  yyyyy- -yyyyyy+      OS: Ubuntu 18.04 bionic
               ://+//////-yyyyyyo      Kernel: x86_64 Linux 4.15.0-36-generic
           .++ .:/++++++/-.+sss/`      Uptime: 5h 55m
         .:++o:  /++++++++/:--:/-      Packages: 1962
        o:+o+:++.`..```.-/oo+++++/     Shell: bash 4.4.19
       .:+o:+o/.          `+sssoo+/    Resolution: 1366x768
  .++/+:+oo+o:`             /sssooo.   DE: GNOME 
 /+++//+:`oo+o               /::--:.   WM: GNOME Shell
 \+/+o+++`o++o               ++////.   WM Theme: Adwaita
  .++.o+++oo+:`             /dddhhh.   GTK Theme: Adwaita [GTK2/3]
       .+.o+oo:.          `oddhhhh+    Icon Theme: Adwaita
        \+.++o+o``-````.:ohdhhhhh+     Font: Cantarell 11
         `:o+++ `ohhhhhhhhyo++os:      CPU: Intel Core i5-2430M @ 4x 2.4GHz [92.0°C]
           .o:`.syhhhhhhh/.oo++o`      GPU: intel
               /osyyyyyyo++ooo+++/     RAM: 2158MiB / 7885MiB
                   ````` +oo+++o\:    
                          `oo++.

odrisci, 2018-10-13:
That's a nice machine, should have no trouble processing 2Msps. Have you compiled gnss-sdr yourself? If so, ensure it is a release build, not a debug build.

From the command line run:

```sh
volk_profile
volk_gnsssdr_profile
```

This will generate the volk config files for you

antoinebarre, 2018-10-13:
Thanks a lot for your help
Vold and Volg_gnsssdr_profile
Less O
but GNSS SDR does not provide localization. 
Alway the lock and loss of lock process
Very strange

I'm using the official gnss-sdr throught the apt-get install gnss-sdr

odrisci, 2018-10-13:
I think that is a relatively old version, so the blocking option probably had no effect. It should still be able to handle the lie sampling rate though. Are you running in a virtual machine?

Otherwise, try a different USB port. After that I'm out of ideas, sorry!

MaxZenith, 2019-03-13:
Antoinebarre did you ever find a solution to your problem? I am also seeing the same thing.

chellaKR, 2019-04-19:
How did you go forward?? Any success ?

carlesfernandez, 2025-03-14:
Please feel free to open a new issue if the problem persists.

closed 2025-03-14 by carlesfernandez