gnss-sdr/gnss-sdr #222: Getting a location fix using RTL-SDR

arani89, 2018-11-12:
I am using the new RTL dongles with TCXO + Bias-T with an active antenna to get GNSS signals (using gnss-sdr version v0.0.9). While I am able to get messages, I was never able to get a real position fix in real time. I am using the config file `gnss-sdr_GPS_L1_rtlsdr_realtime.conf` given with the distribution. What I get is something like this:

> New GPS NAV message received in channel 1: subframe 3 from satellite GPS PRN 05 (Block IIR-M)
> New GPS NAV message received in channel 3: subframe 3 from satellite GPS PRN 13 (Block IIR)
> New GPS NAV message received in channel 0: subframe 3 from satellite GPS PRN 15 (Block IIR-M)
> Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 12 (Block IIR-M)
> Current receiver time: 8 min 14 s
> Loss of lock in channel 2!
> Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 12 (Block IIR-M)
> Current receiver time: 8 min 15 s
> Loss of lock in channel 2!
> Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 12 (Block IIR-M)
> Current receiver time: 8 min 16 s
> Loss of lock in channel 2!
> Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 12 (Block IIR-M)
> Current receiver time: 8 min 17 s
> Loss of lock in channel 2!

My understanding about why it is taking so long to get a position fix is that Channel 2 is always looking for the same satellite (GPS PRN 12). Isn't it possible to check for more satellites and then decide which one provides the best fix? I believe this could be changed from the config file but could not figure out exactly how. My computer has sufficient compute power, so this is not a concern. I already tried to change the parameter `Channels_1C.count` but it does not give a position though it looks for more channels. Any suggestions if this is possible, or is this not feasible (either because the code does not permit or the algorithm does not work that way)?

Thanks a lot for looking into this!

arani89, 2019-06-07:
I got this -- just increasing the number of channels is enough to do this.

closed 2019-06-07 by arani89

andreevdanil, 2020-03-06:
Hi, can you share your config file?