gnss-sdr/gnss-sdr #224: NAV data but no location (Ettus N210)

rfpds1-iscteiulpt, 2018-11-14:
I used this config file:

https://github.com/gnss-sdr/gnss-sdr/blob/master/conf/gnss-sdr_GPS_L1_USRP_X300_realtime.conf

I'm getting navigation data but no location, also some lock loss. I noticed improvements by attaching the antenna to RF2, and it's being fed by a bias-t. I have a u-blox evk-m8t and it detects the gps location, I have the antenna outside and it has a wall above it, might be that?

[image]

rfpds1-iscteiulpt, 2018-11-14:
Might be that I'm only getting ephemeris from 3 satellites and therefore the lock only is working for those 3 (I need 4)?

ZaquL, 2022-06-29:
> only is working for those 3 (I need 4)?

You do not need 4. Lock can work with 3 but you need atomic clock in sync with optical fountain in Geneva to get the lock.