gnss-sdr/gnss-sdr #226: Loss of lock in real time acquisition with bladeRF

matteocutugno, 2018-11-26:
Hello to everyone, 
it's been a month that i'm trying to get PVT with GNSS-SDR and Nuand BladeRF x40.
I feeded the antenna with a bias-tee providing 5V from USB port. I also tryied with different laptops, with a lot of different configuration files and with different version of GNSS-SDR(next and master branch). GNSS-SDR is able to acquire L1 signal and declares positive acquisition, but it fails everytime the tracking with consequent loss of lock.
Below i post my conf file and part of the log file.
Thanks in advance for your answer

[GNSS-SDR]
;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=4000000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=Osmosdr_Signal_Source
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=4000000
SignalSource.freq=1575420000
;# RF Gain: LNA Gain {0, 3, 6}
SignalSource.gain=6
;# IF Gain: N/A
SignalSource.rf_gain=30
;# BB Gain: RX VGA1 + VGA2 [5, 60]
SignalSource.if_gain=33
SignalSource.AGC_enabled=true
SignalSource.samples=0
SignalSource.repeat=true
SignalSource.osmosdr_args=bladerf,bias=1
SignalSource.enable_throttle_control=false
SignalSource.dump=true
SignalSource.dump_filename=./signal_source.dat

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Pass_Through

;######### INPUT_FILTER CONFIG ############
InputFilter.implementation=Pass_Through

;######### RESAMPLER CONFIG ############
Resampler.implementation=Pass_Through

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=8
Channels.in_acquisition=1
Channel.signal=1C

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition_Fine_Doppler
Acquisition_1C.item_type=gr_complex
Acquisition_1C.coherent_integration_time_ms=1
Acquisition_1C.threshold=0.0015
Acquisition_1C.doppler_max=20000
Acquisition_1C.doppler_min=-20000
Acquisition_1C.doppler_step=200
Acquisition_1C.max_dwells=15
Acquisition_1C.make_two_steps=true
Acquisition_1C.second_nbins=4
Acquisition_1C.second_doppler_step=125
Acquisition_1C.dump=false
Acquisition_1C.dump_filename=./acq_dump.dat

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.pll_bw_hz=30.0;
Tracking_1C.dll_bw_hz=3.0;
Tracking_1C.order=3;
Tracking_1C.early_late_space_chips=0.5;
Tracking_1C.dump=false
Tracking_1C.dump_filename=./tracking_ch_
[... 690 lines cut ...]
I1123 15:04:55.503959  1995 pcps_acquisition_fine_doppler_cc.cc:605] doppler -1875
I1123 15:04:55.504046  1959 dll_pll_veml_tracking.cc:469] Number of samples between Acquisition and Tracking = 12382
I1123 15:04:55.504053  1959 dll_pll_veml_tracking.cc:470] Number of seconds between Acquisition and Tracking = 0.0030955
I1123 15:04:55.504271  1959 dll_pll_veml_tracking.cc:601] Starting tracking of satellite GPS PRN 09 (Block IIF) on channel 1
I1123 15:04:55.504290  1959 dll_pll_veml_tracking.cc:608] PULL-IN Doppler [Hz] = -1875. Code Phase correction [samples] = 0.0147366. PULL-IN Code Phase [samples] = 2666.99
I1123 15:04:55.504324  1959 channel_fsm.cc:105] CH = 1. Ev valid acquisition
I1123 15:04:55.504349  1949 gnss_flowgraph.cc:837] Received 1 from 1. Number of applied actions = 9
I1123 15:04:55.504372  1949 gnss_flowgraph.cc:929] Channel 1 ACQ SUCCESS satellite GPS PRN 09 (Block IIF)
I1123 15:04:55.504571  1949 gnss_flowgraph.cc:991] Channel 0 in state 2
I1123 15:04:55.504581  1949 gnss_flowgraph.cc:991] Channel 1 in state 2
I1123 15:04:55.504611  1949 gnss_flowgraph.cc:991] Channel 2 in state 2
I1123 15:04:55.504618  1949 gnss_flowgraph.cc:991] Channel 3 in state 2
I1123 15:04:55.504626  1949 gnss_flowgraph.cc:991] Channel 4 in state 2
I1123 15:04:55.504633  1949 gnss_flowgraph.cc:991] Channel 5 in state 2
I1123 15:04:55.504640  1949 gnss_flowgraph.cc:991] Channel 6 in state 2
I1123 15:04:55.504647  1949 gnss_flowgraph.cc:991] Channel 7 in state 2
I1123 15:04:55.504650  1949 control_thread.cc:713] Processed all control messages
I1123 15:04:55.504654  1949 control_thread.cc:678] Reading control messages from queue
I1123 15:04:55.505589  1997 dll_pll_veml_tracking.cc:728] Loss of lock in channel 0!
I1123 15:04:55.505688  1960 channel_fsm.cc:156] CH = 0. Ev failed tracking standby
I1123 15:04:55.505703  1949 gnss_flowgraph.cc:837] Received 2 from 0. Number of applied actions = 10
I1123 15:04:55.505718  1949 gnss_flowgraph.cc:996] Channel 0 TRK FAILED satellite GPS PRN 01 (Block IIF)
I1123 15:04:55.505735  1949 gnss_flowgraph.cc:997] Number of channels in acquisition = 0
I1123 15:04:55.505738  1949 gnss_flowgraph.cc:1003] Channel 0 Starting acquisition GPS PRN 01 (Block IIF), Signal 1C
I1123 15:04:55.505758  1949 channel_fsm.cc:88] CH = 0. Ev start acquisition

matteocutugno, 2018-12-06:
UPDATE
I was able to get PVT with nuand bladeRF .
Now i'm trying to make an acquisition in multicostellation mode (single band).
Can someone share a working .conf file usefull for acquisition of GPS L1, Galileo E1 and Glonass L1???
Thanks

oscarmezyk, 2018-12-10:
@matteocutugno could you please let me know what have you done to make bladeRF working? Thanks in advance

matteocutugno, 2018-12-11:
I used the .conf file posted below.
I'm experiencing continuous losses of lock but the receiver can receive and decode GPS messages and after 5-6 minutes is able to compute PVT. The computation is done with 4 or maximum 5 satellites so the accuracy is quite bad, but IT WORKS.

[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=2000000

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=Osmosdr_Signal_Source
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=2000000
SignalSource.freq=1575420000
;# RF Gain: LNA Gain {0, 3, 6}
SignalSource.gain=6
;# IF Gain: N/A
SignalSource.rf_gain=40
;# BB Gain: RX VGA1 + VGA2 [5, 60]
SignalSource.if_gain=48
SignalSource.AGC_enabled=false
SignalSource.samples=0
SignalSource.repeat=false
SignalSource.osmosdr_args=bladerf=0  ; This line enables the bladeRF
SignalSource.enable_throttle_control=false
SignalSource.dump=false
SignalSource.dump_filename=./signal_source.dat

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

;######### DATA_TYPE_ADAPTER CONFIG ############
DataTypeAdapter.implementation=Pass_Through

;######### INPUT_FILTER CONFIG ############
InputFilter.implementation=Freq_Xlating_Fir_Filter
InputFilter.decimation_factor=1
InputFilter.input_item_type=gr_complex
InputFilter.output_item_type=gr_complex
InputFilter.taps_item_type=float
InputFilter.number_of_taps=5
InputFilter.number_of_bands=2
InputFilter.band1_begin=0.0
InputFilter.band1_end=0.85
InputFilter.band2_begin=0.9
InputFilter.band2_end=1.0
InputFilter.ampl1_begin=1.0
InputFilter.ampl1_end=1.0
InputFilter.ampl2_begin=0.0
InputFilter.ampl2_end=0.0
InputFilter.band1_error=1.0
InputFilter.band2_error=1.0
InputFilter.filter_type=bandpass
InputFilter.grid_density=16
InputFilter.dump=false
InputFilter.dump_filename=../data/input_filter.dat

;######### RESAMPLER CONFIG ############
Resampler.implementation=Pass_Through

;######### CHANNELS GLOBAL CONFIG ############
Channels_1C.count=8
Channels.in_acquisition=1
Channel.signal=1C

;######### ACQUISITION GLOBAL CONFIG ############
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition_Fine_Doppler
Acquisition_1C.item_type=gr_complex
Acquisition_1C.coherent_integration_time_ms=1
Acquisition_1C.threshold=0.015
Acquisition_1C.doppler_max=10000
Acquisition_1C.doppler_step=250
Acquisition_1C.max_dwells=15
Acquisition_1C.repeat_satellite=true
Acquisition_1C.dump=false
Acquisition_1C.dump_filename=./acq_dump.dat

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.pll_bw_hz=40.0;
Tracking_1C.dll_bw_hz=2.0;
Tracking_1C.order=3;
Tracking_1C.early_late_space_chips=0.5;
Tracking_1C.dump=false
Tracking_1C.dump_filename=./tracking_ch_

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder
TelemetryDecoder_1C.dump=false

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=true
Observables.dump_filename=./observables.dat
Observables.dump_mat=true

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Single
PVT.output_rate_ms=100
PVT.display_rate_ms=500
PVT.iono_model=Broadcast
PVT.trop_model=Saastamoinen
PVT.flag_rtcm_server=false
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.rtcm_tcp_port=2101
PVT.rtcm_MT1019_rate_ms=5000
PVT.rtcm_MT1077_rate_ms=1000
PVT.rinex_version=2

oscarmezyk, 2019-01-05:
Hi, after some time playing with signals it seems that:

1) I needed to take antenna to have a really good sky view - not like laying on the desk near the window but on the balcony on a stick; additionally it must have really poor gain - I needed to add another 20dB LNA to start getting any output. After a while it seems that for poor signal gnss-sdr just needs more time and the additional LNA just speeds things up - sometimes significantly. I will buy "a real-man's 30dB" GPS antenna next week and will see if there is a difference. Overall this seems to make the biggest difference.

2) setting gains in gnss-sdr setup file for osmosdr signal doesn't work for me; I needed to "set agc off" and "set gain rx 48" in bladerf-cli; I commented out gains in config file but if they work for you you can leave them

3) setting bladeRF gains higher seems to worsen things; when looking at FreqSink in gnuradio it seems higher gains tend to introduce more spurs

4) at this point in bladerf-cli you do: "cal lms" and "cal rx dc" (with antenna unplugged) for better IQ dc offsets

5) I added support for SUPL in my config, which seems not to help too much except if you run things several times - then the ephemeris and iono models may be acquired faster and PVT data can show up faster

6) I have changed the filter to normal FirFilter and commented it out - it does not seem to do a good job or is not really needed;  I have tested the filter on the sample "SPAIN....dat" file from the tutorial and seems that it makes it impossible to calculate PVT when used; Anyway bladeRF has it's own pretty good input filters.

7) other changes aren't really justified - I skimmed through different .conf files and played a bit with settings of tracking/acqusition blocks.

Make sure that you set correct data for SUPL if used

; This is a GNSS-SDR configuration file
; The configuration API is described at https://gnss-sdr.org/docs/sp-blocks/

[GNSS-SDR]

;######### GLOBAL OPTIONS ##################
GNSS-SDR.internal_fs_sps=4000000

;######### SUPL RRLP GPS assistance configuration #####
; Check http://www.mcc-mnc.com/
; On Android: https://play.google.com/store/apps/details?id=net.its_here.cellidinfo&hl=en
GNSS-SDR.SUPL_gps_enabled=false
GNSS-SDR.SUPL_read_gps_assistance_xml=false
GNSS-SDR.SUPL_gps_ephemeris_server=supl.google.com
GNSS-SDR.SUPL_gps_ephemeris_port=7275
GNSS-SDR.SUPL_gps_acquisition_server=supl.google.com
GNSS-SDR.SUPL_gps_acquisition_port=7275
GNSS-SDR.SUPL_MCC=260 ;Find for your country - 260 is Poland
GNSS-SDR.SUPL_MNC=03 ;Find for operator of the near-by BTS
GNSS-SDR.SUPL_LAC=56110 ;LAC/TAC - local area code locate near-by BTS, use app in android
GNSS-SDR.SUPL_CI=47768

;######### AGNSS GPS assistance configuration #####
; Check http://www.mcc-mnc.com/
; On Android: https://play.google.com/store/apps/details?id=net.its_here.cellidinfo&hl=en
GNSS-SDR.AGNSS_XML_enabled=false
GNSS-SDR.AGNSS_ref_location= 51.9271, 15.4904

;######### SIGNAL_SOURCE CONFIG ############
SignalSource.implementation=Osmosdr_Signal_Source
SignalSource.item_type=gr_complex
SignalSource.sampling_frequency=4000000
SignalSource.freq=1575420000
;#SignalSource.gain_mode=automatic
;#SignalSource.AGC_enabled=false
;#RF Gain: LNA Gain {0, 3, 6}
;#SignalSource.gain=45
;#IF Gain: N/A
;#SignalSource.rf_gain=45
;#BB Gain: RX VGA1 + VGA2 [5, 60]
;#SignalSource.bb_gain=45
;#set gains in bladerf-cli !!
SignalSource.samples=0
SignalSource.repeat=false
SignalSource.osmosdr_args=bladerf=0  ; This line enables the bladeRF
SignalSource.enable_throttle_control=false
SignalSource.dump=false
SignalSource.dump_filename=./signal_source.dat

;######### SIGNAL_CONDITIONER CONFIG ############
SignalConditioner.implementation=Signal_Conditioner

[... 60 lines cut ...]
;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder
TelemetryDecoder_1C.dump=false

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables
Observables.dump=false
Observables.dump_filename=./observables.dat

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.positioning_mode=Static  ; options: Single, Static, Kinematic, PPP_Static, PPP_Kinematic
PVT.iono_model=Broadcast ; options: OFF, Broadcast, SBAS, Iono-Free-LC, Estimate_STEC, IONEX
PVT.trop_model=Saastamoinen ; options: OFF, Saastamoinen, SBAS, Estimate_ZTD, Estimate_ZTD_Grad
PVT.output_rate_ms=500
PVT.display_rate_ms=1000
PVT.output_path = ./gnss_output
PVT.dump_filename=./PVT
PVT.nmea_dump_filename=./gnss_sdr_pvt.nmea;
PVT.flag_nmea_tty_port=false;
PVT.nmea_dump_devname=/dev/pts/4
PVT.flag_rtcm_server=false
PVT.flag_rtcm_tty_port=false
PVT.rtcm_dump_devname=/dev/pts/1
PVT.dump=false

oscarmezyk, 2019-01-10:
one additional note - it seems that setting bandwitdth to 3000000 also helps a bit. (needs to be done in bladerf-cli)

vrushankstark, 2019-05-03:
Hi I'm working with RTL-SDR V3. I'm continuously getting loss of lock on all channels when I tried running the code above. Could you please help me with the settings. Thanks in advance.

matteocutugno, 2020-03-09:
I post a link for an article that can interest you if you are testing gnss-sdr with low-cost front-ends.

https://www.researchgate.net/publication/339417340_Low-Cost_GNSS_Software_Receiver_Performance_Assessment

In the article are shown all the settings and the parameters useful to reproduce the experiment.

MC

closed 2020-09-18 by matteocutugno