gnss-sdr/gnss-sdr #227: Correct output from My first position fix?

mschuldt, 2018-12-07:
I seem to have gotten the example from My first position fix working but I'm not sure if it's fully correct as my output differs a lot from the sample output in 2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.txt

The only line I get related to the position fix is:
> First position fix at 2013-Apr-04 06:24:02.050000 UTC is Lat = 41.2748 [deg], Long = 1.98757 [deg], Height= 83.766 [m]

While the example output has 100+ lines related to position, such as:
> Position at 2013-Apr-04 06:23:48 is Lat = 41.27471731628889 [deg], Long = 1.987497969066093 [deg], Height= 65.24936532508582 [m]

The example output first finds the position after 26 seconds of signal input / receiver time while the position in my output was found after 56 seconds

Can anyone please provide the updated output that I should expect for this sample data? Or at least let me know if this is expected. Thanks!

Some notes on my setup:

I followed the directions from https://gnss-sdr.org/my-first-fix/ and downloaded the configuration from that page into gps.conf, I've had to make a few changes to it, as suggested by gnss-sdr failure messages, I changed the lines:
> GNSS-SDR.internal_fs_hz=2000000
> Observables.implementation=GPS_L1_CA_Observables
> PVT.implementation=GPS_L1_CA_PVT

to:
>GNSS-SDR.internal_fs_sps=2000000
>Observables.implementation=Hybrid_Observables
>PVT.implementation=RTKLIB_PVT

I ran it with:
> gnss-sdr --config_file=gps.conf

Using gnss-sdr version 0.0.9.git-next-bdb912e9da

acebrianjuan, 2019-01-16:
Hi Michael,

I've just run the _My first position fix_ tutorial.

If you have a look at my output, you'll notice that I start getting position fixes right about the same time as you (~ 56 seconds), and I do get many position reports afterwards.

I've also attached my configuration file at the end.

I'm using GNSS-SDR v0.0.10.git-next-b6f0c92f on Ubuntu 18.04

I hope this helps.

Álvaro

---

**Output:**
```
alvaro@msi:~/work$ $HOME/git/gnss-sdr/install/gnss-sdr -c ./2013_receiver.conf 
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Initializing GNSS-SDR v0.0.10.git-next-b6f0c92f ... Please wait.
Logging will be written at "/tmp"
Use gnss-sdr --log_dir=/path/to/log to change that.
Processing file /home/alvaro/signals/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat, which contains 1600000000 [bytes]
GNSS signal recorded time to be processed: 99.999 [s]
Current receiver time: 1 s
Tracking of GPS L1 C/A signal started on channel 0 for satellite GPS PRN 01 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 3 for satellite GPS PRN 11 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 7 for satellite GPS PRN 17 (Block IIR-M)
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 19 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 1 for satellite GPS PRN 20 (Block IIR)
Tracking of GPS L1 C/A signal started on channel 6 for satellite GPS PRN 32 (Block IIF)
Current receiver time: 2 s
Loss of lock in channel 2!
Current receiver time: 3 s
Current receiver time: 4 s
Current receiver time: 5 s
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 21 (Block IIR)
Current receiver time: 6 s
Loss of lock in channel 5!
Current receiver time: 7 s
Current receiver time: 8 s
Current receiver time: 9 s
Current receiver time: 10 s
Current receiver time: 11 s
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 19 (Block IIR)
Current receiver time: 12 s
Loss of lock in channel 2!
Current receiver time: 13 s
Current receiver time: 14 s
New GPS NAV message received in channel 1: subframe 1 from satellite GPS PRN 20 (Block IIR)
New GPS NAV message received in channel 6: subframe 1 from satellite GPS PRN 32 (Block IIF)
Current receiver time: 15 s
Current receiver time: 16 s
Tracking of GPS L1 C/A signal started on channel 2 for satellite GPS PRN 27 (Block IIF)
Current receiver time: 17 s
Current receiver time: 18 s
Current receiver time: 19 s
Current receiver time: 20 s
New GPS NAV message received in channel 1: subframe 2 from satellite GPS PRN 20 (Block IIR)
New GPS NAV message received in channel 0: subframe 2 from satellite GPS PRN 01 (Block IIF)
New GPS NAV message received in channel 6: subframe 2 from satellite GPS PRN 32 (Block IIF)
New GPS NAV message received in channel 3: subframe 2 from satellite GPS PRN 11 (Block IIR)
New GPS NAV message received in channel 7: subframe 2 from satellite GPS PRN 17 (Block IIR-M)
Current receiver time: 21 s
Current receiver time: 22 s
Current receiver time: 23 s
Tracking of GPS L1 C/A signal started on channel 5 for satellite GPS PRN 26 (Block IIF)
Tracking of GPS L1 C/A signal started on channel 4 for satellite GPS PRN 21 (Block IIR)
[... 283 lines cut ...]
Acquisition_1C.implementation=GPS_L1_CA_PCPS_Acquisition
Acquisition_1C.item_type=gr_complex
Acquisition_1C.threshold=0.008
Acquisition_1C.doppler_max=10000
Acquisition_1C.doppler_step=250

;######### TRACKING GLOBAL CONFIG ############
Tracking_1C.implementation=GPS_L1_CA_DLL_PLL_Tracking
Tracking_1C.item_type=gr_complex
Tracking_1C.pll_bw_hz=40.0;
Tracking_1C.dll_bw_hz=4.0;

;######### TELEMETRY DECODER GPS CONFIG ############
TelemetryDecoder_1C.implementation=GPS_L1_CA_Telemetry_Decoder

;######### OBSERVABLES CONFIG ############
Observables.implementation=Hybrid_Observables

;######### PVT CONFIG ############
PVT.implementation=RTKLIB_PVT
PVT.averaging_depth=100
PVT.flag_averaging=true
PVT.output_rate_ms=10
PVT.display_rate_ms=500
```

YingGhui, 2020-10-08:
Good day. May I know that is this **2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat**  only for GPS_L1_CA ? Is it possible I use the signal source file provided for testing other signal like Glonass L1 C/A ? If yes what should I do for the next ?
Highly appreciate for your replys. Thank you.

acebrianjuan, 2020-10-08:
Hi @YingGhui,

> I know that is this **2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat** only for GPS_L1_CA ? Is it possible I use the signal source file provided for testing other signal like Glonass L1 C/A ?

No, I don't think you can use this particular file for GLONASS L1. This file is typically used for demonstration purposes for decoding GPS L1.

As mentioned in the [documentation](https://gnss-sdr.org/my-first-fix/#step-2-download-a-file-of-raw-signal-samples), the file `2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat`:

> [...] contains 100 seconds of raw GNSS signal samples collected by an RF front-end **centered at 1575.42 MHz** [...]

But the GLONASS L1 band is **centered at 1602 MHz** (see [documentation](https://gnss-sdr.org/docs/tutorials/gnss-signals/#glonass-l1)), so it does not exactly coincide with the GPS L1 band.

[image]

I don't know the chunk of spectrum that the `2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.dat` file covers, but it is very unlikely that the file contains any usable GLONASS signal at all.

Regards,
Álvaro

closed 2024-08-11 by carlesfernandez